espressif/arduino-esp32 #4046: Writing >20 Bytes to Characteristic causes all registered Write Callbacks to fire

emitategh, 2020-05-31:
As explained in title: When sending more than 20 bytes to any Characteristic causes all registered Write Callbacks to fire.

**Board**: ESP32 DevKit v1
**Chip**: ESP32D0WDQ6 (revision 
**Features**: WiFi, BT, Dual Core, 240MHz
**Auto-detected Flash size**: 4MB
**Core Instalation date**: 01/05/2020
**IDE**: Arduino 1.8.12
**Upload Speed**: 115200
**Computer OS**: Windows 10 Pro

**Description**:
Same as title

**Example Sketch**
`#include <Arduino.h>
#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEServer.h>
#include <BLECharacteristic.h>
// See the following for generating UUIDs:
// https://www.uuidgenerator.net/

#define SERVICE_UUID         "4fafc201-1fb5-459e-8fcc-c5c9c331914b"
#define CHARACTERISTIC1_UUID "aeb5483e-36e1-4688-b7f5-ea07361b26a8"
#define CHARACTERISTIC2_UUID "beb5483e-36e1-4688-b7f5-ea07361b26a8"

class Characteristic1_Callbacks: public BLECharacteristicCallbacks {
  void onRead(BLECharacteristic *pCharacteristic){
    Serial.println("Characteristic1 Read");
  }
  void onWrite(BLECharacteristic *pCharacteristic){
    Serial.print("Characteristic1 Written: ");
    Serial.println((char *) pCharacteristic->getData());
  }
};

class Characteristic2_Callbacks: public BLECharacteristicCallbacks {
  void onRead(BLECharacteristic *pCharacteristic){
    Serial.println("Characteristic2 Read");
  }
  void onWrite(BLECharacteristic *pCharacteristic){
    Serial.print("Characteristic2 Written: ");
    Serial.println((char *) pCharacteristic->getData());
  }
};

void setup() {
  Serial.begin(115200);
  Serial.println("Starting BLE work!");

  BLEDevice::init("Long name works now");
  BLEServer *pServer = BLEDevice::createServer();
  BLEService *pService = pServer->createService(SERVICE_UUID);
  BLECharacteristic *pCharacteristic = pService->createCharacteristic(
                                         CHARACTERISTIC1_UUID,
                                         BLECharacteristic::PROPERTY_READ |
                                         BLECharacteristic::PROPERTY_WRITE
                                       );
  pCharacteristic->setCallbacks(new Characteristic1_Callbacks());
  pCharacteristic->setValue("I am Characteristic #1");

  pCharacteristic = pService->createCharacteristic(
                                         CHARACTERISTIC2_UUID,
                                         BLECharacteristic::PROPERTY_READ |
                                         BLECharacteristic::PROPERTY_WRITE
                                       );
  pCharacteristic->setCallbacks(new Characteristic2_Callbacks());
  pCharacteristic->setValue("I am Characteristic #2");

  pService->start();
  // BLEAdvertising *pAdvertising = pServer->getAdvertising();  // this still is working for backward compatibility
  BLEAdvertising *pAdvertising = BLEDevice::getAdvertising();
  pAdvertising->addServiceUUID(SERVICE_UUID);
  pAdvertising->setScanResponse(true);
  pAdvertising->setMinPreferred(0x06);  // functions that help with iPhone connections issue
  pAdvertising->setMinPreferred(0x12);
  BLEDevice::startAdvertising();
  Serial.println("Characteristic defined! Now you can read it in your phone!");
}

void loop() {
  // put your main code here, to run repeatedly:
  delay(2000);
}`

**Debug Messages**: None

emitategh, 2020-05-31:
I found that the same bug was present in the original Esp32 snippets. The following link is the original issue and was fixed and closed

https://github.com/nkolban/esp32-snippets/issues/796

vicatcu, 2020-05-31:
Yes, I still blow away the BLE files in the v1.0.4 arduino-esp32 board support package release and replace them with what is on master from nkolban/esp32-snippets... I wish this would get fixed in this package so I can stop having to do that. Unfortunate that the vestigial code is more stable than the official code in this regard.

me-no-dev, 2020-05-31:
The reason is that nobody actually made a PR for this. This is not the "official" code here. It's just a snapshot from the moment that Colban started his new job and @chegewara stopped supporting the lib. PRs are welcome

emichristiansen, 2020-06-01:
Sorry, I'm not the right one to open a PR. Have almost 0 experience with C++, I'm learning in the process!
@vicatcu  thanks for the info, I guess I'll do the same.

vicatcu, 2020-06-01:
I can open a PR, it's just likely to be a pretty size-able change set and I'm not really equipped to evaluate the full scope of that change set for correctness. My fear would be that it would overwrite / revert fixes that might exist in this repo. I'll give it a shot anyway.

emitategh, 2020-06-01:
Sorry I deleted the other comment (It was from my job account)

Sorry, I'm not the right one to open a PR. Have almost 0 experience with C++, I'm learning in the process!
@vicatcu thanks for the info, I guess I'll do the same.

Either way, thanks for the product and the library!

TryingToGITthis, 2020-06-01:
> Yes, I still blow away the BLE files in the v1.0.4 arduino-esp32 board support package release and replace them with what is on master from nkolban/esp32-snippets... I wish this would get fixed in this package so I can stop having to do that. Unfortunate that the vestigial code is more stable than the official code in this regard. I'm sure there's a good reason though.

Where are the BLE files in Kolban's snippets? I only see the BLE Scanner. 

After checking heap space last night, I realized that the default BLE consumes 120k of heap!

vicatcu, 2020-06-01:
See my PR... they come from ./cpp_utils

TryingToGITthis, 2020-06-02:
Thanks. I downloaded the files and replaced the Espressif files. However, am I doing something wrong or does BLE really consume so much space?

Free heap before BLE: 147564
BLE Name: Test-8866
Free heap after BLE: 49688

chegewara, 2020-06-02:
> After checking heap space last night, I realized that the default BLE consumes 120k of heap!

about 70kB

geeksville, 2020-06-13:
Btw - I had a variant of this bug also, so I tried @vicatcu's awesome PR.  Alas, I can report a regression compared to the code currently in master:

In my app I use light-sleep, and to do so I have to carefully not just turnoff bluetooth, but stop and delete my various services before going to sleep.  That still works fine.

But when I go to recreate my services after wake, something is spinning somewhere - and that step never completes.  I'll debug more - but I thought I should mention it.

geeksville, 2020-06-13:
Okay - all happy now and OMG this PR fixes nasty BLE problems I was having.  So thank ya'll for your work!

geeksville, 2020-08-12:
any questions on the associated #4086 PR?  I suspect others will want it.

commit 2fd3d042 (2020-09-30): Fix #4046 Details below: (#4086)

Informed by the discussion in the bug and the code in 'that other branch'
the fix was clear.  Just set a flag if we start handling a write, and
use that flag to guard the long write complete call.
closed 2020-09-30 by me-no-dev