espressif/arduino-esp32 #4052: wifi: addba response cb: ap bss deleted

TryingToGITthis, 2020-06-01:
I'm getting this error when WiFi clients, I'm assuming, disconnect. Is it an issue? What does it mean? How do I turn it off?

sebhuet, 2020-06-05:
I have the same firmware running on hundreds of ESP32 i also have this message on some chips (10%) not on others. I suspect chip/design issue but i would be happy to know this could be related with software.

TryingToGITthis, 2020-06-05:
Interesting. I have noticed it is tied to certain boards even running the same code.

sebhuet, 2020-07-02:
Need to be confirmed but it looks like connecting with a fixed ip instead of dhcp lowers the amount of fails. could you test this on your side ?

TryingToGITthis, 2020-07-02:
What I found is that using a fixed IP does help a lot. However, I had to drop back to 1.11.2 to keep my code from having issues. DHCP client just dies in the latest versions and won't give out any new IPs.

CelliesProjects, 2020-07-02:
I only heard about this problem from people using my apps. I could not reproduce it on any of my boards, but indeed a fixed IP solved the problem.

@TryingToGITthis 
>  However, I had to drop back to 1.11.2 to keep my code from having issues.

Drop back what? Unclear what you mean.

TryingToGITthis, 2020-07-02:
Versions 1.12.X of arduino-esp32 cause issues. If I use version 1.11.2 it doesn't experience the problem.

TryingToGITthis, 2020-07-02:
I'm am using ESP32-WROOM-32D.

atanisoft, 2020-07-02:
@TryingToGITthis there is no such version of arduino-esp32. However, what you are showing is a platformio-espressif32 version.

TryingToGITthis, 2020-07-02:
My bad, you are correct.

closed 2020-09-14 by stale[bot]

natitomato, 2023-03-20:
Are you using reconnect()? or passing the ssid and password everytime you connect to WiFi?
If you've been using reconnect, try disconnecting first and then starting the WiFi again using WiFi.being(ssid, pwd)
This solved the issue for me (I'm using WROOM-32U).

kevin4dhd, 2024-02-04:
> I'm am using ESP32-WROOM-32D.

Any solution for this model, I also have the same error in ESP32-WROOM-32D, my wifi disconnects after a few seconds