espressif/arduino-esp32 #4053: ESP32  webserver, login process

hyeriii, 2020-06-02:
ESP32 works as a webserver.
I want to implement user sign-up and login process on web pages
How do I solve this problem

chegewara, 2020-06-02:
with html and javascript.

hyeriii, 2020-06-02:
For example, I would like to register an ID, password and email and log in with this registered account. I created a form in html, but what should I do to store, load and compare the data?

pipi61, 2020-06-02:
https://github.com/espressif/arduino-esp32/tree/master/libraries/HTTPClient/examples/Authorization

hyeriii, 2020-06-10:
Is there a need for a separate DB design like mysql?

lbernstone, 2020-06-10:
This site is not for architectural design philosophy.  It is for issues with the arduino-esp32 code- please read the [ISSUE_TEMPLATE](https://github.com/espressif/arduino-esp32/blob/master/docs/ISSUE_TEMPLATE.md).