espressif/arduino-esp32 #4058: Cannot set new service UUID for BLE server - reinit problem

rob-bits, 2020-06-03:
Problem:
In my application I have to change dynamically the device service UUID. If a request is come (from button) then the previous advertised service UUID must be stopped and advertising must be started on new service UUID.
I tried two scenario:
1. Call BLEDevice::deinit(false); -> If the BLEDevice::init("NODE"); is after deinit I got the following error:
BLEDevice: esp_bt_controller_init: rc=259 Invalid state
2.  BLEDevice::deinit() is not called. The following code is called:

`BLEDevice::getAdvertising()->stop(); //stop advertising`
`pServer->removeService(pService);`
`if(pService != NULL) {  delete pService;   }`
`if(serverClbk != NULL)  {    delete serverClbk;  }`
 ` serverClbk = new MyServerCallbacks();`
 ` pServer->setCallbacks(serverClbk); ` 
 `pService = pServer->createService(mDevice.service_uuid); `
   `if(pCharacteristic != NULL)  {    delete pCharacteristic;  } `
// Create a BLE Characteristic
   `pCharacteristic = pService->createCharacteristic(
                      new_char_uuid,
                      //BLECharacteristic::PROPERTY_READ   |
                      //BLECharacteristic::PROPERTY_WRITE  |
                      BLECharacteristic::PROPERTY_NOTIFY |
                      BLECharacteristic::PROPERTY_INDICATE
                    ); `
   ` if(bleDescript != NULL)  {    delete bleDescript;  } ` 
  ` bleDescript = new BLE2902();` 
  // Create a BLE Descriptor
  ` pCharacteristic->addDescriptor(bleDescript);` 

  // Start the service
  ` pService->start();` 
 `  log_e("BLE service started, start advertising");` 
  // Start advertising
  ` pAdvertising = BLEDevice::getAdvertising();` 
 `  pAdvertising->addServiceUUID(mDevice.service_uuid);` 
  ` pAdvertising->setScanResponse(false);` 
  ` pAdvertising->setMinPreferred(0x0);  // set value to 0x00 to not advertise this parameter` 
  ` BLEDevice::startAdvertising(); `

This code does not run to error and everything runs further but the service UUID did not change. The client still sees the same UUID.
What did I wrong?

Hardware:
Board: Olimex Gateway
Core Installation/update date: Espressif 32 1.9.0
IDE name: PlatformIo
Flash Frequency: 80Mhz
Upload Speed: 115200?

Description:
Extending Example BLE_server
The example BLE_server can be compiled and works. Client can connect. There is no issue

chegewara, 2020-06-03:
https://esp32.com/viewtopic.php?t=15691
This is idf explanation, but the idea is to set new serviceUUID and call start advertising.
The problem is you cant delete old serviceUUID. In that case you have to make use of this:
https://github.com/nkolban/esp32-snippets/blob/ac94b3467aa2b4f8a8dd41843e3b49e57b6c2543/cpp_utils/BLEAdvertising.cpp#L151-L178
and start advertising.

rob-bits, 2020-06-04:
Hi @chegewara ,

Thanks for the suggestion. Now I found a member function in the **BLEAdvertismentData** which maybe can be used:
**data.setCompleteServices(new_service_UUID);**

I set a new UUID with this method. The client sees the new UUID but during connection it did not found any services. In the client, inside the **MyAdvertisedDeviceCallbacks** calss **onResult()** callback the **advertisedDevice.isAdvertisingService(newUUID)** function founds the new UUID. But during the **connectToServer()** the following function returns with null pointer:
**BLERemoteService* pRemoteService = pClient->getService(*serviceUUID);**

And the null pointer leads to the message: "Failed to find our service UUID..."

In the server I did the following for changing the UUID:
```
void MyClass::changeUUID(newUUID* newUUID)
{
    BLEAdvertisementData data;
    BLEDevice::getAdvertising()->stop(); //stop advertising
    data.setFlags(0x6); //I do not now what this is means
    data.setCompleteServices(newUUID);
    //pAdvertising is a member data of my class
    pAdvertising->setAdvertisementData(data);
    pAdvertising->start();
}
```

Probably I did not set any characteristics and descriptor for my new UUID. How can I do that?
Can you share an example code with **setCompleteServices**() and setting up a service with characteristics?

Here is the Client log (after the server changed the UUID with the method what I wrote):

```
[E][device.hpp:65] onResult(): BLE Advertised Device found: Name: , Address: 30:ae:a4:36:4a:32, serviceUUID: 00000616-0000-0000-0003-5756fa3738ac
[E][device.hpp:79] onResult(): connecting...
Forming a connection to 30:ae:a4:36:4a:32
 - Created client
 - Connected to server
Failed to find our service UUID: 00000616-0000-0000-0003-5756fa3738ac
onDisconnect
[E][MasterDevice.cpp:54] scanDevicesTask(): We have failed to connect to the server A; there is nothin more we will do.
```

closed 2020-06-05 by rob-bits

ProductOfAmerica, 2024-02-05:
> Hi @chegewara ,
> 
> Thanks for the suggestion. Now I found a member function in the **BLEAdvertismentData** which maybe can be used: **data.setCompleteServices(new_service_UUID);**
> 
> I set a new UUID with this method. The client sees the new UUID but during connection it did not found any services. In the client, inside the **MyAdvertisedDeviceCallbacks** calss **onResult()** callback the **advertisedDevice.isAdvertisingService(newUUID)** function founds the new UUID. But during the **connectToServer()** the following function returns with null pointer: __BLERemoteService_ pRemoteService = pClient->getService(_serviceUUID);__
> 
> And the null pointer leads to the message: "Failed to find our service UUID..."
> 
> In the server I did the following for changing the UUID:
> 
> ```
> void MyClass::changeUUID(newUUID* newUUID)
> {
>     BLEAdvertisementData data;
>     BLEDevice::getAdvertising()->stop(); //stop advertising
>     data.setFlags(0x6); //I do not now what this is means
>     data.setCompleteServices(newUUID);
>     //pAdvertising is a member data of my class
>     pAdvertising->setAdvertisementData(data);
>     pAdvertising->start();
> }
> ```
> 
> Probably I did not set any characteristics and descriptor for my new UUID. How can I do that? Can you share an example code with **setCompleteServices**() and setting up a service with characteristics?
> 
> Here is the Client log (after the server changed the UUID with the method what I wrote):
> 
> ```
> [E][device.hpp:65] onResult(): BLE Advertised Device found: Name: , Address: 30:ae:a4:36:4a:32, serviceUUID: 00000616-0000-0000-0003-5756fa3738ac
> [E][device.hpp:79] onResult(): connecting...
> Forming a connection to 30:ae:a4:36:4a:32
>  - Created client
>  - Connected to server
> Failed to find our service UUID: 00000616-0000-0000-0003-5756fa3738ac
> onDisconnect
> [E][MasterDevice.cpp:54] scanDevicesTask(): We have failed to connect to the server A; there is nothin more we will do.
> ```
> 
> ## UPDATE:
> I finially could configure the esp and now the dynamic UUID works fine. You gave an idea, an I found this example: https://www.gitmemory.com/issue/espressif/arduino-esp32/2600/517083381 Thanks anyway!

@rob-bits 
What is the solution? Please post code, your link is broken.