espressif/arduino-esp32 #4059: No access to Wroom32 throw some time

smacyas, 2020-06-04:
Good afternoon! I use ESP 32.
Libraries:
#include <EEPROM.h>
#include <SPIFFS.h>
#include <MD5Builder.h>
#include <ESPAsyncWebServer.h>
##include <Arduino.h>
#include <AsyncMqttClient.h>
#include <Update.h>
#include <WiFiClient.h>

After some time of operation (sometimes after an hour, sometimes longer), access to the device via WiFi stops, while in debugger I control the status, WiFi. status() -> 1.

Debug verbose: [W][AsyncTCP.cpp:969] _poll(): rx timeout 4
[[W][Async TCP.cp:949] _poll(): pcb is NULL
Disconnected from MQTT: mqtt.eclipse.org:1883
.MQTT disconnect, restarting
.[.[D][Wi Fi Generic.cpp:337] _event Callback(): Event: 5 - STA_DISCONNECTED
.[.[W][Wi Fi Generic.cpp:353] _eventCallback(): Reason: 200 - BEACON_TIMEOUT

What could be the reason?

atanisoft, 2020-06-04:
BEACON_TIMEOUT can also indicate a poor signal quality.

smacyas, 2020-06-05:
> Almost certainly due to your reconnection code. You do not need to reconnect to the AP if the mqtt session fails, and whatever you are doing to reconnect is causing issues.

No, if the connection to the MQTT disappears - then only it is reconnected. If the Internet connection is lost, the MQTT Protocol is stopped, the wireless Internet access is reconnected and then connected to the MQTT Protocol.

mqttReconnectTimer = xTimerCreate("mqttTimer", pdMS_TO_TICKS(2000), pdFALSE, (void*)0, reinterpret_cast<TimerCallbackFunction_t>(connectToMqtt));
  wifiReconnectTimer = xTimerCreate("wifiTimer", pdMS_TO_TICKS(2000), pdFALSE, (void*)0, reinterpret_cast<TimerCallbackFunction_t>(WiFiConnect));
  xTimerStart(wifiReconnectTimer, 0);

void WiFiEvent(WiFiEvent_t event)
{
  Serial.printf("[WiFi/ETH] event: %d\r\n", event);
switch (event)
  {
case SYSTEM_EVENT_STA_CONNECTED:
      Serial.println(F("WiFi connected"));
      xTimerStop(wifiReconnectTimer, 0);
      break;
    case SYSTEM_EVENT_STA_GOT_IP:
      xTimerStop(wifiReconnectTimer, 0);
      eth_connected = false;
      wifi_connected = true;
      isr_timer0_blink = 0;
      wdt_wifiEth = 0;
      Serial.print(F("IP address: ")); Serial.println(WiFi.localIP());
      Serial.print(F("WiFi channel: ")); Serial.println(WiFi.channel());
      Serial.print(F("WiFi RSSI: ")); Serial.println(WiFi.RSSI());
      
      if (mqtt_enabled) MQTTStart();
      
      break;
    case SYSTEM_EVENT_STA_DISCONNECTED:
    case SYSTEM_EVENT_STA_LOST_IP:
      mapps_registered = false;
      wifi_connected = false;
      isr_timer0_blink = 0;
      Serial.println(F("WiFi lost connection"));
      if (mqtt_enabled) xTimerStop(mqttReconnectTimer, 0);
      if (!eth_connected) xTimerStart(wifiReconnectTimer, 0);
      break;
}}

smacyas, 2020-06-05:
> BEACON_TIMEOUT can also indicate a poor signal quality.

STA in 3 meters from router

atanisoft, 2020-06-05:
@smacyas proximity to the AP is irrelevant. Signal strength and quality are more important factors. You didn't say what board you are using or if you are using bare chips. There are some boards that behave poorly due to poor antenna design, improper copper keep-out zones, traces causing feedback to the antenna, etc.

smacyas, 2020-06-05:
> @smacyas proximity to the AP is irrelevant. Signal strength and quality are more important factors. You didn't say what board you are using or if you are using bare chips. There are some boards that [...]

WROOM-32 module, 4M

atanisoft, 2020-06-05:
In that case it sounds like you have a poorly designed/setup PCB. It can also be a missing solder connection for GND. Often times hand soldered chips miss soldering the pad on the backside, that is not an optional pad and must be connected to GND.

closed 2020-08-19 by stale[bot]