espressif/arduino-esp32 #4066: State of development of ESP32-S2

fabiuz7, 2020-06-06:
Hi, 
I was testing the esp32s2 branch, and of course, I have met some problems. I don't if it is my fault or there is a bug in the core (absolute understandable in this stage of development), but I would know if there is a roadmap or table in which is reported the actual state (something like what happen with esp-idf https://esp32.com/viewtopic.php?t=14532).

Thanks for your work.

lbernstone, 2020-06-06:
The next release will be based on 4.2, as that will support the release version of esp32-s2.  There will not be a release of arduino code before the [IDF code](https://github.com/espressif/esp-idf/releases) goes RTM.  Beyond that, YOU are assisting in the testing of beta code.  If you have a problem, report it in [gitter](https://gitter.im/espressif/arduino-esp32), but don't expect an immediate fix.  There are still major hardware units unsupported, esp I2C and CAN.

fabiuz7, 2020-06-06:
OK thank you! I will start to dig into this as soon as possible

closed 2020-06-06 by fabiuz7