espressif/arduino-esp32 #4067: How are gpio_config_t pin_bit_mask supposed to be created?

00coday, 2020-06-06:
I am trying to run gpio_config for multiple input pins with the ultimate goal of adding interrupts - the project is a race track timer using photodiodes as inputs for the lane ends.  Right now I'm using GPIOs  8 and 15.  I get inconsistent requirements with the gpio_config_t.pin_bit_mask - for pin 15, the mask looks like:

`io_conf.pin_bit_mask = 0B00000000000000001000000000000000;`

The "1" is in the 16th position from the right and makes sense if the first pin is pin 0 - everything works fine.  

Based on this I surmise that the pin to be affected is based on bit position in the bit mask.  makes sense.  From there I would think that the bit mask for pin 8 would look like:

`io_conf.pin_bit_mask = 0B00000000000000000000000100000000`

the "1" is in the 9th position - pin 8 if the starting pin is 0.  But that throws:
`ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Core 1 register dump:
PC      : 0x4000c271  PS      : 0x00060034  A0      : 0x80088c0f  A1      : 0x3ffb1e40  
A2      : 0x3ffb0058  A3      : 0x3f4046dc  A4      : 0x00000014  A5      : 0x3ffbe790  
A6      : 0x3ffbe7d8  A7      : 0x00000001  A8      : 0x00000001  A9      : 0x3f4046dd  
A10     : 0x000000a5  A11     : 0x00000000  A12     : 0x800897f7  A13     : 0x3ffbe760  
A14     : 0x00000008  A15     : 0x00000001  SAR     : 0x0000001f  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x4000c271:0x3ffb1e40 0x40088c0c:0x3ffb1e60 0x4008a620:0x3ffb1e80 0x4008a5d6:0x3ffb1ea0 0x400d1d6b:0x3ffb1f78

Core 0 register dump:
PC      : 0x400e9dc2  PS      : 0x00060334  A0      : 0x800d3e16  A1      : 0x3ffbbff0  
A2      : 0x00000000  A3      : 0x00000001  A4      : 0x00000000  A5      : 0x00000001  
A6      : 0x00060320  A7      : 0x00000000  A8      : 0x800d39de  A9      : 0x3ffbbfc0  
A10     : 0x00000000  A11     : 0x40084e58  A12     : 0x00060320  A13     : 0x3ffbb970  
A14     : 0x00000000  A15     : 0x3ffbbce0  SAR     : 0x00000000  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x400e9dc2:0x3ffbbff0 0x400d3e13:0x3ffbc010 0x400897e6:0x3ffbc030 0x400882f5:0x3ffbc050
`
I did some googling and found that for pin 8 the mask needed to be 0x8 or:

`io_conf.pin_bit_mask = 0B00000000000000000000000000001000;`

This works but is the binary representation of the number 8, not a bit map representing pin 8 when starting from 0.

All of the examples I have seen use some sort of bit shifting based on the input pin # - that is backed up by the bit map for 15, but not 8...  Is this an inconsistent behavior in how different pins are represented in the bit mask, or am I missing a fundamental ESP32 development concept here?

I'm using a NodeMCU ESP32S, Arduino IDE 1.8.5, and ESP32 library 1.0.4.  Test code looks like:

`#include <stdio.h>
#include <string.h>
#include <stdlib.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/queue.h"
#include "driver/gpio.h"

void IRAM_ATTR isr_lane(void* arg){
  uint32_t gpio_num = (uint32_t) arg;
  Serial.println( gpio_num );
}

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);

    uint8_t pin = 8;
    gpio_config_t io_conf;
    io_conf.intr_type = (gpio_int_type_t)GPIO_PIN_INTR_DISABLE;
    io_conf.pull_down_en = GPIO_PULLDOWN_DISABLE;
    io_conf.pull_up_en = GPIO_PULLUP_DISABLE;
    //io_conf.pin_bit_mask = 0B00000000000000001000000000000000;  //works for pin 15
    //io_conf.pin_bit_mask = 0B00000000000000000000000000001000;  //works for pin 8
    io_conf.pin_bit_mask = 0B00000000000000000000000100000000;  //should work for pin 8
    io_conf.mode = GPIO_MODE_INPUT;
    gpio_config(&io_conf);
    gpio_set_intr_type((gpio_num_t)pin, GPIO_INTR_HIGH_LEVEL);
    gpio_install_isr_service(0);
    gpio_isr_handler_add((gpio_num_t)pin, isr_lane, (void*) pin);
}

void loop() {
  // put your main code here, to run repeatedly:

}`

chegewara, 2020-06-06:
> the "1" is in the 9th position - pin 8 if the starting pin is 0. But that throws:

Thats correct, but crash is for some other reason, not related to pin mask:
```
Interrupt wdt timeout on CPU1
```

Why you dont use arduino code to attachInterrupt instead?
https://techtutorialsx.com/2017/09/30/esp32-arduino-external-interrupts/

00coday, 2020-06-06:
thanks for pointing that out - attachInterrupt works, but since I was looking to do the same thing with every input pin, I wanted to write the IRAM_ATTR function only once and pass it the parameter of the pin that has changed state.  attachInterrupt won't allow you to pass a parameter into the interrupt handler function.  I may be approaching the problem wrong - is there a way for the interrupt handler function to know what  pin is tripped?

chegewara, 2020-06-06:
In that case probably its better to use idf. Few things:
- dont use Serial.print in ISR function,
- why disable INTR on pin? `GPIO_PIN_INTR_DISABLE`
- why `GPIO_INTR_HIGH_LEVEL`, maybe better suitable is raising edge?

00coday, 2020-06-07:
Thanks for the further input.  The print was just to help see what was going on
Disable INTR - frankly it was in an example I copied.
I'll give raising edge a try.

I iterated through the pins and found that all but 6, 7, 8, 10, 11 work as expected, so I will use one of those that are working.  its still curious on the NodeMCU-32S that these pins are anomalies.

chegewara, 2020-06-07:
Pins 6-11 are used by flash.

00coday, 2020-06-08:
Thanks for the help - I appreciate it.

closed 2020-06-08 by 00coday