espressif/arduino-esp32 #4069: Multible http response headers with the same name get overwritten

yonie1, 2020-06-07:
If the http client gets multible headers with the same name in the response (for example set-cookie headers), they get overwritten. My way to fix it was to change 

https://github.com/espressif/arduino-esp32/blob/b92c58d74b151c7a3b56db4e78f2d3c90c16446f/libraries/HTTPClient/src/HTTPClient.cpp#L1149

to 

if(_currentHeaders[i].key.equalsIgnoreCase(headerName) && _currentHeaders[i].value.length() == 0) {

to check if the header value is not set to avoid overwriting when saving multible headers with the same name

lbernstone, 2020-06-07:
Make a pull request right at that page and mention @Jeroen88

Jeroen88, 2020-06-08:
I just took a quick review of the code. I am not in favor of this change:
- It does not fix the underlying problem: only one header with a specific name can exists, while a server may send more than one
- It only changes the behavior from keeping the last header sent with a certain name to keeping the first with a certain name

A real solution would involve quite some adaptations and very likely an API change, since [requestArgument ](https://github.com/espressif/arduino-esp32/blob/b92c58d74b151c7a3b56db4e78f2d3c90c16446f/libraries/HTTPClient/src/HTTPClient.h#L194) (not a very good name because in this context it is used as a responseArgument) needs to accommodate an array or vector of values. This implies changes in the functions for accessing the values as well.

kurenai-ryu, 2020-08-11:
why not implement like on the esp8266? 
https://github.com/esp8266/Arduino/blob/6feda9ebda8d5588c288145bff2f9f0d816b3ec2/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp#L1277

closed 2020-10-24 by stale[bot]