espressif/arduino-esp32 #4073: Incorrect use of INADDR_NONE in IPAddress.h

kaelef, 2020-06-08:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.8.5
IDE name:						Platform.io
Flash Frequency:					40Mhz
PSRAM enabled:                                    yes
Upload Speed:					115200
Computer OS:                                        Windows 10

### Description:
At the bottom of IPAddress.h there the following line: 
```const IPAddress INADDR_NONE(0, 0, 0, 0);```
This seems to be incorrect, as `INADDR_NONE` is only ever defined as `IPADDR_NONE` which is a UINT32 value.  `IPADDR_NONE(0, 0, 0, 0)` doesn't make sense.

This problem is worked-around near the top of WiFiProv.cpp where you see: 
```#undef IPADDR_NONE```
When this happens, the line in IPAddress.h becomes 
```const IPAddress (0, 0, 0, 0);```
which works because there is a constructor for IPAddress that takes four UINT32 values.

If you include <WiFiClient.h> in your sketch without including <Arduino.h> you'll see the problem come up during compile:

`\user\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\lwip/lwip/ip4_addr.h:79:37: error: expected ')' before numeric constant
 #define IPADDR_NONE         ((u32_t)0xffffffffUL)
                                     ^
\user\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\lwip/lwip/inet.h:71:29: note: in expansion of macro 'IPADDR_NONE'
 #define INADDR_NONE         IPADDR_NONE
                             ^
\user\.platformio\packages\framework-arduinoespressif32\cores\esp32/IPAddress.h:94:17: note: in expansion of macro 'INADDR_NONE'
 const IPAddress INADDR_NONE(0, 0, 0, 0);
 `

If this line in IPAddress were fixed, the #undef in WiFiProv.cpp won't be needed.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

#include <WiFiClient.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:

`\user\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\lwip/lwip/ip4_addr.h:79:37: error: expected ')' before numeric constant
 #define IPADDR_NONE         ((u32_t)0xffffffffUL)
                                     ^
\user\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\lwip/lwip/inet.h:71:29: note: in expansion of macro 'IPADDR_NONE'
 #define INADDR_NONE         IPADDR_NONE
                             ^
\user\.platformio\packages\framework-arduinoespressif32\cores\esp32/IPAddress.h:94:17: note: in expansion of macro 'INADDR_NONE'
 const IPAddress INADDR_NONE(0, 0, 0, 0);
 `

ajaybhargav, 2020-06-23:
I think IPAddress.h should be modified to
`const IPAddress IPADDR_NONE(0, 0, 0, 0);`

so it does not clash with inet.h

closed 2020-09-05 by stale[bot]

PaulMartinsen, 2022-11-02:
This problem still occurs and creates conflicts with code that uses inet.h. In ip4_addr.h we have:
#define IPADDR_NONE         ((u32_t)0xffffffffUL)
so this declaration blows up at compile time. 
extern IPAddress INADDR_NONE;

I suggest the breaking change to avoid conflicts with standard libraries names:
extern IPAddress IPAddress_None;

SuGlider, 2022-11-03:
This issue has been solved in IDF 4.4.2 with Arduino Core 2.0.4. 
The sketch used as example in this issue is now ok.

PaulMartinsen, 2022-11-05:
Thanks SuGlider. I'm using 2.0.5 of the Arduino core and see a related problem when I `#include "FS.h"` (which ultimately `#include "Arduino.h"` which leads to a problem similar to the one described here). I resolved by `#undef INADDR_NONE` before the `#include "FS.h"`. 

I'll post a separate issue when I get a chance, but for now the above might save someone else a bunch of time if they hit this problem too.