espressif/arduino-esp32 #4075: ESP32 works as a web server.

hyeriii, 2020-06-08:
ESP32 works as a web server. I want to edit the IP address of ESP32 on the web page.
Any examples or ideas?

lbernstone, 2020-06-08:
https://www.instructables.com/id/ESP8266-and-ESP32-With-WiFiManager/

hyeriii, 2020-06-10:
I am making an asynchronous web server.
  snprintf(messageBuf,MESSAGE_MAX_LEN,messageStatic,String(server.arg("ipv4static")),String(server.arg("gateway"))),String(server.arg("subnet")),String(server.arg( "ssidStatic")),String(server.arg("passkeyStatic")));
I want to use this part, but the'arg' part gives an error. I think it's part of the synchronous web server example. Do you have any idea how to fix the problem part with an asynchronous web server?