espressif/arduino-esp32 #4076: DOIT esp32 devkit v1 problem about softwareserial with another board

jimmyliiu0104, 2020-06-08:
## Hardware:
Board:							doit ESP32 Devkit v1 
Core Installation version:			
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                    ?no? ?yes?
Upload Speed:					115200
Computer OS:                                        Windows 10

### Description:
 I think the problem is simple. but i just cant figure it out : (
I use this board to transport 3 char(T,H,L) to another uno board by softwareserial. I want uno board  according to "char c", which recive from esp32 board, to do difference action.  At the esp32 side, I include esp's softwareserial .h and let pin(3,1)=rx, tx. As for uno side, I let pin(10, 11)=rx, tx. But when i sent any one of three char to uno board, the board always display all 3 actions! The function i use is while(!mySerial.available ()){}, under that is  if (mySerial.read()=='H') {(here is the action i want it to do when it recieve 'H')}, so does the other two char. 
Where am i wrong? this problem has confused me since last week.
My english is bad, sorry...

picture1 is a part of uno's code which i think may have a problem
[image]

picture2 is esp32's code which sent chars. cuz uno do recieve something ,so i think here is on problem. i use wifi.h to create a simple wifiserver and it works correctly.
[image]

lbernstone, 2020-06-08:
If you want only one character, use Serial.read(1).  Otherwise, it will retrieve the whole buffer and the result will be the number of characters retrieved.

jimmyliiu0104, 2020-06-09:
> If you want only one character, use Serial.read(1).

sorry, i can't understand. Like mySerial.read()=='H' in pic1, do i need to add 1 in ()? or declare serialread(1) at the top of the code first? or use only number not char for both board's code?

maxpromer, 2020-06-16:
You can use `Serial1` and `Serial2` , see more https://circuits4you.com/2018/12/31/esp32-hardware-serial2-example/

closed 2020-08-29 by stale[bot]