espressif/arduino-esp32 #4077: HTTPS error on ESP-Wroom-02

avsmohan, 2020-06-09:
### Hardware:
Board:							ESP-Wroom-02D
Core Installation/update date:			
IDE name:							Arduino IDE
Flash Frequency:					
PSRAM enabled:                                          
Upload Speed:						
Computer OS:                                          Windows 10

### Description:
We are facing some issues as we try to download firmware  using HTTPS  on ESP-Wroom-02D

When doing HTTPS,  OTA its returning an error as "connection refused". 
We are using Arduino IDE with function: ESPhttpUpdate.update(ota_link) for OTA.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//function for fota
WiFiClientSecure client;

if (!client.connect(host, 443)) 
{ 
Serial.println("connection failed"); 
return; 
} 
Serial.println("connected");

t_httpUpdate_return updateReturn = ESPhttpUpdate.update(FOTA_SERVER,8883,ftLnk,String(""),true,String(fingerprint),true);
//HTTPUpdateResult ESP8266HTTPUpdate::update(const String& host, uint16_t port, const String& uri, const String& currentVersion,
        bool https, const String& httpsFingerprint, bool reboot)

switch (updateReturn) 
    {
          case HTTP_UPDATE_FAILED:
            debug.printf("HTTP_UPDATE_FAILD Error (%d): %s", ESPhttpUpdate.getLastError(), ESPhttpUpdate.getLastErrorString().c_str());
            SendJsonPacket(FOTA_FAILED);
            return FOTA_FAIL;
            break;
    
          case HTTP_UPDATE_NO_UPDATES:
            debug.println(F("HTTP_UPDATE_NO_UPDATES"));
            SendJsonPacket(FOTA_FAILED);
            return NO_FOTA_FILE;
            break;
    
          case HTTP_UPDATE_OK:
            debug.println(F("HTTP_UPDATE_OK"));
            debug.printf("settings heap size: %u\n", ESP.getFreeHeap());
            SendJsonPacket(FOTA_SUCCESSFUL);
            return FOTA_SUCCESS;
            break;
    }

```

### Debug Messages:
```
Serial Debug:
connected  
HTTP_UPDATE_FAILD Error (-1): HTTP error: connection refused

```

Jeroen88, 2020-06-09:
Take a look at the httpUpdateSecure [example](https://github.com/espressif/arduino-esp32/blob/master/libraries/HTTPUpdate/examples/httpUpdateSecure/httpUpdateSecure.ino)

- I am not sure, but possibly a fingerprint can not be used, a certificate will work, but then you also need to set the time
- You should not make the connection, httpUpdate will do that
- The connection you make is over port 443, while you pass port 8883 to httpUpdate. The port should match where your server is hosting the firmware

atanisoft, 2020-06-09:
Esp-wroom-02 is an esp8266, this is not the correct place to report or seek assistance for that platform.

Jeroen88, 2020-06-09:
@atanisoft sharp!

@avsmohan example for esp8266 is [here](https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266httpUpdate/examples/httpUpdateSecure), fingerprint will work, use [this ](https://github.com/esp8266/Arduino/blob/5473656eac6a765dfea8dfa1befa72cccb4ef1fc/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.h#L87) function.