espressif/arduino-esp32 #4081: BluetoothSerial reporting "connected" even when connection fails

vanniaz, 2020-06-11:
I have been trying the SerialToSerialBTM example to read data from a barcode scanner, and I noticed that the message "Connected Successfully" appeared even when the scanner was off.
So I took a look at the waitForConnect function:
`static bool waitForConnect(int timeout) {
    TickType_t xTicksToWait = timeout / portTICK_PERIOD_MS;
    return (xEventGroupWaitBits(_spp_event_group, SPP_CONNECTED, pdFALSE, pdTRUE, xTicksToWait) != 0);
}
`
If I am not wrong, this function will return "true" even if other bits are set than SPP_CONNECTED (because according to the xEventGroupWaitBits description, it returns the value of the whole xEventGroup, not only the tested bits).
So in my opinion the function should be modified as follows:
`static bool waitForConnect(int timeout) {
    TickType_t xTicksToWait = timeout / portTICK_PERIOD_MS;
    return ((xEventGroupWaitBits(_spp_event_group, SPP_CONNECTED, pdFALSE, pdTRUE, xTicksToWait) & SPP_CONNECTED) != 0);
}
`
In fact, after having modified the code I see that it is working correctly.
The same change should be made to the "disconnect" and "isReady" functions.
Comments are welcome (I am not very expert in ESP-IDF so maybe that I am overlooking something).

lbernstone, 2020-06-11:
Go ahead and submit a PR.  You may get comments there, but your change looks appropriate to me.

vanniaz, 2020-06-11:
I see that a pull request for this bug has already been merged #3446, so I am going to close the issue.

closed 2020-06-11 by vanniaz