espressif/arduino-esp32 #4082: SD_MMC works but after a random time fails until I power it off

hepa94meca, 2020-06-11:
**Hardware**

- ESP32_Wroover B
- uSD 32GB
- LAN8720
- IDE Arduino
- Upload Speed  921600
- Flash 80Mhz

**Description**
I follow the instructions of the connection on the next link.. https://github.com/espressif/esp-idf/tree/master/examples/storage/sd_card#note-about-gpio12
 
- D2   - GPIO 12 (PULL_UP by code => pinMode(12, INPUT_PULLUP)
- D3   - GPIO 13 (10Kh pull-up)
- CMD -GPIO 15 (10Kh pull-up)
- VSS   -GND
- VDD  -3.3V
- CLK   -GPIO 14 (10Kh pull-up)
- VSS   -GND 
- D0    -GPIO2  (10Kh pull-up)
- D1    -GPIO4  (10Kh pull-up)

When I try just the SD_MMC it works great without problem, I tasted 7 days continuously, but when I add Ethernet (LAN8720) after 2 or 3 days the SD stops working, I have to restart the ESP by  powering off other ways such as restarting by code doesn’t initialize the SD and it fails from the beginning .

I try also with SPI_SD protocol and happens more less the same, after some days the SD fails.

**Steps to reproduce.**
Use SD_MMC with Ethernet Connection, append some data on a file and wait until it fails 

The LAN8720 connection is the follow...
https://sautter.com/blog/ethernet-on-esp32-using-lan8720/ 

**Debug Messages**
E (30975) sdmmc_cmd: sdmmc_read_sectors_dma: sdmmc_send_cmd returned 0x107
E (30975) diskio_sdmmc: sdmmc_read_blocks failed (263)
E (32012) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
E (32012) sdmmc_cmd: sdmmc_read_sectors_dma: sdmmc_send_cmd returned 0x107
E (32023) diskio_sdmmc: sdmmc_read_blocks failed (263)
E (33082) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
E (33082) sdmmc_cmd: sdmmc_read_sectors_dma: sdmmc_send_cmd returned 0x107
E (33093) diskio_sdmmc: sdmmc_read_blocks failed (263)
[E][vfs_api.cpp:265] VFSFileImpl(): fopen(/sdcard/dbin.csv) failed

**Code**
`
void setup(void)
{
	Serial.begin(9600);
	

    pinMode(12, INPUT_PULLUP);
    if(!SD_MMC.begin()){
        Serial.println("Card Mount Failed");
        return;
    }

  WiFi.onEvent(WiFiEvent);
  ETH.begin(ETH_ADDR, ETH_POWER_PIN, ETH_MDC_PIN, ETH_MDIO_PIN, ETH_TYPE, ETH_CLK_MODE);
	InitServer();
	InitWebSockets();
}
long contador=0;
long previousMillisSD=0;
#define Minuto 60000
void loop(void)
{

  unsigned long currentMillis = millis(); 
if (currentMillis - previousMillisSD >=Minuto*10) {
   previousMillisSD = currentMillis;
   appendFile(SD_MMC, "/hello.txt", "World!\r\n");
  }
    
    
}

void appendFile(fs::FS &fs, const char * path, const char * message){
    Serial.printf("Appending to file: %s\n", path);

    File file = fs.open(path, FILE_APPEND);
    if(!file){
        Serial.println("Failed to open file for appending");
        return;
    }
    if(file.print(message)){
        Serial.println("Message appended");
    } else {
        Serial.println("Append failed");
    }
}
`

cgmckeever, 2020-07-29:
Did you ever have a resolution of this -- seeing a similar issue

closed 2020-10-11 by stale[bot]

hall1235, 2021-01-05:
I got the same issue, how can you fix ?

hepa94meca, 2021-01-12:
The problem was solved, it was a power supply issue, my current was not enough.

hall1235, 2021-01-12:
Thank you so much **hepa94meca**