espressif/arduino-esp32 #4087: ESP32 and Temperature Sensor MCP9808 in I2C communication

HotToCold, 2020-06-13:
Hello every body!
I use this web site for the first time...
This is a Program Arduino dedicated to the Temperature Sensor MCP9808 connected to the ESP32 Device.
I hope it will help you.

```
#include <Wire.h>

//GPIO-33 dedicated to I2C SDA and GPIO-32 dedicated to I2C SCL, for example
#define I2C_SDA 33
#define I2C_SCL 32

/******************************* TEMPERATURE SENSOR MCP9808 *******************************/
const uint8_t MCP9808_ADDR 				= 0x18; //MCP9808 Sensor Address by default

const uint8_t MCP9808_CFG_RESOLUTION0 	= 0x00; //Resolution Configuration to 0.5°C 	| Tconv= 30ms 	| 33 Samples/s
const uint8_t MCP9808_CFG_RESOLUTION1 	= 0x01; //Resolution Configuration to 0.25°C 	| Tconv= 65ms 	| 15 Samples/s
const uint8_t MCP9808_CFG_RESOLUTION2 	= 0x02; //Resolution Configuration to 0.125°C 	| Tconv= 130ms 	| 7 Samples/s
const uint8_t MCP9808_CFG_RESOLUTION3 	= 0x03; //Resolution Configuration to 0.0625°C 	| Power-up Default, Tconv= 250ms | 4 Samples/s

const uint8_t MCP9808_REG_CONFIG 		= 0x01; //Register of Configuration
const uint8_t MCP9808_REG_UPPER_TEMP 	= 0x02; //Register of Threshold Limit Temperature Upper (TUPPER)
const uint8_t MCP9808_REG_LOWER_TEMP 	= 0x03; //Register of Threshold Limit Temperature Lower (TLOWER)
const uint8_t MCP9808_REG_CRIT_TEMP 	= 0x04; //Register of Threshold Limit Temperature Critical (TCRIT)
const uint8_t MCP9808_REG_AMBIENT_TEMP 	= 0x05; //Register of Ambiant Temperature (TA)
const uint8_t MCP9808_REG_MANUF_ID 		= 0x06; //Register of Manufacturer ID
const uint8_t MCP9808_REG_DEVICE_ID 	= 0x07; //Register of Device ID/Revision
const uint8_t MCP9808_REG_RESOLUTION 	= 0x08; //Register of Resolution

const uint16_t MCP9808_CFG_THYST0 		= 0x0000; //Hysteresis Limit (Tupper and Tlower) 	| Bits 10-9 - 00: 0°C (power-up default)
const uint16_t MCP9808_CFG_THYST1 		= 0x0200; //Hysteresis Limit (Tupper and Tlower) 	| Bits 10-9 - 01: +1.5°C
const uint16_t MCP9808_CFG_THYST2 		= 0x0400; //Hysteresis Limit (Tupper and Tlower) 	| Bits 10-9 - 10: +3.0°C
const uint16_t MCP9808_CFG_THYST3 		= 0x0600; //Hysteresis Limit (Tupper and Tlower) 	| Bits 10-9 - 11: +6.0°C
const uint16_t MCP9808_CFG_SD 			= 0x0100; //Shutdown Mode bit						| Bit 8
const uint16_t MCP9808_CFG_CRITLOCK 	= 0x0080; //Critical Trip bit						| Bit 7
const uint16_t MCP9808_CFG_WINLOCK 	= 0x0040; //Alarm window lock						| Bit 6
const uint16_t MCP9808_CFG_INTCLR 		= 0x0020; //Interrupt clear							| Bit 5
const uint16_t MCP9808_CFG_ALERTSTAT 	= 0x0010; //Alert output status						| Bit 4
const uint16_t MCP9808_CFG_ALERTCTRL 	= 0x0008; //Alert output control					| Bit 3
const uint16_t MCP9808_CFG_ALERTSEL 	= 0x0004; //Alert output select						| Bit 2
const uint16_t MCP9808_CFG_ALERTPOL 	= 0x0002; //Alert output polarity					| Bit 1
const uint16_t MCP9808_CFG_ALERTMODE 	= 0x0001; //Alert output mode						| Bit 0

const uint16_t MCP9808_STAT_SIGN 		= 0x1000; //Measurement Sign
const uint16_t MCP9808_STAT_TLOWER 		= 0x2000; //Lower Trigger Status
const uint16_t MCP9808_STAT_TUPPER 		= 0x4000; //Upper Trigger Status
const uint16_t MCP9808_STAT_TCRIT 		= 0x8000; //Critical Trigger Status
/* FIN ************************** TEMPERATURE TRANSMITTER MCP9808 *******************************/

void setup() {
	if(1){
		Serial.begin(115200);
		while (!Serial); //Waiting until establishment of serial communication
		Serial.println();
		Serial.println("Etablissement de la communication série USB");
		Serial.println();
	}

	if(1){ /* TEMPERATURE SENSOR MCP9808 **********************/
		uint16_t CheckCfg;
		Serial.println();
		Serial.println();
		Serial.println("Tentative de Communication I2C avec le capteur de température MCP9808");
		//Initialisation de la liaison I2C
		Wire.begin(I2C_SDA, I2C_SCL);
		//The MCP9808 has an internal Power-On Reset (POR) circuit.
		//If the power supply voltage Vdd glitches below the VPOR (2.2Vdd) threshold, the device resets the registers to the power-on default settings.
		//Register : MCP9808_REG_CONFIG			| Configuration Value: 0x0000
		//Register : MCP9808_REG_UPPER_TEMP		| Value: 0x0000 (0°C)
		//Register : MCP9808_REG_LOWER_TEMP		| Value: 0x0000 (0°C)
[... 298 lines cut ...]
	uint8_t Status;
	Wire.beginTransmission(Addr);
	Wire.write(Reg);
	Status = Wire.endTransmission();
	
	if (Status == 0) {
		Wire.requestFrom(Addr, 2);
		Value = Wire.read();
		Value <<= 8;
		Value |= Wire.read();
	}
	return Value;
}

// Sensor MCP9808 I2C: Writing of Register (16 bits)
void Sensor_WriteReg16bits(uint8_t Addr, uint8_t Reg, uint16_t Param) {
	Wire.beginTransmission(Addr);
	Wire.write(Reg);
	Wire.write(Param >> 8);
	Wire.write(Param & 0xFF);
	Wire.endTransmission(); 
}
/* END ****************** I2C COMMUNICATION **************************************/

```

lbernstone, 2020-06-13:
This is not instructables.com  .  Please don't abuse the site.

HotToCold, 2020-06-13:
Could you explain to me what happens?
Because, i just send a program for arduino...
Otherwhise, no problem i will unsubscribe.

WereCatf, 2020-06-17:
@HotToCold This is for reporting issues with the Arduino-core for the ESP32, not for posting your personal projects.

closed 2020-08-30 by stale[bot]