espressif/arduino-esp32 #4091: handle website: Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

dsyleixa, 2020-06-14:
```
Core Installation version:  1.0.4
IDE name: Arduino 1.8.9
Flash Frequency: 80MHz
PSRAM enabled: ?no? ?yes?
Upload Speed: 921600
Computer OS: Windows 7/64 pro
Browser: Firefox 70.x.x
Adafruit Feather ESP32
```

setup() code for wifiserver and webserver works fine on ESP8266 and appearently also is fine for ESP32, but then it crashes when creating the website.
What is wrong with the website code? 
What is it not esp8266-compatible but crashes?

```
#include <WiFi.h>
#include <WebServer.h>
IPAddress   this_ip(192, 168, 2, 201); //  
IPAddress   gateway(192, 168, 2, 1);       // <<< LAN Gateway IP
IPAddress   subnet(255, 255, 255, 0);      // <<< LAN Subnet Mask
WiFiServer  wifiserver(http_port);
WebServer   webserver(8081);
//...

void setup() {
//----------------------------------------
   // Connect to WiFi network
   Serial.println();
   Serial.println();
   Serial.println("Connecting to WiFi: ");
   Serial.println( WiFi.gatewayIP().toString() );
   
   WiFi.mode(WIFI_STA);
   WiFi.config(this_ip, gateway, subnet, gateway, gateway);   // dns = gateway
   WiFi.begin(ssid, password);

   while (WiFi.status() != WL_CONNECTED) {
      delay(500);
      Serial.print(".");      
   }   
   
   Serial.println("");
   Serial.print("WiFi connected: ");
   Serial.println(WiFi.gatewayIP());

   //----------------------------------------
   wifiserver.begin();
   Serial.println("WiFi Server started");

   //----------------------------------------
   webserver.on("/", handleRoot) ;
   webserver.on("/client/client0/", handleClients);
   delay(10);   
   webserver.begin();
   Serial.println("ESP webserver started");

   // Print the IP address
   Serial.print("Use this URL to connect: ");
   Serial.print("http://");    Serial.print(WiFi.localIP());
   Serial.print(":");    Serial.print(http_port);
   Serial.println("/");
   Serial.print((String)"myweb.sytes.net" + ":" + http_port + "/");
}

void loop() {
   Serial.println("\nDEBUG:\n new loop");

   static double ftmp;
   static unsigned long tsec = millis(), tms = millis();
      
   Serial.println("\nDEBUG:\n handleWebsite()");
   handleWebsite();                               // <<<<<<<<<<<<<<<<<<<<<<<<<< CRASH !!

   Serial.println("\nDEBUG:\n webserver.handleClient()");  // <<<<<<<<<< not executed any more
   webserver.handleClient();

   //---------------------------------------
   // Read local  data   
   ftmp = fINVAL;
   ftmp = bmp_x77.readTemperature();
   //
}

void handleWebsite() {

   WiFiClient client = wifiserver.available();

   String request = client.readStringUntil('\r');
   Serial.println(request);
   client.flush();
   String script = "";

   script += ("HTTP/1.1 200 OK \n");
   script += ("Content-Type: text/html \n");
   script += ("\n");  
   script += ("<!DOCTYPE html> \n");
   script += ("<html> \n");
   // head + title
   script += ("<head> \n");
   // autom. Aktualisierung alle 20 sec.
   script += "<meta http-equiv=\"refresh\" content=\"20; URL=";
   script += (String)website_url + ":" + (String)http_port + "\"> \n" ;

   // utf-8 für "°" Zeichen
   script += "<meta http-equiv=\"Content-Type\" content=\"text/html; charset=utf-8\"> \n" ;
   script += ("<title>");
   script += "myweb.sytes.net";
   script += ("</title> \n");
   script += ("</head> \n");
   // body + caption
   script += ("<body> \n");
   script += ("<h1> <p> ");      
   script += ver + " " + WiFi.localIP().toString() + " " + (String)ssid + " <br>" ;
   //script += "</font> </p> \n";
   script += "</font>   \n";
   script += "</body> \n";
   script += "</html> \n";
   script += ("<br> \n");
   script += ("<br> \n");

   client.print(script);
   script = "";

   delay(1);
}

```

```
Connecting to WiFi: 
0.0.0.0
.
WiFi connected: 192.168.2.1
WiFi Server started
ESP webserver started
Use this URL to connect: http://192.168.2.201:8080/
myweb.sytes.net:8080/setup done 

DEBUG:
 new loop

DEBUG:
 handleWebsite()
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d6ab4  PS      : 0x00060630  A0      : 0x800d6b88  A1      : 0x3ffb1e10  
A2      : 0x00000000  A3      : 0x3ffb1e5f  A4      : 0x00000001  A5      : 0x00000001  
A6      : 0x00060220  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x00000000  
A10     : 0x00000000  A11     : 0x00000080  A12     : 0x00000050  A13     : 0x00000001  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000008  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffff9  

Backtrace: 0x400d6ab4:0x3ffb1e10 0x400d6b85:0x3ffb1e30 0x4014ae16:0x3ffb1e50 0x400dbfb1:0x3ffb1e80 0x400dc049:0x3ffb1ea0 0x400d4491:0x3ffb1ec0 0x400d4827:0x3ffb1f80 0x400deac5:0x3ffb1fb0 0x400896d9:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

```

atanisoft, 2020-06-14:
Client may be null in loop is the only thing I can see. You will need to decide the backtrace to know for certain where the null dereference is happening.

dsyleixa, 2020-06-15:
tbh. I do not understand.
The code is working for esp8266 since many years and is also Arduino Wifi shield compatible, so what shall I do exactly?

me-no-dev, 2020-06-15:
repeating how it works on 8266 will not help you much. decoding the issue with the ExceptionDecoder might actually help. Link and instructions were in the issue template that you edited. Also can be found in the readme. And since you so much "works on 8266 for ages" 8266 uses the same exception decoder.

dsyleixa, 2020-06-15:
I don't know how to do that encoding with the ExceptionDecoder.
OTOH, for  esp8266 there is no issue, there the code runs like a charm, so I never needed the ExceptionDecoder so far.

me-no-dev, 2020-06-15:
then maybe good time to learn to use it if you want help.

dsyleixa, 2020-06-15:
I do not understand that - any help highly appreciated!

me-no-dev, 2020-06-15:
ok, let me try to say it more clear.

We can not give you any help unless you decode your backtraces. There are tutorials, youtube videos and more on how to install and use the exception decoder. Also in your code you did not show `handleRoot()`.

dsyleixa, 2020-06-15:
yes, a c+p error.
I meanwhile updated the code which now compiles fine.

dsyleixa, 2020-06-15:
PS, 
after installing the decoder I get no elf trace or whatever

me-no-dev, 2020-06-15:
you need to paste the backtrace into the window... and the sketch to be compiled and running. Did you look at any video?

dsyleixa, 2020-06-15:
I don't get a window, just a file to select (which is not existent).

Can please test this thing anyone who is experienced?

me-no-dev, 2020-06-15:
You open the sketch in Arduino IDE, you compile and upload the sketch to the board, then open the serial monitor so you can receive the error. Now open the exception decoder, it will not ask you for file, paste the error in the top half and you will get the result.

dsyleixa, 2020-06-15:
I did all that, I pasted the code, but I but I don't get a result.

usane1, 2020-06-15:
In your first post, there is a debug part. Guru Meditation Error....
The end is rebooting and date. Before that ther is a line with **Backtrace: .....**
That line have to e copied to the expectiondecoder  including the "Backtrace:" , while your board connected and running the code that cause the error.

usane1, 2020-06-15:
Btw, probably it is the HTTPClient related issue that cause a lot of problem in 1.0.3 and 1.0.4 core.
Read #4039 and #3347 .

dsyleixa, 2020-06-15:
yes, I copied the line Backtrace, but got no result.
I also tried 1.0.2, same crash.
What is the solution to my problem?

chegewara, 2020-06-15:
> What is the solution to the Guru crash problem?

Find and fix issue.

atanisoft, 2020-06-15:
@dsyleixa you must compile your sketch in the same ardunio ide as you are using the backtrace decoder. Copy/paste the trace from the serial monitor in the ide into the trace window and it should produce a meaningful decode output. The decoder has a required dependency on the binary produced by the ide in order to decode the trace.

dsyleixa, 2020-06-15:
yes, I did that, but no result.

> @dsyleixa you must compile your sketch in the same ardunio ide as you are using the backtrace decoder. Copy/paste the trace from the serial monitor in the ide into the trace window and it should [...]

dsyleixa, 2020-06-15:
Can please test my code  anyone who is experienced?

usane1, 2020-06-15:
Then you probably not installed correctly the decoder. It gives me result if i put your backtrace to my decoder but thats not gives the good result because i havent got your code, and i have different operating system, file system etc...

atanisoft, 2020-06-15:
Then I don't think we can help narrow down the cause for the failure if the decoding is not happening. From a very quick read of the code it doesn't look like there is any error checking being done in handleWebsite() which would include checking that wifiserver.available() returned a valid client object (if none are available it can return nullptr).

Also assigning script at the end of that method is not necessary. The variable will go out of scope and be cleaned up automatically upon exit from the function. You could also improve the function a bit by directly calling client.print() directly

chegewara, 2020-06-15:
You may dislike my answer, but its only correct.
There is tens of tens situations that can cause Guru crash. My bet is you are trying to access not initialized variable.

dsyleixa, 2020-06-15:
chegewara, if you cannot provide a reasonable fix, please troll somewhere else.

atanisoft, 2020-06-15:
@dsyleixa he isn't trolling. He is staying fact. There are many ways to encounter such crashes and often times they are due to simple mistakes

dsyleixa, 2020-06-15:
I do not undrstand what to do now, I am just an Arduino hobby-user.
Can you please try my code and see if you can find the reason?

> Then I don't think we can help narrow down the cause for the failure if the decoding is not happening. From a very quick read of the code it doesn't look like there is any error checking being done [...]

chegewara, 2020-06-15:
> I do not understand what to do now, I am just an Arduino hobby-user.
> Can you please try my code and see if you can find the reason?

No, because you will never learn to use backtrace decoder and will get back here with every single crash.

chegewara, 2020-06-15:
@me-no-dev Please close this issue

dsyleixa, 2020-06-15:
@me-no-dev Please block chegewara.

dsyleixa, 2020-06-15:
The code is esp8266 and Arduino Wifi shield compatible and I never had crashes there.

chegewara, 2020-06-15:
http://www.iotsharing.com/2017/12/how-to-debugging-errors-using-arduino-esp-exception-decoder.html

chegewara, 2020-06-15:
https://www.youtube.com/watch?v=323CS87h6WU

atanisoft, 2020-06-15:
@dsyleixa esp32 is not the same as esp8266 and it never will be.

You can try adjusting your handleWebsite() to look like this:
```
void handleWebsite() {
   WiFiClient client = wifiserver.available();
   if (!client) {
      return;
   } 

   String request = client.readStringUntil('\r');
```
This checks that the client object that was returned from the server listener is valid (ie: connected to something) and if not it gets out of the method before trying to use it.

As for the other suggestion you can do something like this directly and save yourself a lot of memory on the stack:
```
   client.println("HTTP/1.1 200 OK");
   client.println("Content-Type: text/html");
   client.println(""); // blank newline after header data
```

usane1, 2020-06-15:
We can't test your code because github has not private messaging. If you can send your code an other way to me, i give it a test, but today i only have about 30 min then i go.

dsyleixa, 2020-06-15:
@usane1:
the code is completely what I posted in the top, nothing else yet

dsyleixa, 2020-06-15:
I applied your changes, now no crashes but also still no website generated:
```

#include <WiFi.h>
#include <WebServer.h>
IPAddress   this_ip(192, 168, 2, 201); //  
IPAddress   gateway(192, 168, 2, 1);       // <<< LAN Gateway IP
IPAddress   subnet(255, 255, 255, 0);      // <<< LAN Subnet Mask
WiFiServer  wifiserver(8080);
WebServer   webserver(8081);
const char* ssid="WLAN-3YA7LD";              // WIFI network name
const char* password="18658446694458594657";          // WIFI network password
//...

void setup() {
//----------------------------------------
   Serial.begin(115200);
   delay(1000);
   // Connect to WiFi network
   Serial.println();
   Serial.println();
   Serial.println("Connecting to WiFi: ");
   Serial.println( WiFi.gatewayIP().toString() );
   
   WiFi.mode(WIFI_STA);
   WiFi.config(this_ip, gateway, subnet, gateway, gateway);   // dns = gateway
   WiFi.begin(ssid, password);

   while (WiFi.status() != WL_CONNECTED) {
      delay(500);
      Serial.print(".");      
   }   
   
   Serial.println("");
   Serial.print("WiFi connected: ");
   Serial.println(WiFi.gatewayIP());

   //----------------------------------------
   wifiserver.begin();
   Serial.println("WiFi Server started");

   //----------------------------------------
   webserver.on("/", handleRoot) ;
   webserver.on("/client/client0/", handleClients);
   delay(10);   
   webserver.begin();
   Serial.println("ESP webserver started");

   // Print the IP address
   Serial.print("Use this URL to connect: ");
   Serial.print("http://");    Serial.print(WiFi.localIP());
   Serial.print(":");    Serial.print(8080);
   Serial.println("/");
   Serial.println((String)"myweb.sytes.net" + ":" + (String)(8080) + "/");
   Serial.println("setup done ");
}

void loop() {
   Serial.println("\nDEBUG:\n new loop");

   static double ftmp;
   static unsigned long tsec = millis(), tms = millis();
      
   Serial.println("\nDEBUG:\n handleWebsite()");
   handleWebsite();                               // <<<<<<<<<<<<<<<<<<<<<<<<<< CRASH !!

   Serial.println("\nDEBUG:\n webserver.handleClient()");  // <<<<<<<<<< not executed any more
   webserver.handleClient();

   //---------------------------------------
   // Read local  data   
   ftmp = -99;
   //ftmp = bmp_x77.readTemperature();
   //
}

void handleWebsite() {

   WiFiClient client = wifiserver.available();
   if (!client) {
      return;
   } 

   String request = client.readStringUntil('\r');
   Serial.println(request);
   client.flush();
   String script = "";

   script += ("HTTP/1.1 200 OK \n");
   script += ("Content-Type: text/html \n");
   script += ("");  
   script += ("<!DOCTYPE html> \n");
   script += ("<html> \n");
   // head + title
   script += ("<head> \n");
   // autom. Aktualisierung alle 20 sec.
   script += "<meta http-equiv=\"refresh\" content=\"20; URL=";
   script += (String)"myweb.sytes.net" + ":" + (String)(8080) + "\"> \n" ;

   // utf-8 für "°" Zeichen
   script += "<meta http-equiv=\"Content-Type\" content=\"text/html; charset=utf-8\"> \n" ;
   script += ("<title>");
   script += "myweb.sytes.net";
   script += ("</title> \n");
   script += ("</head> \n");
   // body + caption
   script += ("<body> \n");
   script += ("<h1> <p> ");      
   script += (String)"test" + " " + WiFi.localIP().toString() + " " + (String)ssid + " <br>" ;
   script += "</p> </h1>  \n"; 
   script += "</body> \n";
   script += "</html> \n";
   script += ("<br> \n");
   script += ("<br> \n");

   client.print(script);
   script = "";

   delay(1);
}

void handleRoot() {;}

void handleClients() {;}

```
> @dsyleixa esp32 is not the same as esp8266 and it never will be.
>
> You can try adjusting your handleWebsite() to look like this:
>
> ```
> void handleWebsite() {
>    WiFiClient client = [...]

usane1, 2020-06-15:
Here is what i got from decoding:
```
PC: 0x400d18dc: WiFiClientRxBuffer::read(unsigned char*, unsigned int) at C:\Users\Office-1\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi\src\WiFiClient.cpp line 107
EXCVADDR: 0x00000008

Decoding stack results
0x400d18dc: WiFiClientRxBuffer::read(unsigned char*, unsigned int) at C:\Users\Office-1\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi\src\WiFiClient.cpp line 107
0x400d19ad: WiFiClient::read(unsigned char*, unsigned int) at C:\Users\Office-1\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi\src\WiFiClient.cpp line 434
0x4014245a: WiFiClient::read() at C:\Users\Office-1\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi\src\WiFiClient.cpp line 345
0x400d67ad: Stream::timedRead() at C:\Users\Office-1\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\Stream.cpp line 36
0x400d6845: Stream::readStringUntil(char) at C:\Users\Office-1\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\Stream.cpp line 287
0x400d12d1: handleWebsite() at C:\Users\Office-1\Desktop\sketch_jun15a/sketch_jun15a.ino line 79
0x400d165f: loop() at C:\Users\Office-1\Desktop\sketch_jun15a/sketch_jun15a.ino line 62
0x400d74f9: loopTask(void*) at C:\Users\Office-1\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\main.cpp line 19
0x40088b7d: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

```

dsyleixa, 2020-06-15:
and what does that mean now?

atanisoft, 2020-06-15:
> I applied your changes, now no crashes but also still no website generated:

well, it looks like it hasn't accepted a connection. You may need to turn up the core debug log level to see what might be going on in the underlying layers. Set your board type to the ESP32 Dev board (should be one of the first ones in the list) and you will find a core debug log level menu option, set it to DEBUG and reflash the sketch.

@usane1 has decoded the backtrace and it points to using a null client object which you have a fix for now.

dsyleixa, 2020-06-15:
what is a null client object and how to use that?

chegewara, 2020-06-15:
> @usane1 has decoded the backtrace and it points to using a null client object which you have a fix for now.

which i pointed here:

> My bet is you are trying to access not initialized variable.

dsyleixa, 2020-06-15:
chegewara, stop your superior attitude and move along!

dsyleixa, 2020-06-15:
qatanisoft:
what is a null client object and how to use that?

atanisoft, 2020-06-15:
> what is a null client object and how to use that?

The listener did not have a connected client so it defaulted to return a null value. It is always a good idea to validate the return values from APIs to ensure they did not report an error condition (which this could be taken as).

> chegewara, stop your superior attitude and move along!

We are all trying to help you and continued attacks will result in no further help. Please reign it in a bit.

dsyleixa, 2020-06-15:
> The listener did not have a connected client so it defaulted to return a null value. It is always a good idea to validate the return values from APIs to ensure they did not report an error condition [...]

I don't understand a word, I never needed that for Arduino Wifi oder esp8266.
what is the correct code for that?

atanisoft, 2020-06-15:
> what is the correct code for that?

https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/SimpleWiFiServer/SimpleWiFiServer.ino

dsyleixa, 2020-06-15:
I still don't understand.
What is the null client code there?

chegewara, 2020-06-15:
> We are all trying to help you and continued attacks will result in no further help. Please reign it in a bit.

He is right, im not trying to help him to solve that issue. All he is is doing is wining to test his code and this is not his first issue like that:
https://github.com/espressif/arduino-esp32/issues/3444
https://github.com/espressif/arduino-esp32/issues?q=is%3Aissue+author%3Adsyleixa+is%3Aclosed

If someone does not want to learn then does not deserve help and attention.

dsyleixa, 2020-06-15:
chegewara, troll somewhere else!

dsyleixa, 2020-06-15:
I highly appreciate help from people who want to help though.

usane1, 2020-06-15:
Chekt the line in the example code:
if(client){
....
}
That means if it doesnt run to dis loop then nothing connected to the ESP, that means nullclient.

chegewara, 2020-06-15:
> I highly appreciate help from people who want to help though.

i doubt
https://github.com/espressif/arduino-esp32/issues/3444#issuecomment-551930831

dsyleixa, 2020-06-15:
ok, so I just have to wrap 
if(client){
....
}
around?
I'll try immediately!

usane1, 2020-06-15:
That only solve the Meditation error, but doesnt't solve why the client cannot connect to the ESP, that what is need to figure out.

usane1, 2020-06-15:
Now i have to go. Hope you will have a solution.

dsyleixa, 2020-06-15:
yes, same as before:
the code runs through, no Guru errror, but also no website created!
```

#include <WiFi.h>
#include <WebServer.h>
IPAddress   this_ip(192, 168, 2, 201); //
IPAddress   gateway(192, 168, 2, 1);       // <<< LAN Gateway IP
IPAddress   subnet(255, 255, 255, 0);      // <<< LAN Subnet Mask
WiFiServer  wifiserver(8080);
WebServer   webserver(8081);
const char* ssid="WLAN-3YA7LD";              // WIFI network name
const char* password="18658446694458594657";          // WIFI network password
//...

void setup() {
//----------------------------------------
   Serial.begin(115200);
   delay(1000);
   // Connect to WiFi network
   Serial.println();
   Serial.println();
   Serial.println("Connecting to WiFi: ");
   Serial.println( WiFi.gatewayIP().toString() );

   WiFi.mode(WIFI_STA);
   WiFi.config(this_ip, gateway, subnet, gateway, gateway);   // dns = gateway
   WiFi.begin(ssid, password);

   while (WiFi.status() != WL_CONNECTED) {
      delay(500);
      Serial.print(".");
   }

   Serial.println("");
   Serial.print("WiFi connected: ");
   Serial.println(WiFi.gatewayIP());

   //----------------------------------------
   wifiserver.begin();
   Serial.println("WiFi Server started");

   //----------------------------------------
   webserver.on("/", handleRoot) ;
   webserver.on("/client/client0/", handleClients);
   delay(10);
   webserver.begin();
   Serial.println("ESP webserver started");

   // Print the IP address
   Serial.print("Use this URL to connect: ");
   Serial.print("http://");    Serial.print(WiFi.localIP());
   Serial.print(":");    Serial.print(8080);
   Serial.println("/");
   Serial.println((String)"myweb.sytes.net" + ":" + (String)(8080) + "/");
   Serial.println("setup done ");
}

void loop() {
   Serial.println("\nDEBUG:\n new loop");

   static double ftmp;
   static unsigned long tsec = millis(), tms = millis();

   Serial.println("\nDEBUG:\n handleWebsite()");
   handleWebsite();                               // <<<<<<<<<<<<<<<<<<<<<<<<<< CRASH !!

   Serial.println("\nDEBUG:\n webserver.handleClient()");  // <<<<<<<<<< not executed any more
   webserver.handleClient();

   //---------------------------------------
   // Read local  data
   ftmp = -99;
   //ftmp = bmp_x77.readTemperature();
   //
}

void handleWebsite() {

   WiFiClient client = wifiserver.available();
   if (!client) {
      return;
   }
   if(client) {

      String request = client.readStringUntil('\r');
      Serial.println(request);
      client.flush();
      String script = "\n";

      script += ("HTTP/1.1 200 OK \n");
      script += ("Content-Type: text/html \n");
      script += ("");
      script += ("<!DOCTYPE html> \n");
      script += ("<html> \n");
      // head + title
      script += ("<head> \n");
      // autom. Aktualisierung alle 20 sec.
      script += "<meta http-equiv=\"refresh\" content=\"20; URL=";
      script += (String)"myweb.sytes.net" + ":" + (String)(8080) + "\"> \n" ;

      // utf-8 für "°" Zeichen
      script += "<meta http-equiv=\"Content-Type\" content=\"text/html; charset=utf-8\"> \n" ;
      script += ("<title>");
      script += "myweb.sytes.net";
      script += ("</title> \n");
      script += ("</head> \n");
      // body + caption
      script += ("<body> \n");
      script += ("<h1> <p> ");
      script += (String)"test" + " " + WiFi.localIP().toString() + " " + (String)ssid + " <br>" ;
      script += "</p> </h1>  \n";
      script += "</body> \n";
      script += "</html> \n";
      script += ("<br> \n");
      script += ("<br> \n");a

      client.print(script);
      script = "";
   }
   delay(1);
}

void handleRoot() {
   ;
}

void handleClients() {
   ;
}

```

atanisoft, 2020-06-15:
> the code runs through, no Guru errror, but also no website created!

Just a quick thought... you are connecting to port 8080 and not 8081 correct? I see you have a WebServer instance configured for 8081 and the handlers are stubbed off to not do anything which could give you the results you are seeing. But if you are connecting to 8081 I would expect it to work, perhaps you can add some Serial.println() statements in your handleWebsite() method to see where it is getting to?

dsyleixa, 2020-06-15:
yes, ports 8080 and 8081 work fine with esp8266.
I will add some prints.

dsyleixa, 2020-06-15:
runs through, no client connection:

```
WiFi connected: 192.168.2.1
WiFi Server started
ESP webserver started
Use this URL to connect: http://192.168.2.201:8080/
myweb.sytes.net:8080/
setup done 

DEBUG:
 new loop

DEBUG:
 handleWebsite()

DEBUG:
 !client => return

DEBUG:
 webserver.handleClient()

DEBUG:
 new loop

DEBUG:
 handleWebsite()

DEBUG:
 !client => return

DEBUG:
 webserver.handleClient()

DEBUG:
 new loop

DEBUG:
 handleWebsite()

DEBUG:
 !client => return

DEBUG:
 webserver.handleClient()

DEBUG:
 new loop

DEBUG:
 handleWebsite()

DEBUG:
 !client => return

DEBUG:
 webserver.handleClient()

DEBUG:
 new loop

```

```
#include <WiFi.h>
#include <WebServer.h>
IPAddress   this_ip(192, 168, 2, 201); //
IPAddress   gateway(192, 168, 2, 1);       // <<< LAN Gateway IP
IPAddress   subnet(255, 255, 255, 0);      // <<< LAN Subnet Mask
WiFiServer  wifiserver(8080);
WebServer   webserver(8081);
const char* ssid="WLAN-3YA7LD";              // WIFI network name
const char* password="18658446694458594657";          // WIFI network password
//...

void setup() {
//----------------------------------------
   Serial.begin(115200);
   delay(1000);
   // Connect to WiFi network
   Serial.println();
   Serial.println();
   Serial.println("Connecting to WiFi: ");
   Serial.println( WiFi.gatewayIP().toString() );

   WiFi.mode(WIFI_STA);
   WiFi.config(this_ip, gateway, subnet, gateway, gateway);   // dns = gateway
   WiFi.begin(ssid, password);

   while (WiFi.status() != WL_CONNECTED) {
      delay(500);
      Serial.print(".");
   }

   Serial.println("");
   Serial.print("WiFi connected: ");
   Serial.println(WiFi.gatewayIP());

   //----------------------------------------
   wifiserver.begin();
   Serial.println("WiFi Server started");

   //----------------------------------------
   webserver.on("/", handleRoot) ;
   webserver.on("/client/client0/", handleClients);
   delay(10);
   webserver.begin();
   Serial.println("ESP webserver started");

   // Print the IP address
   Serial.print("Use this URL to connect: ");
   Serial.print("http://");    Serial.print(WiFi.localIP());
   Serial.print(":");    Serial.print(8080);
   Serial.println("/");
   Serial.println((String)"myweb.sytes.net" + ":" + (String)(8080) + "/");
   Serial.println("setup done ");
}

void loop() {
   Serial.println("\nDEBUG:\n new loop");

   static double ftmp;
   static unsigned long tsec = millis(), tms = millis();

   Serial.println("\nDEBUG:\n handleWebsite()");
   handleWebsite();                               // <<<<<<<<<<<<<<<<<<<<<<<<<< CRASH !!

   Serial.println("\nDEBUG:\n webserver.handleClient()");  // <<<<<<<<<< not executed any more
   webserver.handleClient();

   //---------------------------------------
   // Read local  data
   ftmp = -99;
   //ftmp = bmp_x77.readTemperature();
   //
}

void handleWebsite() {

   WiFiClient client = wifiserver.available();
   if (!client) {
      Serial.println("\nDEBUG:\n !client => return");
      return;
   }
   if(client) {
      
      String request = client.readStringUntil('\r');
      Serial.println(request);
      client.flush();
      String script = "\n";
      client.readStringUntil('\r');

      Serial.println("\nDEBUG:\n  script += (HTTP/1.1 200 OK \n) ");

      script += ("HTTP/1.1 200 OK \n");
      script += ("Content-Type: text/html \n");
      script += ("");
      script += ("<!DOCTYPE html> \n");
      script += ("<html> \n");
      // head + title
      script += ("<head> \n");
      // autom. Aktualisierung alle 20 sec.
      script += "<meta http-equiv=\"refresh\" content=\"20; URL=";
      script += (String)"myweb.sytes.net" + ":" + (String)(8080) + "\"> \n" ;

      // utf-8 für "°" Zeichen
      script += "<meta http-equiv=\"Content-Type\" content=\"text/html; charset=utf-8\"> \n" ;
      script += ("<title>");
      script += "myweb.sytes.net";
      script += ("</title> \n");
      script += ("</head> \n");
      // body + caption
      script += ("<body> \n");
      script += ("<h1> <p> ");
      script += (String)"test" + " " + WiFi.localIP().toString() + " " + (String)ssid + " <br>" ;
      script += "</p> </h1>  \n";
      script += "</body> \n";
      script += "</html> \n";
      script += ("<br> \n");
      script += ("<br> \n");

      Serial.println("\nDEBUG:\n client.print(script)");

      client.print(script);
      script = "";
   }
   delay(1);
   Serial.println("\nDEBUG:\n end of handleWebsite()");
}

void handleRoot() {
   ;
   Serial.println("\nDEBUG:\n handleRoot()");
}

void handleClients() {
   ;
   Serial.println("\nDEBUG:\n handleClients()");
}

```

dsyleixa, 2020-06-15:
runs through, no client connection:

```
WiFi connected: 192.168.2.1
WiFi Server started
ESP webserver started
Use this URL to connect: http://192.168.2.201:8080/
myweb.sytes.net:8080/
setup done 

DEBUG:
 new loop

DEBUG:
 handleWebsite()

DEBUG:
 !client => return

DEBUG:
 webserver.handleClient()

DEBUG:
 new loop

DEBUG:
 handleWebsite()

DEBUG:
 !client => return

DEBUG:
 webserver.handleClient()

DEBUG:
 new loop

DEBUG:
 handleWebsite()

DEBUG:
 !client => return

DEBUG:
 webserver.handleClient()

DEBUG:
 new loop

DEBUG:
 handleWebsite()

DEBUG:
 !client => return

DEBUG:
 webserver.handleClient()

DEBUG:
 new loop

```

```
#include <WiFi.h>
#include <WebServer.h>
IPAddress   this_ip(192, 168, 2, 201); //
IPAddress   gateway(192, 168, 2, 1);       // <<< LAN Gateway IP
IPAddress   subnet(255, 255, 255, 0);      // <<< LAN Subnet Mask
WiFiServer  wifiserver(8080);
WebServer   webserver(8081);
const char* ssid="WLAN-";              // WIFI network name
const char* password="1865657";          // WIFI network password
//...

void setup() {
//----------------------------------------
   Serial.begin(115200);
   delay(1000);
   // Connect to WiFi network
   Serial.println();
   Serial.println();
   Serial.println("Connecting to WiFi: ");
   Serial.println( WiFi.gatewayIP().toString() );

   WiFi.mode(WIFI_STA);
   WiFi.config(this_ip, gateway, subnet, gateway, gateway);   // dns = gateway
   WiFi.begin(ssid, password);

   while (WiFi.status() != WL_CONNECTED) {
      delay(500);
      Serial.print(".");
   }

   Serial.println("");
   Serial.print("WiFi connected: ");
   Serial.println(WiFi.gatewayIP());

   //----------------------------------------
   wifiserver.begin();
   Serial.println("WiFi Server started");

   //----------------------------------------
   webserver.on("/", handleRoot) ;
   webserver.on("/client/client0/", handleClients);
   delay(10);
   webserver.begin();
   Serial.println("ESP webserver started");

   // Print the IP address
   Serial.print("Use this URL to connect: ");
   Serial.print("http://");    Serial.print(WiFi.localIP());
   Serial.print(":");    Serial.print(8080);
   Serial.println("/");
   Serial.println((String)"myweb.sytes.net" + ":" + (String)(8080) + "/");
   Serial.println("setup done ");
}

void loop() {
   Serial.println("\nDEBUG:\n new loop");

   static double ftmp;
   static unsigned long tsec = millis(), tms = millis();

   Serial.println("\nDEBUG:\n handleWebsite()");
   handleWebsite();                               // <<<<<<<<<<<<<<<<<<<<<<<<<< CRASH !!

   Serial.println("\nDEBUG:\n webserver.handleClient()");  // <<<<<<<<<< not executed any more
   webserver.handleClient();

   //---------------------------------------
   // Read local  data
   ftmp = -99;
   //ftmp = bmp_x77.readTemperature();
   //
}

void handleWebsite() {

   WiFiClient client = wifiserver.available();
   if (!client) {
      Serial.println("\nDEBUG:\n !client => return");
      return;
   }
   if(client) {
      
      String request = client.readStringUntil('\r');
      Serial.println(request);
      client.flush();
      String script = "\n";
      client.readStringUntil('\r');

      Serial.println("\nDEBUG:\n  script += (HTTP/1.1 200 OK \n) ");

      script += ("HTTP/1.1 200 OK \n");
      script += ("Content-Type: text/html \n");
      script += ("\n");
      script += ("<!DOCTYPE html> \n");
      script += ("<html> \n");
      // head + title
      script += ("<head> \n");
      // autom. Aktualisierung alle 20 sec.
      script += "<meta http-equiv=\"refresh\" content=\"20; URL=";
      script += (String)"myweb.sytes.net" + ":" + (String)(8080) + "\"> \n" ;

      // utf-8 für "°" Zeichen
      script += "<meta http-equiv=\"Content-Type\" content=\"text/html; charset=utf-8\"> \n" ;
      script += ("<title>");
      script += "myweb.sytes.net";
      script += ("</title> \n");
      script += ("</head> \n");
      // body + caption
      script += ("<body> \n");
      script += ("<h1> <p> ");
      script += (String)"test" + " " + WiFi.localIP().toString() + " " + (String)ssid + " <br>" ;
      script += "</p> </h1>  \n";
      script += "</body> \n";
      script += "</html> \n";
      script += ("<br> \n");
      script += ("<br> \n");

      Serial.println("\nDEBUG:\n client.print(script)");

      client.print(script);
      script = "";
   }
   delay(1);
   Serial.println("\nDEBUG:\n end of handleWebsite()");
}

void handleRoot() {
   ;
   Serial.println("\nDEBUG:\n handleRoot()");
}

void handleClients() {
   ;
   Serial.println("\nDEBUG:\n handleClients()");
}

```

[7 more replies]

closed 2020-06-15 by me-no-dev