espressif/arduino-esp32 #4092: ESP 32 Bluetooth dropping out

DavidARBrown, 2020-06-14:
Good morning everyone,

Could someone far smarter than me through some ideas at me around the following issue :

I have an ESP 32 WROOM dev kit that I am interfacing through the Arduino IDE. I have a 2015 Macbook Pro Retina 13" i5 and am using a USB cable with a USB A connect to the Mac and a micro USB connect to power the board.

The issue : When I try and connect my ESP 32 to the Mac via Bluetooth the following happens :

I load the sketch - no issues
I reboot the chip and see it visible for pairing on the Mac - no issues
I pair the ESP 32 and Mac - so far so good.
Then after ca. 4 secs the ESP 32 drops out and loses the connection ?

Can anyone shed any light on potential reasons / solutions for this ?

Any ideas very welcome.

David.

sansillusion, 2020-06-14:
Try using a good external power source. The esp will often go in brownout
with wifi or Bluetooth due to high power consumption. Some computer do not
provide enough power on the usb port or the usb cable might have too much
resistance.

DavidARBrown, 2020-06-15:
Wonderful - thank you very much for the feedback. I had a feeling it may be pulling too much power and shuts down. I'll give it a go. Thanks again for getting back to me.

closed 2020-08-28 by stale[bot]