espressif/arduino-esp32 #4093: IO00 power issues on custom circuit board

joeltbennett, 2020-06-14:
I am trying to run WiFi sketch. It uploads successfully but when I open serial monitor after upload, the messages said "ets Jun 8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download" 

This is self assembled board (using ESP-WROOM-32 module).
Previously it works just fine. Today I got this problem.
I have to do manual reset(EN) and IO0 to gnd before sketch upload.

_Originally posted by @studuino in https://github.com/espressif/arduino-esp32/issues/577#issuecomment-321994132_

Hi - I have a problem that exhibits a lot of the same symptoms as this old one referenced above. Basically, my custom circuit board design must not be powering IO00 enough or some such thing. Even in the very rare event I am able to get a proper serial read (meaning the board is finally running the program after a restart and the correct combination of power/ground wires set up), the serial communication still fails to occur for Serial1 (running out to a nextion screen).

I've tested it up and down, baud rates correct, re-ran all the code on a breakout board setup. Everything is good - it's an issue with this circuit board and I think it's power related.

Is there anyone here who has a custom ESP WROOM 32 based circuit board which works properly? I would like to compare it to my own design and see if there's something wrong

Here's the PDF:

https://drive.google.com/file/d/1Q-sIIrPPeWGy_laVgasTINMRztvn6sar/view?usp=sharing

Any help would be greatly appreciated, I've been going mental with this. I didn't design the circuit so I'm limited with what I can do, the guy who did is in Netherlands and not communicating atm.

lbernstone, 2020-06-14:
IO0 is not reset.  It is the programmer jumper.  It only needs to be pulled down during boot to upload new firmware.

joeltbennett, 2020-06-14:
Thank you for looking at it, that's very helpful. I am going to forward this thread to the circuit guy, he'll know how to translate that into a circuit. I just do the software end. Thank you again.

closed 2020-06-14 by joeltbennett