espressif/arduino-esp32 #4094: Spindle Enable Pin Behavior

reynolds087, 2020-06-15:
Is the spindle enable pin supposed to be on when the device boots up? It seems that in order for it to be useful, it should only be active when the spindle is supposed to be rotating.

What you are trying to do?

I want the spindle enable pin to be default inactive to prevent the spindle from ever being on except when the spindle is activated. 

Describe your system( Hardware, computer, O/S, core version, environment).

ESP-32S development board, Gigabyte Aero 15 laptop, Windows 10 x64, PWM is optocoupled to a homemade mosfet driver.

Describe what is failing.

When I turn on the ESP32, the spindle enable pin is active meaning the contacts of the relay are connected. In Candle, I turn the spindle on (no change to the pin, but the spindle begins rotating), and then turn it off, the spindle enable pin turns off, relay disconnects the contacts and the motor stops rotating). I am debugging my mosfet driver, and sometimes it activates when it shouldn't. The relay is connected to spindle enable with the intention of using it as a safety feature, but it doesn't work that way because the circuit is closed at startup and allows the motor to engage when it shouldn't.

atanisoft, 2020-06-15:
@reynolds087 no clue what you are referring to as "spindle enable pin"...

reynolds087, 2020-06-16:
Sorry, meant to post this in GRBL.

closed 2020-06-16 by reynolds087