espressif/arduino-esp32 #4097: enable FF_USE_FASTSEEK

yetifrisstlama, 2020-06-18:
This might be more of an esp-idf feature request, but Arduino is also affected as it's using the pre-compiled esp-idf libraries and there is no easy way to modify them.

Without `FF_USE_FASTSEEK`, seeking __backwards__ on a FAT32 FS takes __hundreds of ms__.

This shows up, for example, when streaming animation frames from a SD card, where some frames need to be shown repeatably.

Enabling `FF_USE_FASTSEEK` costs ~64 words of memory per open file and reduces seek time to << 1 ms.

I tested this by overriding the `ff.c` file with my patched version:
https://github.com/yetifrisstlama/espirgbani_pio/commit/2c957e3f77fba3410e91f84a3a8304582586eda8

See also:
http://www.elm-chan.org/fsw/ff/doc/lseek.html

yetifrisstlama, 2020-06-18:
Here is a benchmark project:
https://github.com/yetifrisstlama/fast_seek_test

Needs platform.io and an ESP32 with SD card connected.

```
------------------------
 without FAST_SEEK
------------------------
[I][main.cpp:34] setup(): Creating 200 MB file: /sd/test.dat ...
[I][main.cpp:58] setup(): fseek( 104857750) dt: 128002 us, read 4096
[I][main.cpp:58] setup(): fseek( 104857751) dt: 128022 us, read 4096
[I][main.cpp:58] setup(): fseek( 104857752) dt: 128044 us, read 4096
[I][main.cpp:58] setup(): fseek( 104857753) dt: 128061 us, read 4096
[I][main.cpp:58] setup(): fseek( 104857754) dt: 128012 us, read 4096
[I][main.cpp:58] setup(): fseek( 104857755) dt: 127987 us, read 4096
[I][main.cpp:58] setup(): fseek( 104857756) dt: 128001 us, read 4096

------------------------
 with FAST_SEEK enabled
------------------------
(comment last line in platform.ini)
[I][main.cpp:34] setup(): Creating 200 MB file: /sd/test.dat ...
[V][ff.c:3736] f_open(): FAST_SEEK enabled
[I][main.cpp:58] setup(): fseek( 104857750) dt:   1143 us, read 4096
[I][main.cpp:58] setup(): fseek( 104857751) dt:    692 us, read 4096
[I][main.cpp:58] setup(): fseek( 104857752) dt:    684 us, read 4096
[I][main.cpp:58] setup(): fseek( 104857753) dt:    681 us, read 4096
[I][main.cpp:58] setup(): fseek( 104857754) dt:    679 us, read 4096
[I][main.cpp:58] setup(): fseek( 104857755) dt:    682 us, read 4096
[I][main.cpp:58] setup(): fseek( 104857756) dt:    681 us, read 4096
```

G2Barbour, 2020-07-10:
This sort of thing should get more attention. I was looking for answers to something else, but I was happy to stop and read this along the way.

lbernstone, 2020-07-10:
This is an upstream issue.  If someone opens the issue at https://github.com/espressif/esp-idf/issues, it can be added.  It's not fixable here.

closed 2020-09-24 by stale[bot]