espressif/arduino-esp32 #4098: BLE manufacturer data should be able to have '0' in it

rhoggarth, 2020-06-18:
### Hardware:
Board:							WEMOS LOLIN32

### Description:
Trying to set manufacturer data to 0x00,0xFF,0x28,0x50.  This fails because the BLEAdvertinsementData::setManufacturerData function expects a std:str.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch

//Relevant subset of code below
void setup() {
  BLEAdvertisementData advertisementData;
  char manData[] = {0x00,0xff,0x28,0x50,0x00};  // Final 0x00 added only as a string terminator
  advertisementData.setManufacturerData(manData);

 // Add other advertisementData, setAdvertisementData() and startAdvertising()

// Raw data in nRF Connect shows 0x01, 0xFF (data type FF is manufacturer data  total length 1 byte)

}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

Jeroen88, 2020-06-18:
Try:

```
char char manData[] = {0x00,0xff,0x28,0x50};
advertisementData.setManufacturerData(std::string(manData, sizeof manData -1));

```
I think that will work.

rhoggarth, 2020-06-18:
Thanks yep, that does indeed work.  Closing the ticket.  Didn't realize you could override the default zero-terminator of std::string.

closed 2020-06-18 by rhoggarth