espressif/arduino-esp32 #4100: ESP32-S2  Pin interrupt does not seem to work with example code.

RudyFiero, 2020-06-18:
### Hardware:
Board:							ESP32-S2-Saola-1
Core Installation version:			?1.0.0? ?1.0.1-rc4? ?1.0.1? ?1.0.1-git? ?1.0.2? ?1.0.3?
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                    no
Upload Speed:				      115200
Computer OS:                                        Windows 10

### Description:
I tried to run the interrupt code example. It never produced the interrupt response. 
https://github.com/espressif/arduino-esp32/blob/esp32s2/libraries/ESP32/examples/GPIO/GPIOInterrupt/GPIOInterrupt.ino

I used two different port pins compared to the example, otherwise no change to the code.  I never thought to enable debug. 

After running that code unsuccessfully, I added a digital read of the port pin in Loop() along with a digital write to a different pin with an LED connected. The LED followed the change on the port pin watched.

### Sketch:  
#include <Arduino.h>

struct Button {
    const uint8_t PIN;
    uint32_t numberKeyPresses;
    bool pressed;
};

Button button1 = {23, 0, false};
Button button2 = {18, 0, false};

void ARDUINO_ISR_ATTR isr(void* arg) {
    Button* s = static_cast<Button*>(arg);
    s->numberKeyPresses += 1;
    s->pressed = true;
}

void ARDUINO_ISR_ATTR isr() {
    button2.numberKeyPresses += 1;
    button2.pressed = true;
}

void setup() {
    Serial.begin(115200);
    pinMode(button1.PIN, INPUT_PULLUP);
    attachInterruptArg(button1.PIN, isr, &button1, FALLING);
    pinMode(button2.PIN, INPUT_PULLUP);
    attachInterrupt(button2.PIN, isr, FALLING);
}

void loop() {
    if (button1.pressed) {
        Serial.printf("Button 1 has been pressed %u times\n", button1.numberKeyPresses);
        button1.pressed = false;
    }
    if (button2.pressed) {
        Serial.printf("Button 2 has been pressed %u times\n", button2.numberKeyPresses);
        button2.pressed = false;
    }
    static uint32_t lastMillis = 0;
    if (millis() - lastMillis > 10000) {
      lastMillis = millis();
      detachInterrupt(button1.PIN);
    }
}
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

lbernstone, 2020-06-18:
Ok.  Please edit your OP to indicate that your version is esp32s2.

RudyFiero, 2020-06-18:
When I get home tonight I will see if there is anything more I can add.

chegewara, 2020-06-18:
Bug is found and will be fixed soon. Thanks.

chegewara, 2020-06-19:
In meantime you can patch it:
https://github.com/espressif/arduino-esp32/blob/esp32s2/cores/esp32/esp32-hal-gpio.c#L350

chegewara, 2020-06-19:
In meantime you can patch it:
https://github.com/espressif/arduino-esp32/blob/esp32s2/cores/esp32/esp32-hal-gpio.c#L350

```
#if CONFIG_IDF_TARGET_ESP32S2
        GPIO.pin[pin].int_ena = 1;
#else
        GPIO.pin[pin].int_ena = 4;
#endif
```

RudyFiero, 2020-06-19:
Great! Thanks for this.

closed 2020-06-19 by RudyFiero