espressif/arduino-esp32 #4101: TTGO-CAMERA-PLUS ESP32 COMPILER SAYS SKETCH TOO LARGE FOR WEBSERVER

johncblacker, 2020-06-18:
### Hardware:
Board:							?TTGO-CAMERA-PLUS? specified as WROVER board
Core Installation version:			?commit b92c58d74b151c7a3b56db4e78f2d3c90c16446f?
IDE name:						?Arduino IDE?
Flash Frequency:					?40Mhz?
PSRAM enabled:                                    ?no? ?yes? Don't know how????????????
Upload Speed:					?115200?
Computer OS:                                        ?Windows 10? 

### Description:
Errors on compile camerawebserver on TTGO-CAMERA-PLUS.  I chose ESP32 wrover module, 
8mb flash partition scheme (just guessing, board is 4mb + 8mb PSRAM, but don't know what option
to pick.  Compile error messages:
***************** error messages start ********************
text section exceeds available space in boardSketch uses 2106895 bytes (160%) of program storage space. Maximum is 1310720 bytes.

Global variables use 53768 bytes (16%) of dynamic memory, leaving 273912 bytes for local variables. Maximum is 327680 bytes.
Sketch too big; see http://www.arduino.cc/en/Guide/Troubleshooting#size for tips on reducing it.
Error compiling for board ESP32 Wrover Module.
***************** error messages end **********************
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

chegewara, 2020-06-18:
You have in `Tools->Partition scheme` different options. Default is with 1.3MB partition and its way too smal for 2MB+ app.

johncblacker, 2020-06-18:
Regarding previous errors, I changed to partition scheme: Huge app and my app loaded, I connected to the webserver, requested a still picture and also a stream, but got errors: 
**************  serial monitor messages *****************
WiFi connected
Starting web server on port: '80'
Starting stream server on port: '81'
Camera Ready! Use 'http://192.168.0.25' to connect
[E][camera.c:1335] esp_camera_fb_get(): Failed to get the frame on time!
Camera capture failed
[E][camera.c:1335] esp_camera_fb_get(): Failed to get the frame on time!
Camera capture failed
************** end of serial monitor messages *****************
I don't know how to determine the next course of action. 
Thanks for help.

closed 2020-09-01 by stale[bot]