espressif/arduino-esp32 #4102: EEPROM data erases everytime

gsaikiran29, 2020-06-19:
Hello, 
I am new to esp32, I am using esp32 dev module. I want to store data in internal (Flash as EEPROM) using EEPROM.h library functions, successfully done with writing data and reading data.
but when I press reset button on board the data erased. why?

I have one more problem i.e., when i use strcpy(); function to copy char array. totally not working and gives guru meditation error on serial. why I am not able to use this function?
please help me to go forward?

usane1, 2020-06-19:
No problem with the EEPROM library at all. I use it without problem. It even keep the data after OTA update.

lbernstone, 2020-06-19:
1) EEPROM requires a commit().  It is deprecated, and the Preferences library should be used if code does not need to portable.
2) strcpy requires an allocated char* to receive the copied data.

gsaikiran29, 2020-06-20:
yeah! I missed commit(); now its working perfect. Thank you for the Information about strcpy also.

closed 2020-06-20 by gsaikiran29