espressif/arduino-esp32 #4104: WiFiUDP::parsePacket() CRASH

piotrus04, 2020-06-19:
Somebody has the same crash than me using WiFiUDP ?

abort() was called at PC 0x40165d87 on core 1

Backtrace: 0x4008cf6c:0x3ffd4850 0x4008d19d:0x3ffd4870 0x40165d87:0x3ffd4890 0x40165dce:0x3ffd48b0 0x40165e7b:0x3ffd48d0 0x40165efa:0x3ffd48f0 0x40165f11:0x3ffd4910 0x400f3c92:0x3ffd4930 0x400d5d8c:0x3ffd4970 0x400f9159:0x3ffd49a0 0x400d7200:0x3ffd49c0 0x400d722b:0x3ffd49e0 0x40089285:0x3ffd4a00
  #0  0x4008cf6c:0x3ffd4850 in invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
  #1  0x4008d19d:0x3ffd4870 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
  #2  0x40165d87:0x3ffd4890 in __cxxabiv1::__terminate(void (*)()) at /Volumes/build/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc:112
  #3  0x40165dce:0x3ffd48b0 in std::terminate() at /Volumes/build/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc:112
  #4  0x40165e7b:0x3ffd48d0 in __cxa_throw at /Volumes/build/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_throw.cc:87
  #5  0x40165efa:0x3ffd48f0 in operator new(unsigned int) at /Volumes/build/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_op.cc:54
  #6  0x40165f11:0x3ffd4910 in operator new[](unsigned int) at /Volumes/build/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_opv.cc:32
  #7  0x400f3c92:0x3ffd4930 in WiFiUDP::parsePacket() at /Users/pierregufflet/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiUdp.cpp:221
  #8  0x400d5d8c:0x3ffd4970 in ArtnetWifi_V2::receive_udp_task(void*) at src/ArtnetV2/ArtnetWifi_V2.h:57

I just call udp.parsepacket()

This only occurs if i send several Artnet Universes : 32 packets of 530 bytes instantly crashes.

i log the esp32 freeheap just before crash : 
/// 
N: getFreeHeap 56472
N: getMaxAllocHeap 51016
///

Espressif last stable version : 
1.12.4

### Hardware:
Board:							ESP32devkitc v4
Core Installation version:			?1.0.0? ?1.0.1-rc4? ?1.0.1? ?1.0.1-git? ?1.0.2? ?1.0.3?
IDE name:						Arduino IDE Platform.io 
Flash Frequency:					40Mhz
PSRAM enabled:                                    no
Upload Speed:					115200
Computer OS:                                        Mac OSX

atanisoft, 2020-06-19:
without code to reproduce this it won't go anywhere. It is crashing due to inability to allocate memory for the UDP packet, how much etc is not clear due to lack of code.

piotrus04, 2020-06-19:
ok thanks for your answer, i will post soon detailled source

AnzuTeam, 2021-06-25:
had this issue, changed the implementation of handlePacket to use buf on stack instead of re-allocating it over and over with every call. problem seems to be solved.

crisreimberg, 2021-06-27:
> had this issue, changed the implementation of handlePacket to use buf on stack instead of re-allocating it over and over with every call. problem seems to be solved.

Could you describe the new code and where did you put it? I had this issue, it is very random, but it every comes again.
Thanks.

michaelbadichi, 2021-06-27:
> Could you describe the new code and where did you put it? I had this issue, it is very random, but it every comes again.
> Thanks.
sure, in the file Arduino15/packages/esp32/hardware/esp32/2.0.0-alpha1/libraries/WiFi/src/WiFiUdp.cpp
change the implementation to this:
`int WiFiUDP::parsePacket(){
  if(rx_buffer)
    return 0;
  struct sockaddr_in si_other;
  int slen = sizeof(si_other) , len;
  char buf[1460];
  if ((len = recvfrom(udp_server, buf, 1460, MSG_DONTWAIT, (struct sockaddr *) &si_other, (socklen_t *)&slen)) == -1){
    if(errno == EWOULDBLOCK){
      return 0;
    }
    log_e("could not receive data: %d", errno);
    return 0;
  }
  remote_ip = IPAddress(si_other.sin_addr.s_addr);
  remote_port = ntohs(si_other.sin_port);
  if (len > 0) {
    rx_buffer = new cbuf(len);
    rx_buffer->write(buf, len);
  }
  return len;
}`

crisreimberg, 2021-06-27:
> sure, in the file Arduino15/packages/esp32/hardware/esp32/2.0.0-alpha1/libraries/WiFi/src/WiFiUdp.cpp
> change the implementation to this:
> int WiFiUDP::parsePacket(){
> if(rx_buffer)
> return 0;
> struct [...]

Thank you a lot, I already implement it and seems to be good. 👍

g3gg0, 2022-06-29:
As I also permanently have issues with exactly this code and the change frrom @michaelbadichi seems to work fine for me as well - will this get integrated?

lbernstone, 2022-06-29:
[Edit the file](https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/WiFiUdp.cpp) and submit a PR.

DamronDan, 2023-02-14:
We have been experiencing issues as well, this seems to have fixed our issue as well.
I will edit the file and submit a PR.

g3gg0, 2023-02-16:
sorry for this dumb question, as I am not that experienced with C++.

Isn't new() and the cleanup still using heap via malloc, which would result in the same behavior just in different clothing?
or is cbuf using singletons?

closed 2023-02-20 by me-no-dev

bmedici, 2023-07-18:
same problem here :
```
#0  0x400840c1:0x3ffb2610 in panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:402
  #1  0x4008d459:0x3ffb2630 in esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:128
  #2  0x400939d1:0x3ffb2650 in abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/abort.c:46
  #3  0x40177b0b:0x3ffb26d0 in __cxxabiv1::__terminate(void (*)()) at /Users/brnomac003/.gitlab-runner/builds/qR2TxTby/0/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:47
  #4  0x40177b52:0x3ffb26f0 in std::terminate() at /Users/brnomac003/.gitlab-runner/builds/qR2TxTby/0/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:57
  #5  0x40176ef7:0x3ffb2710 in __cxa_throw at /Users/brnomac003/.gitlab-runner/builds/qR2TxTby/0/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_throw.cc:95
  #6  0x40177036:0x3ffb2730 in operator new(unsigned int) at /Users/brnomac003/.gitlab-runner/builds/qR2TxTby/0/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/new_op.cc:54
  #7  0x401768d5:0x3ffb2750 in operator new[](unsigned int) at /Users/brnomac003/.gitlab-runner/builds/qR2TxTby/0/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/new_opv.cc:32
  #8  0x40192521:0x3ffb2770 in WiFiUDP::parsePacket() at /Users/bruno/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiUdp.cpp:210
  #9  0x40182d8a:0x3ffb27c0 in ArduinoOTAClass::handle() at /Users/bruno/.platformio/packages/framework-arduinoespressif32/libraries/ArduinoOTA/src/ArduinoOTA.cpp:379
  #10 0x400e7acd:0x3ffb27e0 in ota_tick() at .pio/libdeps/ser/vanbox-lib/src/ota.cpp:87
  #11 0x400dc48b:0x3ffb2800 in loop() at src/main.cpp:263
  #12 0x400f01fd:0x3ffb2820 in loopTask(void*) at /Users/bruno/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:50
```

PR #7847 (2023-02-20): WiFiUDP:parsePacket() Crashfix
-----------
## Description of Change
Resolves crash in WiFiUDP:parsePacket() where allocation of `new char[1460];` throws an exception under low memory conditions.

## Tests scenarios
Tested on FeatherESP32 using UDP socket listener.  Using arduino-esp32 core v2.0.5.
Additional testing welcome!

## Related links
Closes #4104 (stale/closed), #7558, #7845