espressif/arduino-esp32 #4105: Failure connecting to PC-Hosted Access Point

Ivorforce, 2020-06-20:
### Hardware:
Board:							ESP32-WROOM32-D
Core Installation version:			1.12.2
IDE name:						Arduino IDE / Platform.io
Computer OS:                                        macOS / Windows 10

### Description:
Hi, I'm trying to connect my ESP32 to hotspots created on some PC host without going over an additional node (e.g. router).

The code from the sketch works fine connecting to my established internet router (fritz.box) and for hotspots created on my android phone, but not for hotspots created on macOS Catalina or Windows 10. Other devices can connect to these networks fine, so it must be some technical difference that causes the issue.

3 hotspot-creation methods were attempted, but failed to work, as described here:

1) Hotspot via "Internet Sharing".
Create loopback network:
```
sudo networksetup -createnetworkservice Loopback lo0
sudo networksetup -setmanual Loopback 172.20.42.42 255.255.255.255
```
Share via internet sharing.
[image]

2) Hotspot via "Create Network":

[image]

3) Hotspot via "Share my Internet over Wifi/Bluetooth" on Windows 10.

### Sketch: 
```cpp

//Change the code below by your sketch
#include <Arduino.h>
void setup() {
    WiFi.mode(WIFI_MODE_STA);
    WiFi.begin("someSSID", "somePassword");
}

void loop() {
   if (!WiFi.isConnected())
        WiFi.begin("someSSID", "somePassword");

   delay(2000);
}
```

### Debug Messages:
```
(Debug: None that I can discern relevant, WIP)
```

There are no error messages. The network simply tries to connect, but fails.
I have affirmed all connections are 2.4GHz, and passwords and SSIDs are correct.

Ivorforce, 2020-06-20:
From the debug messages, I was able to figure it out. Calling WiFi.begin() again before the connection is fully established can choke out the connection attempt, and computer WiFis seem to be slower in setup than the router.

A simple fix is found in simply tracking desired SSID / Password states and only calling WiFi.begin() again if any of the two change. The WiFi chip handles disconnects, reconnection attempts and anything of the sort just fine. It only stops attempting connection when it is specifically requested by the caller.

closed 2020-06-20 by Ivorforce

CelliesProjects, 2020-07-02:
I would do:
```c++
void setup() {
    WiFi.mode(WIFI_MODE_STA);
    WiFi.begin("someSSID", "somePassword");
    if (WiFi.waitForConnectResult() == WL_CONNECTED)
	Serial.println(F("WiFi connected"));
    else
	Serial.println(F("WiFi not connected"));
}
```