espressif/arduino-esp32 #4107: (.literal._Z41__static_initialization_and_destruction_0ii+0x10): undefined reference to `IRrecv::IRrecv(unsigned short, unsigned short, unsigned char, bool)'

MarkEvens, 2020-06-21:
I am using arduino as component with esp-idf4.0 in eclipse with esp-idf plugin.
And trying to use IRremoteESP8266 liberary with esp-idf but it gives below error.
in this line 
*************************************************************************
IRrecv irrecv(kRecvPin, kCaptureBufferSize, kTimeout, true);
*************************************************************************
.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/main/libmain.a(gatts_demo.cpp.obj):(.literal._Z41__static_initialization_and_destruction_0ii+0x10): undefined reference to `IRrecv::IRrecv(unsigned short, unsigned short, unsigned char, bool)'

.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/main/libmain.a(gatts_demo.cpp.obj):(.literal._Z41__static_initialization_and_destruction_0ii+0x14): undefined reference to `IRrecv::~IRrecv()'

.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/main/libmain.a(gatts_demo.cpp.obj): in function `__static_initialization_and_destruction_0(int, int)':

undefined reference to `IRrecv::IRrecv(unsigned short, unsigned short, unsigned char, bool)'
/home/bhautik/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/bhautik/eclipse-workspace/gatt/build/../main/gatts_demo.cpp:41: undefined reference to `IRrecv::~IRrecv()'
co

lbernstone, 2020-06-21:
We don't support 3rd party libraries here, and definitely are not going to support a esp8266 library.
https://github.com/lbernstone/IR32

MarkEvens, 2020-06-22:
It has any tutorial or guid about adding third-party library esp-idf when we using Arduino as component

MarkEvens, 2020-06-22:
today tried to compile using terminal and it's work fine. but when i compile it inside eclipse it gives above error.

closed 2020-09-04 by stale[bot]