espressif/arduino-esp32 #4108: ESP32-S2 support for USB HID

HelloDB, 2020-06-21:
Does esp32-s2 support USB mouse and keyboard? Can you provide sample code

lbernstone, 2020-06-21:
Yes, but you will need to be patient or use example code from IDF for now.  esp32-s2 is still in beta.

me-no-dev, 2020-06-22:
working on it ... HID needs to be tied to IDF's HID lib.

closed 2020-09-04 by stale[bot]

pefford, 2020-12-29:
Hi, I couldn't find any library to use ESP32 as an HID/MIDI...
Is there any solution? Any code example, so far?
Thanks in advance!

chegewara, 2020-12-29:
There is one i know about, but it is not complete yet, especially in HID.
https://github.com/chegewara/EspTinyUSB

chegewara, 2020-12-29:
And another i just found:
https://github.com/lathoub/Arduino-AppleMIDI-Library

pefford, 2020-12-29:
Thanks @chegewara, though seems to me that these libr are intended for the new ESP32-S2?

chegewara, 2020-12-29:
The first one, as described in readme, is for esp32 s2 and is using native USB.
Second one is using ethernet/wifi and can be used with old esp32 so i am guessing it will works with S2 too (its midi only).

pefford, 2020-12-29:
Thanks @chegewara, i will manage an ESP32-S2 to test... i was initially expecting to use the "old" ESP32 as an HID, but seems not to be possible, except for using BLE...

chegewara, 2020-12-29:
If you have some interesting project i may update my library to add support for more HID devices.

chegewara, 2021-01-04:
@HelloDB i just added HID to my library:
https://github.com/chegewara/EspTinyUSB/tree/master/examples/hid

pefford, 2021-01-04:
Hi @chegewara, is your library update then now valid for the "old" ESP32 to let it act as an HID when physically plugged into the USB slot of a PC (not pairing it over BLE) ?

chegewara, 2021-01-04:
esp32-S1 will never have standalone USB support. Hardware is not capable of it. But you can add S2 to your project (cheap) or wait till S3 is available: `S1 + S2 + extras = S3`

pefford, 2021-01-04:
Ahhh... Thanks for clarifying!

Murilo-la, 2021-06-09:
Hello @chegewara im working in a project to read barcodes with hid scanner, can you help me with the code? Does the keyboard example works with the device or do i need to make changes? Thank you in advance

chegewara, 2021-06-09:
Hi,
actually my library is no longer working with arduino-esp32 v2.0.0 or master branch. It is caused by conflict with hal library in arduino-esp32. Its because some tinyusb callbacks has to be called in library, but are used in hal code and there is conflict.
Sorry for that

Murilo-la, 2021-06-09:
No problem, thank you for the reply

chegewara, 2021-06-09:
@Murilo-la 
There are few older commits of arduino-esp32 you could still use with my library, but they are considered as not stable.

Murilo-la, 2021-06-09:
@chegewara
My team have a solution using esp32 wroom 32u and a max3421 ic, i ll try to adapt it in the esp32 s2, and if it shows difficulties i ll stay with the 32u solution

Angeliccare, 2022-01-23:
Just a fix to a link with the examples: https://github.com/chegewara/EspTinyUSB/blob/master/examples/device/hid/keyboard/keyboard.ino

Pol-Moreno, 2022-04-05:
Hi @chegewara,
I'm starting at this world and I need some help.
I'm working too in a project to read barcodes with hid barcode scanner. Is it possible to do that with any library? 
Could you help me with the code I must upload to my ESP32-s2 for reading the codes sent by barcode scanner since the serial monitor from this one?

Thanks!

vigyanabikshu1, 2022-08-14:
I can see USB HID device examples in esp-idf repository