espressif/arduino-esp32 #4109: Device does not execute script after upload

joeltbennett, 2020-06-21:
Hi - I am having a very frustrating time here with a custom board I had made. I posted a problem about it here a few days ago for an issue I had with the thing failing to go into upload mode (I dunno the official names): https://github.com/espressif/arduino-esp32/issues/4093

There was a missing resistor, which I have since soldered onto the board (as instructed in the linked thread). This is the schematic of the board being used: 

https://drive.google.com/file/d/1Q-sIIrPPeWGy_laVgasTINMRztvn6sar/view?usp=sharing

Anyway, now I have a new issue - after uploading, there is no program execution happening at all. Not after reset or power off or anything, just complete stasis. Do I have to now disconnect this resistor that I've soldered on to allow the program to run now?

atanisoft, 2020-06-21:
If IO0 is shorted to GND during power-up it will go direct to download mode and not run the app.

joeltbennett, 2020-06-21:
OK so I'll just try clipping that resistor off then?

atanisoft, 2020-06-21:
you can try that as long as IO0 is shorted to 3v3 afterwards

joeltbennett, 2020-06-21:
ok so the thing is...i have no idea what i'm doing anymore. i'm going to send this to my circuit design guy though and i'll leave it open for a day if he has any questions to add. thank you for clarifying though, i figured there was some new mode-related issues after that first fix.

lbernstone, 2020-06-21:
This forum is specifically for issues with the arduino-esp32 codebase.  Hardware questions should be on esp32.com . The reference designs for the chips and modules are available at espressif.com

joeltbennett, 2020-06-21:
I just really like you guys

closed 2020-06-21 by joeltbennett