espressif/arduino-esp32 #4110: feature request / example code request: website by esp32 server, plus remote control also of esp8266 clients

dsyleixa, 2020-06-23:
feature request / example code request for noobs: 
code for a website by esp32 with buttons, connect with an extra remote esp8266 client and contol remote pins.
Problems: 
- no example code in this repo available how to connect an esp8266 to an esp32 web-or wifiserver over a website,
- different source codes for esp8266 client to  esp8266 (!!)-server(!!) work, but cannot be ported to work with  esp32 .

e.g., to start from:
https://lastminuteengineers.com/creating-esp32-web-server-arduino-ide/
now with  1st button still for esp32 LED_BUILTIN (that part of course works already)
but 2nd button  instead for remote esp8266 client LED_BUILTIN

what I tried (but failed):
problem:
both programs run, but esp8266 client can never connect to esp32 server/website.
So a working example _for all noobs_ in this repo would be highly appreciated.
If already exists somewhere: a link would be appreciated.

ESP32 Server:
```
// Create A Simple ESP32 Web Server In Arduino IDE
// https://lastminuteengineers.com/creating-esp32-web-server-arduino-ide/

#include <WiFi.h>
#include <WiFiClient.h>
#include <ESPmDNS.h>
#include <ArduinoOTA.h>
#include <WebServer.h>
//#include <WiFiServer.h>

String version = "0.03a";

/*Put your SSID & Password*/
const char* ssid = "WLAN";  // Enter SSID here
const char* password = "18658";  //Enter Password here

char www_username[64] = "admin";
char www_password[64] = "esp32";

IPAddress   this_ip(192, 168, 2, 202);       // <<< static local IP of this ESP-webserver
IPAddress   gateway(192, 168, 2, 1);         // <<< LAN Gateway IP
IPAddress   subnet(255, 255, 255, 0);        // <<< LAN Subnet Mask

#define wwwport 8008
#define wifiport 8081  // Port für die esp clients
WebServer  webserver(wwwport);  // www website: website http port 
WebServer  server(wifiport);     // client comm: router wifi port  

// allows you to set the realm of authentication Default:"Login Required"
const char* www_realm = "Custom Auth Realm";
// the Content of the HTML response in case of Unautherized Access Default:empty
String authFailResponse = "Authentication Failed";

uint8_t LED1pin = LED_BUILTIN;
bool LED1status = LOW;

uint8_t LED2pin = 12;
bool LED2status = LOW; // c4out1

volatile int8_t  c4out1=0, C4OUT2=0, C4OUT3=0; // Client4; actual output pin states; stop=0, fwd=1, rev=-1;

//----------------------------------------------------------
void handleNotFound() {
 
  String message = "File Not Found\n\n";
  message += "URI: ";
  message += webserver.uri();
  message += "\nMethod: ";
  message += (webserver.method() == HTTP_GET) ? "GET" : "POST";
  message += "\nArguments: ";
  message += webserver.args();
[... 485 lines cut ...]
         cstringarg(cmsg, "c4out2", carg);  // switch on/off
         Sargval=(String)carg;
         if(Sargval!="") {
            c4out2=Sargval.toInt();
         }
      }

      // debug
      Serial.println( (String)"c4out0=" + (String)c4out0 + " <<<<<<<<<<<<" );
      Serial.println( (String)"c4out1=" + (String)c4out1 + " <<<<<<<<<<<<" );
      Serial.println( (String)"c4out2=" + (String)c4out2 + " " );

      if(c4out0==0) digitalWrite(PIN_OUT0, LOW); else digitalWrite(PIN_OUT0, HIGH);
      if(c4out1==0) digitalWrite(PIN_OUT1, LOW); else digitalWrite(PIN_OUT1, HIGH);
      if(c4out2==0) digitalWrite(PIN_OUT2, LOW); else digitalWrite(PIN_OUT2, HIGH);

   }  // if(client.available())
   client.flush();
   client.stop();

   Serial.println("\nPausiere jetzt ... \n");
   delay(3000);
}

```

atanisoft, 2020-06-23:
With all of the options available to accomplish this sort of task, it doesn't exactly make sense for the arduino-esp32 repository to dictate what option is best as that is very subjective and project specific.

Now as to your issue, it isn't clear why it is not working as you haven't included any logs etc which might help shed light as to why the code you are using is not working as you expect. It doesn't look like an issue in arduino-esp32 at this point.

dsyleixa, 2020-06-23:
as stated, it's just a request for an example  code e.g. in the library Webserver examples.
As to my code examples, I just wanted to show what does not work.
Currenly nowhere is explained so far how to build a website to connect and to communicate to an esp8266 client, which  is what probably 1000s of Arduino users have.

If esp32 code was 100% compatibel to esp8266 code then it was no issue, because esp8266 server to esp8266 client works. Nonetheless, an upgrade to an esp32 server does not work anymore.

Arduino users also are mostly no professonals but  noobs  or hobbyists and program by c+p'ing functioning code.
So this topic about an code example is meant for general example code to Arduino noobs to c+p (like e.g. the "hello server" example.

lbernstone, 2020-06-23:
This is not instructables.com.  The examples are to demonstrate the library and peripheral functions, and often to act as tests for them.  There are 1000s of sites that are specifically for the sort of example you want.

dsyleixa, 2020-06-23:
no, there is no single one - or show one. And "1000s of sites" is basically just hollow chatter.
And it's not up to YOU to decide. It's a feature request to the esp-arduino developers, to the benefit of all Aduino users  who like to purchase an esp32 and work with it.

chegewara, 2020-06-23:
> And it's not up to YOU to decide.

Neither up to you.
This is not feature request either, it request to build app for you.

As you pointed, there is 1000s of users:
> Currently nowhere is explained so far how to build a website by a esp32 to connect and to communicate to an esp8266 client, which is hardware what probably 1000s of Arduino users have.

There is also 1000s of users with RPi, nrf, arduino R3 etc. Im guessing we (community) have to write example that will show how to use those boards with esp32, just because noobs like you (you words) cant build it?
If you dont know how to build code on esp32 then maybe its not for you and stay with esp8266.

dsyleixa, 2020-06-23:
> Neither up to you.
> This is not feature request either, it request to build app for you.
>
> As you pointed, there is 1000s of users:
>
>
> There is also 1000s of users with RPi, nrf, arduino R3 etc. Im [...]

seldom read such a shit like yours.
But quite often here by you and different other snooty and arrogant "contributors" who think they have leased the wisdom and that they have decision-making powers over this repo.

dsyleixa, 2020-06-23:
> Neither up to you.
> This is not feature request either, it request to build app for you.
>
> As you pointed, there is 1000s of users:
>
>
> There is also 1000s of users with RPi, nrf, arduino R3 etc. Im [...]

seldom read such a shit like yours.
But quite often here by you and different other snooty and arrogant "contributors" who troll around and think they have leased the wisdom and that they have decision-making powers over this repo.

lbernstone, 2020-06-23:
Stop feeding the trolls, please.  Github has a block user feature.

dsyleixa, 2020-06-25:
I am always nice, like this time and all times before, as I asked this request, because source code examples  about this topic are missing.
But I am not supposed to be nice to insulting trolling idiots, from which I do not expect help actually (e.g., claiming that there existed already 1000s of examples about this in the web but are unable to link to just 1.
It's also completely unreasonable to close this issue because it's not is not solved yet, and instead take sides with the other fools.
Honestly, I have rarely seen such a horde of narrow-minded professional  idiots like here in this repo. You can be sure that I will share my esp32  experiences and  I already started to advise users in different forums strictly against the ESP32 because of the effing poor support to reported issues.

dsyleixa, 2020-06-25:
I am always nice, like this time and all times before, as I asked this request, because source code examples about this topic are missing.
But I am not supposed to be nice to insulting trolling idiots, from which I do not expect help actually (e.g., claiming that there existed already 1000s of examples about this in the web but are unable to link to just 1).
It's also completely unreasonable that you close this issue because it's not solved yet, and instead take sides with the other fools.
Honestly, I have rarely seen such a horde of narrow-minded professional idiots like here in this repo. You can be sure that I will share my esp32 experiences and I already started to advise users in different forums strictly against the ESP32 because of the effing poor (actually not even existing) support to reported issues.

chegewara, 2020-06-25:
> You can be sure that I will share my esp32 experiences and I already started to advise users in different forums strictly against the ESP32 because of the effing poor (actually not even existing) support to reported issues.

Uh oh, we will miss you.

me-no-dev, 2020-06-26:
I hope you realise that us fools also talk to each other. I am well aware of your history on the ESP8266 side. You were never nice, you do not chose who tries to help you, you do not even try to do what they propose. I am afraid you are expecting too much from an issue tracker. This is a tool to solve bugs and issues in the core, not to help user achieve their goals or add unnecessarily complex examples. You are free to submit such example as PR, discuss implementations and approaches in the gitter chat and so on.

The issue was closed because it's invalid in the scope of the "github issues" tracker. It can be discussed further here, but there is no point in it being shown as active.

p.s. You can continue to insult my competency, I am not bothered. You are the one that does not know how to do things though... And don't forget to link to your examples of our poor professionalism when you spread the word :)

dsyleixa, 2020-06-27:
as stated, I am always nice, like this time and all times before, when I asked questions about issues and answering to constructive replies.
But I never will be nice to insulting snooty idiotic trolls  like e.g. chegewara and others, and admittedly on the esp8266 side there are also a couple of trolling assholes I am not expected to act nicely to.
OTOH, feature requests about extensions are a permissible concern for issue trackers, especially like here if code works for esp8266 (or in other cases, e.g. for original Arduino.cc libs) but not for esp32 any more, so the the principle of cross-board compatibility is viotated and thus working or fixed code is needed.