espressif/arduino-esp32 #4112: esp32 direct port manipulation

Nirvana-56, 2020-06-23:
Hardware:
Board:							ESP32 Dev Module 
Core Installation version:			        1
IDE name:						Arduino IDE 
Flash Frequency:					80Mhz
PSRAM enabled:                                       no
Upload Speed:					       115200
Computer OS:                                         Windows 7

Hi 
esp32 lose some data on its output ( it could be checked via software ,LCD or LED matrix ) and if you check the outputs, it is not stable in some cases it ( like a random noise even on Version 1 esp32 chip ).
I searched it seriously, this is a problem for many people, and I hope it will be solved by helping the expertise:
https://github.com/espressif/esp-idf/issues/3862
https://qiita.com/h_nari/items/d52734440a7256072e8e
I decided to use it as a parallel port instead of I2S ( without DMA) because it is not very good for many cases. In addition, as you know in other microcontrollers do it easily and I think the modern soc like esp32 should do it!!

Regards

`      
for ( int i=0; i<=1919 ;i++) {

        GPIO.out_w1tc=clear_parameter;
 
        GPIO.out_w1ts**=temp[i];
       
        };
`

chegewara, 2020-06-23:
Im not saying there is or there is not an issue, but i can tell few things:
1) your device to make measurements doesnt look to be very professional, and my guess is it is way too slow to make accurate measurements with this code; you may want to check this:
https://articles.saleae.com/logic-analyzers/choosing-a-logic-analyzer#sample-rate
2) ive been trying to check whats going on with that code and esp32 and i failed too, because my LA is too slow, it is only 24MHz, then i switched CPU to 80MHz which allows me to make reliable measurements with my LA and all looks good, nice 4MHz output
3) during tests i discovered that every 1000ms there is short gap, and with help of @atanisoft i know it is freertos ISR tick; you have to remember that esp32 is running on freertos which is working under hood with scheduler and is taking control sometimes over your code
4) finally i did my tests with just a bit changed code, but this should be not that important:
```

#include "esp_task_wdt.h"

void setup()
{
  gpio_config_t io_conf;
  io_conf.mode = GPIO_MODE_OUTPUT;
  io_conf.pin_bit_mask = 0b10100;
  gpio_config(&io_conf);
  xTaskCreatePinnedToCore(task, "t", 1024, NULL, 24, NULL, 1); // pin task to core 1, just like loop is
  vTaskDelete(NULL); // delete loop task
}

void loop()
{
}

void task(void*p)
{
  if(esp_task_wdt_delete(NULL) != ESP_OK){ // WDT is disabled in loop task on arduino
      log_e("Failed to remove task from WDT");
  }
  while(1)
  {
    taskDISABLE_INTERRUPTS();
    GPIO.out_w1ts = 0b10100;
    GPIO.out_w1tc = 0b10100;
    taskENABLE_INTERRUPTS();
  }
}
```
Here is screenshot with CPU 80MHz and very regulat 4MHz output:
[image]

Here is gap i mentioned:
[image]

And some irregular output because my logic analyzer is too slow and CPU is 240MHz:
[image]

Nirvana-56, 2020-06-24:
Hi
Thank you and I really respected your efforts; I appreciated your code, this is something that I did not know ( time intervals reason, irregular output reason, …. )
There are some question:
1-	Due to RTOS nature of esp32, in each freertos ISR tick, it do another work; what is the measure of that period of time ( I mean gap in 240MHZ ) , and is there any control on it ? What is the CPU frequency of the second graph, I measure the gap from your second graph, it is about 0.021 ms (rough estimate) am I right ?
2-	Is there any way to minimize it (the gap) for example task switching…?
3-	Why freertos ISR tick does not stopped even with your command “ taskDISABLE_INTERRUPTS(); 
4-	How we could manage ISR ticks in any code?

Thank you again
Regards

chegewara, 2020-06-24:
1) IIRC it was 240MHz and gap is about 22us
2) i have no idea, this is something i didnt have to work with
3) i dont know, but it is also possible that freertos tick is handled after ISR is enabled, but before it is disabled again, or maybe its priority is high and cant be disabled
4) maybe @atanisoft or @me-no-dev can answer that

DerfJagged, 2020-06-24:
You can use portENTER_CRITICAL() to pause freertos ticks:

In Setup():
`static portMUX_TYPE my_mutex;
vPortCPUInitializeMutex(&my_mutex);`

Wherever:
`portENTER_CRITICAL(&my_mutex);
//timing sensitive stuff
portEXIT_CRITICAL(&my_mutex);`

If you stay in CRITICAL too long, the watchdog will expire and reboot the ESP32, so only use it for spots where precise timing is needed (like in your ISR). That fixes the tick every 1000ms, but there seems to be something else that takes priority even over ISRs. I've been trying to figure out how to fix this as some code of mine works fine but randomly after 20-30 seconds, it takes too long to respond to an ISR, I think it might be possible to esp_intr_free() a port's interrupt, and esp_intr_alloc() it with the highest priority or set ESP_INTR_FLAG_LEVELx for that port to something else. I haven't tested that though. Let me know if you find a way to get that working!

closed 2020-09-07 by stale[bot]