espressif/arduino-esp32 #4113: getting "xtensa-esp32-elf-g++": file does not exist" error." compile error, though my Arduino core for the ESP32 is at C:\\Users\\[USER_NAME]\\Documents\\Arduino\\hardware\\espressif\\esp32/ saved at

shudutta, 2020-06-24:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do?  to toggle LED at PIN 2
Describe your system( Hardware, computer, O/S, core version, environment). Lenovo Thinkpad, WIN10
Describe what is failing. : Compilation
Show the shortest possible code that will duplicate the error.

Show the EXACT error message(it doesn't work is not enough). 
  Arduino: 1.8.13 (Windows 10), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

exec: "C:\\Users\\shudutta\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++": file does not exist

Error compiling for board ESP32 Dev Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.4
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                     no
Upload Speed:					115200
Computer OS:                                        Windows 10

### Description:
Describe your problem here

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

   int led =2;

void setup() {
  // put your setup code here, to run once:
  pinMode(led, OUTPUT);
  Serial.begin(115200);
}

void loop() {
  // put your main code here, to run repeatedly:
  digitalWrite(led,HIGH);
  delay(500);
  digitalWrite(led,LOW);
  delay(500);
}
### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

I've gone through other similar issue but none of the fix is working for me.
Could anyone like to resolve the "xtensa-esp32-elf-g++": file does not exist" error.
I have saved espressif core in correct path, still getting the "xtensa-esp32-elf-g++": file does not exist" error.

lbernstone, 2020-06-24:
Run get.exe in the tools directory.  I would recommend doing this from a cmd window so you can see the output.

shudutta, 2020-06-25:
[image]
I'm getting above error while running get.exe from cmd.
Can you guess why I'm getting the error. Since I'm working on a organisation Laptop,  I doubt the firewall(which I can't turn OFF) is throwing the error.
If the firewall is the issue, would you like to suggest any workaround @lbernstone . Thanks in advance.

sansillusion, 2020-06-25:
I would guess either an antivirus stoping the download or a bad internet
connection

lbernstone, 2020-06-25:
https://desire.giesecke.tk/index.php/2018/04/02/get-exe-get-py-fails-to-download-required-files/

closed 2020-09-07 by stale[bot]