espressif/arduino-esp32 #4116: ESP32 HIMEM

Nirvana-56, 2020-06-25:
Hardware:
Board: Mini32 ESP32-WROVER-B 8MB PSRAM
IDE name: Arduino IDE
Flash Frequency: 80Mhz
Upload Speed: 115200
Computer OS: Windows 7

Hi
why the top 4MB command (ESP32 HIMEM) in the Arduino is not be accessible ? 

extern "C" {
#include <esp_spiram.h>
#include <esp_himem.h>
}

void setup() {
  Serial.begin(115200);
  esp_spiram_init();
  printf("spiram size %u\n", esp_spiram_get_size());
  printf("himem free %u\n", esp_himem_get_free_size());
  printf("himem phys %u\n", esp_himem_get_phys_size());
  printf("himem reserved %u\n", esp_himem_reserved_area_size());
}

void loop() {}

result :

**E (127) spiram: SPI RAM not initialized
E (127) esp_himem: Cannot allocate memory for meta info. Not initializing!**
spiram size 8388608
himem free 0
himem phys 4456448
himem reserved 262144

[image]

atanisoft, 2020-06-25:
can you try:
```
void setup()
{
  Serial.begin(115200);
  esp_himem_init();
  printf("spiram size %u\n", esp_spiram_get_size());
  printf("himem free %u\n", esp_himem_get_free_size());
  printf("himem phys %u\n", esp_himem_get_phys_size());
  printf("himem reserved %u\n", esp_himem_reserved_area_size());
}
void loop() {}
```

As long as you have selected a PSRAM enabled board you do not need to initialize PSRAM yourself as the arduino-esp32 code should do that automatically.

chegewara, 2020-06-25:
Try to pick regular devKit and enable this option:
[image]

lbernstone, 2020-06-25:
and just for clarity, is this release 1.0.4 or esp32s2?

Nirvana-56, 2020-06-26:
Hi, Thank you again @atanisoft   
But there is an error during compiling  Arduino version 1.8.13 and esp_himem_init();
**exit status 1
'esp_himem_init' was not declared in this scope**
Does it work for you ?
May i ask you please attach your drivers, i mean ( esp_himem.h and esp_himem.c )

@chegewara 
This chip is ESP32-D0WD series (LILYGO TTGO Mini32- ESP-WROVER-B module with 8MB psram )
"Try to pick regular devKit and enable this option" It does not work. as you could see in the above picture when you using  ESP-WROVER module option, the enable option will be disappear.

atanisoft, 2020-06-26:
@Nirvana-56 add:
```
#include <esp_himem.h>
```
and it should compile. no extra drivers etc are needed as they are included within the arduino-esp32 codebase.

Nirvana-56, 2020-06-26:
[image]
@atanisoft what is wrong ? the error is still unresolved !!

Nirvana-56, 2020-06-26:
The code is :

#include <esp_himem.h>
#include <esp_spiram.h>

void setup()
{
  Serial.begin(115200);
  esp_himem_init();
  printf("spiram size %u\n", esp_spiram_get_size());
  printf("himem free %u\n", esp_himem_get_free_size());
  printf("himem phys %u\n", esp_himem_get_phys_size());
  printf("himem reserved %u\n", esp_himem_reserved_area_size());
}
void loop() {}

atanisoft, 2020-06-26:
@Nirvana-56 what version of arduino-esp32 are you using? You answered with the IDE version which is not very meaningful.

lbernstone, 2020-06-26:
Himem got a major rework from v3.3-v4.0. There is no init, but allocation is more manual. Please look at the example from v3.3 to see how to allocate:
https://github.com/espressif/esp-idf/blob/release/v3.3/examples/system/himem/main/himem_test_main.c

Nirvana-56, 2020-06-26:
[image]

@atanisoft  I download in this page and using clone !  how i can check the version ?

Nirvana-56, 2020-06-26:
@atanisoft 
[image]

@lbernstone 
Did you have any code for Arduino ?

Nirvana-56, 2020-06-26:
[image]

@atanisoft : sorry i am not expert as you are, what is the main difference between master and other releases idf-release v4.0

atanisoft, 2020-06-26:
@Nirvana-56 you can copy / paste these lines into setup(): 
https://github.com/espressif/esp-idf/blob/release/v3.3/examples/system/himem/main/himem_test_main.c#L92-L94

Or, simply remove the init call as that appears to only be valid from IDF v4.0+ and was inlined as part of the PSRAM initialization calls.

> sorry i am not expert as you are, what is the main difference between master and other releases idf-release v4.0

The branches are used for different versions of IDF support, the only branch outside of "master" that would be useful for you is "esp32s2" which is planned to be the basis of the next arduino-esp32 release. I wouldn't advise cloning/downloading the code directly from GitHub but instead use the [boards manager](https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/boards_manager.md) approach instead.

lbernstone, 2020-06-26:
Change app_main to setup and that code will compile just fine.  I don't have a device with 8MB psram, so I cannot test it.

Nirvana-56, 2020-06-26:
@lbernstone  i rewrite code in Arduino as you mentioned, it seem that it need some modification:
[image]

#include <stdio.h>
#include <stdbool.h>
#include <stdint.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/queue.h"
#include "esp_system.h"
#include "nvs_flash.h"
#include "esp_heap_caps.h"
#include "esp_spiram.h"
#include "rom/cache.h"
#include "sdkconfig.h"
#include "esp_himem.h"

//Fill memory with pseudo-random data generated from the given seed.
//Fills the memory in 32-bit words for speed.
static void fill_mem_seed(int seed, void *mem, int len)
{
    uint32_t *p = (uint32_t *)mem;
    unsigned int rseed = seed ^ 0xa5a5a5a5;
    for (int i = 0; i < len / 4; i++) {
        *p++ = rand_r(&rseed);
    }
}

//Check the memory filled by fill_mem_seed. Returns true if the data matches the data
//that fill_mem_seed wrote (when given the same seed).
//Returns true if there's a match, false when the region differs from what should be there.
static bool check_mem_seed(int seed, void *mem, int len, int phys_addr)
{
    uint32_t *p = (uint32_t *)mem;
    unsigned int rseed = seed ^ 0xa5a5a5a5;
    for (int i = 0; i < len / 4; i++) {
        uint32_t ex = rand_r(&rseed);
        if (ex != *p) {
            printf("check_mem_seed: %x has 0x%08x expected 0x%08x\n", phys_addr+((char*)p-(char*)mem), *p, ex);
            return false;
        }
        p++;
    }
    return true;
}

//Allocate a himem region, fill it with data, check it and release it.
static bool test_region(int check_size, int seed)
{
    esp_himem_handle_t mh; //Handle for the address space we're using
    esp_himem_rangehandle_t rh; //Handle for the actual RAM.
    bool ret = true;

    //Allocate the memory we're going to check.
    ESP_ERROR_CHECK(esp_himem_alloc(check_size, &mh));
    //Allocate a block of address range
    ESP_ERROR_CHECK(esp_himem_alloc_map_range(ESP_HIMEM_BLKSZ, &rh));
    for (int i = 0; i < check_size; i += ESP_HIMEM_BLKSZ) {
        uint32_t *ptr = NULL;
        //Map in block, write pseudo-random data, unmap block.
        ESP_ERROR_CHECK(esp_himem_map(mh, rh, i, 0, ESP_HIMEM_BLKSZ, 0, (void**)&ptr));
        fill_mem_seed(i ^ seed, ptr, ESP_HIMEM_BLKSZ); //
        ESP_ERROR_CHECK(esp_himem_unmap(rh, ptr, ESP_HIMEM_BLKSZ));
    }
    vTaskDelay(5); //give the OS some time to do things so the task watchdog doesn't bark
    for (int i = 0; i < check_size; i += ESP_HIMEM_BLKSZ) {
        uint32_t *ptr;
        //Map in block, check against earlier written pseudo-random data, unmap block.
        ESP_ERROR_CHECK(esp_himem_map(mh, rh, i, 0, ESP_HIMEM_BLKSZ, 0, (void**)&ptr));
        if (!check_mem_seed(i ^ seed, ptr, ESP_HIMEM_BLKSZ, i)) {
            printf("Error in block %d\n", i / ESP_HIMEM_BLKSZ);
            ret = false;
        }
        ESP_ERROR_CHECK(esp_himem_unmap(rh, ptr, ESP_HIMEM_BLKSZ));
        if (!ret) break; //don't check rest of blocks if error occurred
    }
    //Okay, all done!
    ESP_ERROR_CHECK(esp_himem_free(mh));
    ESP_ERROR_CHECK(esp_himem_free_map_range(rh));
    return ret;
}

 void setup()
{
    size_t memcnt=esp_himem_get_phys_size();
    size_t memfree=esp_himem_get_free_size();
    printf("Himem has %dKiB of memory, %dKiB of which is free. Testing the free memory...\n", (int)memcnt/1024, (int)memfree/1024);
    assert(test_region(memfree, 0xaaaa));
    printf("Done!\n");
}

void loop() {
  // put your main code here, to run repeatedly:

}

Nirvana-56, 2020-06-26:
@lbernstone  the code was compiled but there is error in this line according to the back trace :

ESP_ERROR_CHECK(esp_himem_alloc_map_range(ESP_HIMEM_BLKSZ, &rh));

Nirvana-56, 2020-06-26:
@atanisoft : this modification also does not work 

#include <stdio.h>
#include <stdbool.h>
#include <stdint.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/queue.h"
#include "esp_system.h"
#include "nvs_flash.h"
#include "esp_heap_caps.h"
#include "esp_spiram.h"
#include "rom/cache.h"
#include "sdkconfig.h"
#include "esp_himem.h"

void setup() {
  // put your setup code here, to run once:
  size_t memcnt=esp_himem_get_phys_size();
  size_t memfree=esp_himem_get_free_size();
  printf("Himem has %dKiB of memory, %dKiB of which is free. Testing the free memory...\n", (int)memcnt/1024, (int)memfree/1024);

}

void loop() {
  // put your main code here, to run repeatedly:

}

result : 

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
**E (125) spiram: SPI RAM not initialized
E (125) esp_himem: Cannot allocate memory for meta info. Not initializing!
[D][esp32-hal-psram.c:47] psramInit(): PSRAM enabled
Himem has 4352KiB of memory, 0KiB of which is free. Testing the free memory...**

lbernstone, 2020-06-26:
This is the error:
`E (125) esp_himem: Cannot allocate memory for meta info. Not initializing!`
This is originating from https://github.com/espressif/esp-idf/blob/release/v3.3/components/esp32/esp_himem.c#L159, where it tries to allocate the himem.
As mentioned, I do not have an 8MB device, so I cannot help you any further.

lbernstone, 2020-06-26:
WROVER-B is 4MB psram.  WROVER-I is 8MB psram.

atanisoft, 2020-06-26:
@Nirvana-56 I can confirm the same behavior but unsure why it is failing.

```
E (134) spiram: SPI RAM not initialized
E (134) esp_himem: Cannot allocate ⸮fcmory for meta info. Not initializing!
[D][esp32-hal-cpu.c:205] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[D][esp32-hal-psram.c:85] psramInit(): PSRAM enabled
Himem has 4352KiB of memory, 0KiB of which is free. Testing the free memory...
ESP_ERROR_CHECK failed: esp_err_t 0x101 (ESP_ERR_NO_MEM) at 0x40087ba8
```
This was with the esp32s2 branch using DevKit with PSRAM enabled. Similar results were observed with the WROVER selection.

@me-no-dev any ideas?

me-no-dev, 2020-06-26:
s2 does not have highmem. It can map up to 4GB of RAM. ESP32 can do 4MB, thus high-mem.
the first issue is that you get `E (134) spiram: SPI RAM not initialized` before `[D][esp32-hal-psram.c:85] psramInit(): PSRAM enabled`. I do not have 8MB modules here either, so can not test. You can try to disable PSRAM in the board menu and then init it yourself.

atanisoft, 2020-06-26:
@me-no-dev my test was with an ESP32-WROVER on the esp network kit. Why would PSRAM fail to init on startup in IDF but arduino claim it enabled it?

chegewara, 2020-06-26:
It looks like himem calls are before psram is init. Then psram init is enabled.

chegewara, 2020-06-26:
Try to call this before himem in setup:
`psramInit();`

atanisoft, 2020-06-26:
in my test psramInit() was called before setup() was invoked (as shown in the log as well)

chegewara, 2020-06-26:
> E (134) spiram: SPI RAM not initialized
> E (134) esp_himem: Cannot allocate ⸮fcmory for meta info. Not initializing!
> [D][esp32-hal-cpu.c:205] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
> [D][esp32-hal-psram.c:85] psramInit(): PSRAM enabled

logs order shows himem before psramInit

atanisoft, 2020-06-26:
that is before setup() or arduino startup is called

Nirvana-56, 2020-06-26:
@chegewara @me-no-dev @me-no-dev 

Why in first code PSRAM will be enabled and in second one it does not initialized ?

extern "C" {
#include <esp_spiram.h>
#include <esp_himem.h>
}

uint8_t* Buffer0 = NULL;

void setup() {
Serial.begin(115200);

Buffer0 = (uint8_t*)ps_malloc(900000);
psramInit();
//esp_spiram_init();
//printf("spiram size %u\n", esp_spiram_get_size());
//printf("himem free %u\n", esp_himem_get_free_size());
//printf("himem phys %u\n", esp_himem_get_phys_size());
//printf("himem reserved %u\n", esp_himem_reserved_area_size());
}

void loop() {}

Result :

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
[D][esp32-hal-psram.c:47] psramInit(): PSRAM enabled

---------------------------------------------------------------------------------------------

extern "C" {
#include <esp_spiram.h>
#include <esp_himem.h>
}

uint8_t* Buffer0 = NULL;

void setup() {
Serial.begin(115200);

Buffer0 = (uint8_t*)ps_malloc(900000);
psramInit();
//esp_spiram_init();
printf("spiram size %u\n", esp_spiram_get_size());
printf("himem free %u\n", esp_himem_get_free_size());
printf("himem phys %u\n", esp_himem_get_phys_size());
printf("himem reserved %u\n", esp_himem_reserved_area_size());
}

void loop() {}

Result : 
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
E (127) spiram: SPI RAM not initialized
E (127) esp_himem: Cannot allocate memory for meta info. Not initializing!
[D][esp32-hal-psram.c:47] psramInit(): PSRAM enabled
spiram size 8388608
himem free 0
himem phys 4456448
himem reserved 262144

Nirvana-56, 2020-06-26:
@atanisoft

chegewara, 2020-06-26:
Ok, i tested code with arduino as component and sdkconfig from esp32s2 branch for esp32. Here is the thing:
https://github.com/espressif/arduino-esp32/blob/esp32s2/tools/sdk/esp32/sdkconfig#L385
If i enable it then it works:
```
entry 0x400805ec
Himem has 4352KiB of memory, 4352KiB of which is free. Testing the free memory...
Done!
```
with disabled:
```
 (138) spiram: SPI RAM not initialized
E (138) esp_himem: Cannot allocate memory for meta info. Not initializing!
E (40) spiram: SPI RAM not initialized
Himem has -3840KiB of memory, 0KiB of which is free. Testing the free memory...
ESP_ERROR_CHECK failed: esp_err_t 0x101 (ESP_ERR_NO_MEM) at 0x400860fc
```

atanisoft, 2020-06-26:
@chegewara what happens with that flag enabled on a non-psram esp32?

chegewara, 2020-06-26:
There is additional flag to ignore if no psram found.

Nirvana-56, 2020-06-26:
@chegewara : 
Thank you again :+1: 
you mean that it just work when we use it arduino as component of esp-idf  !!
Why it does not work in arduino environment ? How we could change the flag in sdkconfig of arduino?

chegewara, 2020-06-26:
```
Name: SPIRAM_IGNORE_NOTFOUND
Prompt: Ignore PSRAM when not found
Type: bool
Value: y

Help:

  Normally, if psram initialization is enabled during compile time but not found at runtime, it
  is seen as an error making the CPU panic. If this is enabled, booting will complete
  but no PSRAM will be available.

Direct dependencies (=y):
     SPIRAM_BOOT_INIT(=y)
  && !SPIRAM_ALLOW_BSS_SEG_EXTERNAL_MEMORY(=n)  (=y)
  && ESP32_SPIRAM_SUPPORT(=y)

Default:
  - n

Kconfig definition, with parent deps. propagated to 'depends on'
================================================================

At /home/chegewara/esp/esp32s2/components/esp_common/Kconfig.spiram.common:17
Included via /home/chegewara/esp/esp32s2/Kconfig:334 -> kconfigs.in:7 -> /home/chegewara/esp/esp32s2/components/esp32/Kconfig:115
Menu path: (Top) -> Component config -> ESP32-specific -> Support for external, SPI-connected RAM -> SPI RAM config -> Initialize SPI RAM during startup

  config SPIRAM_IGNORE_NOTFOUND
```

> @chegewara :
> Thank you again 
> you mean that it just work when we use it arduino as component of esp-idf !!
> Why it does not work in arduino environment ? How we could change the flag in sdkconfig of arduino?

@me-no-dev need to rebuild arduino with both flags enabled most likely

lbernstone, 2020-06-27:
Works in 3.3 also

atanisoft, 2020-06-27:
@lbernstone is 3.3 a concern since the 4.x branch is the next major release update?

lbernstone, 2020-06-27:
No, just what I was using to test.

closed 2020-09-10 by stale[bot]

arkhipenko, 2021-07-15:
https://github.com/arkhipenko/ESP32Himem

Better late than never...