espressif/arduino-esp32 #4119: CameraWebServer example on ESP32-S2

BackLogers, 2020-06-27:
I created my own OV2640 test board PCB, works fine with ESP32 but on ESP32-S2 example code has a comment:
```cpp
//Disable Example for now
```
When try to compile it, I receive:
### Debug Messages:
```
CameraWebServer:3:10: fatal error: esp_camera.h: No such file or directory
```
Is there any way to run this example on ESP32-S2?

### Hardware:
Board:							ESP32-S2-Saola
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                     yes
Upload Speed:					921600
Computer OS:                                         Windows 10

lbernstone, 2020-06-27:
This sort of comment in the code implies that the developer already knows it isn't working.  S2 has some major changes to psram that should improve camera performance, but will need serious driver changes.

BackLogers, 2020-07-02:
Thank you for the answer, I didn't know that ESP32 and ESP32-S2 are so different. In that case I'll be waiting.

closed 2020-07-02 by BackLogers