espressif/arduino-esp32 #4120: SPIFFS Upload fails for ESP32-S2.  Invalid head of packet (0x50)

RudyFiero, 2020-06-27:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do?
Describe your system( Hardware, computer, O/S, core version, environment).
Describe what is failing.
Show the shortest possible code that will duplicate the error.
Show the EXACT error message(it doesn't work is not enough).
All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							ESP32-S2 Dev Module
Core Installation version:			ESP32-S2 Dev
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                    no
Upload Speed:					115200
Computer OS:                                        Windows 10

### Description:
I am trying to use the ESP32 Sketch Upload Data tool in the Arduino IDE. I get the error shown below. 

It seems that the sketch is loaded into the ESP32-S2 with a newer esptool, but the data uploader is using a previous esptool.py

Where and how can I easily add the required uploader? 

I tried to convert esptool.py to esptool.exe but that resulted in failure. I am not a programmer, I am a hardware developer. I am using the Arduino IDE because of it's simplicity. I need a relatively simple solution. I'm to old to deal with a lot of complexity. 

### Debug Messages:
```

esptool.py v2.6
Serial port COM4
Connecting....
Chip is ESP32-PICO-D4 (revision 0)
Features: WiFi, Single Core, Embedded Flash, VRef calibration in efuse, BLK3 partially reserved, Coding Scheme Repeat (UNSUPPORTED)
MAC: 18:20:69:d1:56:7f
Uploading stub...
Running stub...

A fatal error occurred: Invalid head of packet (0x50)
SPIFFS Upload failed!

```

chegewara, 2020-06-27:
`ESP32-PICO-D4 ` this is not esp32 S2.

lbernstone, 2020-06-27:
Issue belongs at https://github.com/me-no-dev/arduino-esp32fs-plugin, but do a search for esptool.py on your machine.  It should not be anywhere outside of <.arduino>/packages/esp32 or <Arduino_sketch>/hardware/espressif/esp32.
If you also use arduino esp8266 on your machine this can cause conflicts.

RudyFiero, 2020-06-27:
@chegewara  "ESP32-PICO-D4 this is not esp32 S2."

Yes I agree. I have ESP32-S2 Dev Module selected. The problem I believe is that there is no ESP32-S2 in the release code for the esptool version. From what I have found esptool.py v3.0-dev is required for the ESP32-S2. And this (v3.0-dev) is what is being used to upload the code to the ESP32-S2.

@lbernstone   I have found **esptool.py v3.0-dev** but I believe the problem is that it is a python file and not an .EXE
I had tried to convert esptool.py v3.0-dev to esptool.exe but I got errors and am doubtful that I am capable of resolving them. 

Have either of you been able to use the ESP32 Sketch Data Upload in the Arduino environment?

lbernstone, 2020-06-27:
Yes.  Please move issue to the correct repo.

chegewara, 2020-06-27:
I think i dont understand something. You are trying to compile code and flash with tools for esp32 S2, but you have esp32 pico D4, which is old not S2 chip. How do you want to do it?

RudyFiero, 2020-06-27:
I have a ESP32-S2-Saola-1 and I am compiling with the board set for ESP32-S2 DEV MODULE. 
It compiles, upload, and works as expected. It uses esptool.py v3.0-dev

What doesn't work is trying to use the tool in Arduino to upload into SPIFFS. 
From what I have the upload tool uses esptool.exe version 2.6.1
That version does not support the ESP32-S2 and I believe it is misinterpreting the device to be a  esp32 pico D4.

I found the esptool.py v3.0-dev  but it is a python file. The Arduino uploader for Windows 10 is an .exe file. 
I had tried to convert the .py to an executable but I got errors. I know next to nothing about Python and the tools.

Basically I can't upload to SPIFFS. I have been working during my vacation to try and get some kind of a demo done using the ESP32-S2 so that I can get it to be the processor for a new product project a different engineer is working on at my work. I'm not a programmer. But I trying to do what I can to steer the project toward the Espressif part.

chegewara, 2020-06-27:
Maybe @atanisoft can help, i know he recently did some flashing to spiffs on S2.

atanisoft, 2020-06-27:
I don't use that tool unfortunately.  It is possible the tool is picking the wrong upload tool in it's automation.

lbernstone, 2020-06-27:
If you just want to make it work, use the command line to upload.
`python esptool.py --port=COM3 --baud=961200 write_flash 0x290000 <tmp>/arduino_build/<sketch_name>.spiffs.bin`
This assumes you are using the default partition scheme.  I don't know where those files end up in windows.

RudyFiero, 2020-06-27:
I'll give that a try. 
Thanks.

closed 2020-09-10 by stale[bot]