espressif/arduino-esp32 #4121: Librairy conflict WiFi.h when include MySQL_Connection

Vincentimetr, 2020-06-27:
### Hardware:
Board:							ESP32 Dev kitV1
Core Installation version:			1.0.4
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                      no
Upload Speed:					921600
Computer OS:                                         Windows 10

### Description:
Hi,
I try to connect my esp32 Devkit V1 to a database hosted on my raspberry PI.

I use [ESP_WiFiManager](https://github.com/khoih-prog/ESP_WiFiManager) to connect the esp32 on the local network.

This is working fine!

But at the moment I try to include [MySQL_Connection](https://github.com/ChuckBell/MySQL_Connector_Arduino/), an error occured!

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```
// WIFI LIBRAIRIES
#include <esp_wifi.h>
#include <WiFi.h>
#include <WiFiClient.h>
#include <ESP_WiFiManager.h>              //https://github.com/khoih-prog/ESP_WiFiManager

// MYSQL LIBRAIRIES
// #include <MySQL_Connection.h>
// #include <MySQL_Cursor.h>

void setup(){
  Serial.begin(115200);

  Serial.println("----- Wifi START -----");
  ESP_WiFiManager ESP_wifiManager("AutoConnectAP");
  ESP_wifiManager.setDebugOutput(true);
  ESP_wifiManager.setAPStaticIPConfig(IPAddress(192, 168, 100, 1), IPAddress(192, 168, 100, 1), IPAddress(255, 255, 255, 0));
  ESP_wifiManager.setMinimumSignalQuality(-1);
  ESP_wifiManager.setSTAStaticIPConfig(IPAddress(192, 168, 2, 114), IPAddress(192, 168, 2, 1), IPAddress(255, 255, 255, 0), IPAddress(192, 168, 2, 1), IPAddress(8, 8, 8, 8));
  String Router_SSID = ESP_wifiManager.WiFi_SSID();
  String Router_Pass = ESP_wifiManager.WiFi_Pass();
  ESP_wifiManager.setConfigPortalTimeout(1);
  String chipID = String((uint32_t)ESP.getEfuseMac(), HEX);
  chipID.toUpperCase();
  Serial.println("Router_SSID stored : " + Router_SSID);
  Serial.println("Router_Pass stored : " + Router_Pass);
  Serial.println("chipID : " + chipID);
  String AP_SSID = "ESP32_" + chipID;
  String AP_PASS = chipID;
  Serial.println("AutoConnect...");
  ESP_wifiManager.autoConnect(AP_SSID.c_str(), AP_PASS.c_str());
  Serial.println("Router_SSID : " + Router_SSID);
  Serial.println("Router_Pass : " + Router_Pass);
  Serial.print("WiFi.status()");Serial.println(WiFi.status());
  Serial.println("----- Wifi END -----");

  Serial.println("----- MYSQL START -----");
  //WiFiClient client;
  //MySQL_Connection conn((Client *)&client);
  Serial.println("----- MYSQL END -----");
}

void loop(){
  delay(1000);
  Serial.print("WiFi.status()");Serial.println(WiFi.status());
}
```

### Debug Messages:
```
C:\Users\vince\Documents\Arduino\libraries\MySQL_Connector_Arduino\src\MySQL_Encrypt_Sha1.cpp: In member function 'virtual size_t Encrypt_SHA1::write(uint8_t)':
C:\Users\vince\Documents\Arduino\libraries\MySQL_Connector_Arduino\src\MySQL_Encrypt_Sha1.cpp:89:1: error: no return statement in function returning non-void [-Werror=return-type]
 }
 ^
C:\Users\vince\Documents\Arduino\libraries\MySQL_Connector_Arduino\src\MySQL_Encrypt_Sha1.cpp: In member function 'virtual size_t Encrypt_SHA1::write(uint8_t*, int)':
C:\Users\vince\Documents\Arduino\libraries\MySQL_Connector_Arduino\src\MySQL_Encrypt_Sha1.cpp:95:1: error: no return statement in function returning non-void [-Werror=return-type]
 }
 ^
cc1plus.exe: some warnings being treated as errors
C:\Users\vince\Documents\Arduino\libraries\MySQL_Connector_Arduino\src\MySQL_Packet.cpp: In member function 'int MySQL_Packet::read_lcb_int(int)':
C:\Users\vince\Documents\Arduino\libraries\MySQL_Connector_Arduino\src\MySQL_Packet.cpp:523:7: warning: unused variable 'size' [-Wunused-variable]
   int size = 0;
       ^
Multiple libraries were found for "WiFi.h"
 Used: C:\Users\vince\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi
 Not used: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.39.0_x86__mdqgnx93n4wtt\libraries\WiFi
```

lbernstone, 2020-06-27:
We don't support 3rd party libraries.  The errors are the lines that say "error" in them.

Vincentimetr, 2020-06-27:
Hi @lbernstone 
The problem is a librairy conflict between official arduino librairy and the WiFi librairy of esp32. The librairy MYSQL connector just raise the problem.

closed 2020-06-27 by Vincentimetr

programmerShourya, 2020-07-11:
very simple just go to pefrences > arduino.15.text in that box > go to stagging when clicking on arduino 15 > go to libraries in staging > search for wifi that will be at end only > now delete the file starting with wifi101 or wifi .SO simple