espressif/arduino-esp32 #4122: ArduinoOTA is failing with W5500 Ethernet module

Miq1, 2020-06-28:
### Hardware:
Board:							ESP32 WROOM DevkitC V4
Core Installation version:			1.12.4
IDE name:						PlatformIO
Flash Frequency:					?
PSRAM enabled:                                    ?
Upload Speed:					n/a
Computer OS:                                   Windows 10

### Description:
I am trying to get [ArduinoOTA.h](https://github.com/jandrassy/ArduinoOTA) working over Ethernet without success. I have a W5500 module connected to the ESP32 and successfully am using that to do several TCP and UDP communication with the ESP. I installed ArduinoOTA from within PlatformIO and put in the required lines
At the very beginning:
```// OTA update interface
#include <ArduinoOTA.h>
```
In setup():
``` // Start ArduinoOTA service
  ArduinoOTA.begin(lIP, "myBridge", "xxxx", InternalStorage);
```
In loop():
```  // Check for OTA updates
  ArduinoOTA.poll();
```

My platformio.ini has:
```upload_protocol = espota
upload_port = "192.168.178.77"
upload_flags = -p 3232
upload_flags = --auth=xxxx
```

When trying to upload a firmware, I get the following output, though:
```esptool.py v2.6
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   5.3% (used 28372 bytes from 532480 bytes)
Flash: [===       ]  30.6% (used 400518 bytes from 1310720 bytes)
Configuring upload protocol...
AVAILABLE: esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = espota
Uploading .pio\build\az-delivery-devkit-v4\firmware.bin
18:11:20 [DEBUG]: Options: {'timeout': 10, 'esp_ip': '192.168.178.77', 'host_port': 41297, 'image': '.pio\\build\\az-delivery-devkit-v4\\firmware.bin', 'host_ip': '0.0.0.0', 'auth': 'xxxx', 'esp_port': 3232, 'spiffs': False, 'debug': True, 'progress': True}
18:11:20 [INFO]: Starting on 0.0.0.0:41297
18:11:20 [INFO]: Upload size: 400640
Sending invitation to 192.168.178.77 ..........
18:13:00 [ERROR]: No response from the ESP
*** [upload] Error 1
```
For me it looks like the OTA packets are sent to the W5500 okay, but are never worked on in the ESP.

I tried to enable debug messages with ``build_flags = -DCORE_DEBUG_LEVEL=4`` in platformio.ini, but that resulted in no additional data being put out.

Miq1, 2020-07-04:
I am doubting someone is interested at all, but anyway...

There seems to be another issue with ArduinoOTA/W5500. After having run the ESP32 for a day or so, espota.py is unable to trigger an OTA process on the ESP32 anymore. Only a reset of the ESP32 will solve the issue temporarily.
Other TCP and UDP connections are working regardless, so this is confined to the ArduinoOTA communication.

TheNetStriker, 2020-08-24:
@Miq1 I've just tested your code changes with the Adafruit Ethernet FeatherWing and it works perfectly. It would be nice to have this feature in the official library.

Miq1, 2020-08-26:
Thanks for the confirmation. For all that it is worth I created a PR for it: [#4298](https://github.com/espressif/arduino-esp32/pull/4298)

Miq1, 2020-10-25:
No one is doing anything about this, so I am giving it up. 😠

closed 2020-11-09 by stale[bot]

BobLynas, 2021-03-12:
HI Miq1, I just came across this post, and I am very interested in this solution, to be honest I didn't think ESP32/W5500/OTE was possible, but now i'm going to give it a go, thanks

Miq1, 2021-03-13:
Hi Bob, 

the patch is really simple (as is described in the first post of this issue). You also may look at the commit I mode: [in my forked repository](https://github.com/Miq1/arduino-esp32/commit/feff52b86eb71df0668aefaebc67ab02efd414d3). But please note that the master repository had quite some commits since. Plus: I left in the ``mDNS`` stuff, knowing it will not work via EThernet right away. Using the IP address of your ESP will do.