espressif/arduino-esp32 #4125: BLE dependencies seem to be included when using WiFi Provisioning

jwktje, 2020-06-29:
### Hardware:
Board:							Wemos Lolin32
Core Installation version:			Git master
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                    no
Upload Speed:					115200
Computer OS:                                        Mac OSX

### Description:
When running the example for WiFi provisioning I noticed I had to change the partitions to something non-default to even allow the sketch to fit. This surprised me as I had only experienced this before when working with BLE.

However I wanted to use the SoftAP transport method for provisioning. I modified the [example sketch ](https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiProv/WiFiProv.ino)for this. I will put my code below.

### Expected behaviour:
When I use SoftAP and don't explicitly use the BLE transport for provisioning the compiled binary shouldn't be so massive as it currently is.

### Actual behaviour:
Just excluding the "WiFi.beginProvision" excludes all dependencies it seems.
But as soon as this line is uncommented he Sketch jumps from 724.780 to 1.472.272 bytes.

### Sketch:
```
#include "WiFi.h"

void SysProvEvent(system_event_t *sys_event,wifi_prov_event_t *prov_event)
{
    if(sys_event) {
      switch (sys_event->event_id) {
      case SYSTEM_EVENT_STA_GOT_IP:
          Serial.print("\nConnected IP address : ");
          Serial.println(ip4addr_ntoa(&sys_event->event_info.got_ip.ip_info.ip));
          break;
      case SYSTEM_EVENT_STA_DISCONNECTED:
          Serial.println("\nDisconnected. Connecting to the AP again... ");
          break;
      default:
          break;
      }      
    }

    if(prov_event) {
        switch (prov_event->event) {
        case WIFI_PROV_START:
            Serial.println("\nProvisioning started\nGive Credentials of your access point using \" Android app \"");
            break;
        case WIFI_PROV_CRED_RECV: { 
            Serial.println("\nReceived Wi-Fi credentials");
            wifi_sta_config_t *wifi_sta_cfg = (wifi_sta_config_t *)prov_event->event_data;
            Serial.print("\tSSID : ");
            Serial.println((const char *) wifi_sta_cfg->ssid);
            Serial.print("\tPassword : ");
            Serial.println((char const *) wifi_sta_cfg->password);
            break;
        }
        case WIFI_PROV_CRED_FAIL: { 
            wifi_prov_sta_fail_reason_t *reason = (wifi_prov_sta_fail_reason_t *)prov_event->event_data;
            Serial.println("\nProvisioning failed!\nPlease reset to factory and retry provisioning\n");
            if(*reason == WIFI_PROV_STA_AUTH_ERROR) 
                Serial.println("\nWi-Fi AP password incorrect");
            else
                Serial.println("\nWi-Fi AP not found....Add API \" nvs_flash_erase() \" before beginProvision()");        
            break;
        }
        case WIFI_PROV_CRED_SUCCESS:
            Serial.println("\nProvisioning Successful");
            break;
        case WIFI_PROV_END:
            Serial.println("\nProvisioning Ends");
            break;
        default:
            break;
        }      
    }
}

void setup() {
  Serial.begin(115200);
  
  //Delete saved credentials every time for debugging reasons
  WiFi.enableSTA(true); // must be sta to disconnect erase
  WiFi.disconnect(true,true);
  
  WiFi.onEvent(SysProvEvent);
  WiFi.beginProvision(WIFI_PROV_SCHEME_SOFTAP, WIFI_PROV_EVENT_HANDLER_NONE, WIFI_PROV_SECURITY_1, "abcd1234", "PROV_ARDUINO_TEST");
}

void loop() {
  
}
```

jwktje, 2020-06-30:
Additionally I've found that after provisioning the Info messages suggest that the device will connect to the saved credentials. This doesn't actually occur however. I consider this a bug. Is it possible to make it actually start to connect to the saved credentials upon calling beginProvision?

The line that logs this [is here](https://github.com/espressif/arduino-esp32/blob/4d118b36a29684e65425716632110b23c34ee85a/libraries/WiFi/src/WiFiProv.cpp#L153)

Below is the way I had to implement it in Arduino to actually have it conditionally connect or provision; 
```
void provision_reset() {
  WiFi.enableSTA(true); // must be sta to disconnect erase
  WiFi.disconnect(true,true);

  delay(1000);
  
  Serial.println("Reset the thing");
}

void provision_start() {
  WiFi.onEvent(SysProvEvent);
  WiFi.beginProvision(WIFI_PROV_SCHEME_SOFTAP, WIFI_PROV_EVENT_HANDLER_NONE, WIFI_PROV_SECURITY_1, "abcd1234", "SETUP_MORNING_WINNER");
}

void provision_init() {
  int con_status = WL_IDLE_STATUS;
  int retryCount = 0;
  while ( con_status != WL_CONNECTED) { // attempt to connect to Wifi network:
    if(retryCount < 6) {
      Serial.println("Attempting to connect to Network");
      con_status = WiFi.begin();
      delay(10000);// wait 10 seconds for connection:
      retryCount++;
    } else {
      Serial.println("Retry limit reached! Connection failed. Resetting");
      provision_reset();
      provision_start();
    }
    
  }
  Serial.println("You're connected to the network");
  Serial.println(WiFi.localIP());
}
```

lbernstone, 2020-07-01:
I don't know anything about the provisioning stuff, but when you connect to an AP, it should save your configuration.  You can do something like this in your setup to check:
```
#include <WiFi.h>
#include <esp_wifi.h>

void setup() {
  Serial.begin(115200);
  WiFi.mode(WIFI_MODE_STA);
  wifi_config_t conf;
  esp_wifi_get_config(WIFI_IF_STA, &conf);
  String saved_ssid = String(reinterpret_cast<char*>(conf.sta.ssid));
  if(saved_ssid=="") {
    Serial.println("No saved SSID");
    // do provisioning
  } else {
    Serial.printf("Saved SSID: %s\n", saved_ssid);
    WiFi.begin();
  }
}
```

sweetymhaiske, 2020-08-24:
@jwktje Few fixes are made for BLE dependencies. While using SoftAP it won't be pulled.

jwktje, 2020-09-03:
@sweetymhaiske I had time to test it today and this seems to fix the size issue! The compiled size is now way lower. I will still have to functionally test it later, but I will close this issue for now. If anything comes up I will reopen.

Thanks you and @me-no-dev for the help on this!

closed 2020-09-03 by jwktje