espressif/arduino-esp32 #4127: BLE esp32 sending data

baamiis, 2020-07-01:
### Hardware:
Board:							ESP32 Dev Module?
Core Installation version:			1.4
IDE name:						Arduino IDE
Flash Frequency:					?40Mhz?
PSRAM enabled:                                    no
Upload Speed:					115200
Computer OS:                                        Windows 10

### Description:
this is used from the idf examples the program hangs waiting for retrieve the service attributes indefinitly so I turned debug on and the output from debug is included: 

Can someone please help me resolve this the code bellow, thank yoy :)

###the code:
#define TARGET_SERVICE  "6E400001-B5A3-F393-E0A9-E50E24DCCA9E" // UART service UUID
#define TARGERT_CHARAC  "6E400003-B5A3-F393-E0A9-E50E24DCCA9E"
class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
    void onResult(BLEAdvertisedDevice advertisedDevice) {
      
      Serial.printf("Advertised Device: %s \n", advertisedDevice.toString().c_str());
      if(String(advertisedDevice.toString().c_str()).indexOf("dd:5b:80:6a:c9:06") >= 0) {
        Serial.println(" - FOUND Address:");
        
        pServerAddress = new BLEAddress("dd:5b:80:6a:c9:06");
        MicrobitAlive = true;
 
      }

    }
};

void setup() {
  Serial.begin(115200);
  Serial.println("Scanning...");

  BLEDevice::init("");
  pBLEScan = BLEDevice::getScan(); //create new scan
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
  pBLEScan->setActiveScan(true); //active scan uses more power, but get results faster
  pBLEScan->setInterval(100);
  pBLEScan->setWindow(99);  // less or equal setInterval value
}

void loop() {
  if(MicrobitAlive == true) {
        Serial.print("Forming a connection to ");
        BLEClient*  pClient  = BLEDevice::createClient();
        pClient->connect(*pServerAddress);
        Serial.println(" - Connected to server");
        // Obtain a reference to the service we are after in the remote BLE server.
        BLERemoteService* pRemoteService = pClient->getService(TARGET_SERVICE);
        if (pRemoteService == nullptr) {
          Serial.print("Failed to find our service UUID: ");
          Serial.println(TARGET_SERVICE);
        }
        Serial.println(" - Found our service");
        // Obtain a reference to the characteristic in the service of the remote BLE server.
        pRemoteCharacteristic = pRemoteService->getCharacteristic(TARGERT_CHARAC);
        if (pRemoteCharacteristic == nullptr) {
          Serial.print("Failed to find our characteristic UUID: ");
          Serial.println(TARGERT_CHARAC);
        }
        MicrobitAlive = false;
  }
  else {
      // put your main code here, to run repeatedly:
      BLEScanResults foundDevices = pBLEScan->start(scanTime, false);
      Serial.print("Devices found: ");
      Serial.println(foundDevices.getCount());
[... 146 lines cut ...]
[D][BLEScan.cpp:97] handleGAPEvent(): Ignoring dd:5b:80:6a:c9:06, already seen it.
[D][BLEScan.cpp:97] handleGAPEvent(): Ignoring dd:5b:80:6a:c9:06, already seen it.
[D][BLEScan.cpp:97] handleGAPEvent(): Ignoring 2a:5a:db:2e:6f:f5, already seen it.
[D][BLEScan.cpp:97] handleGAPEvent(): Ignoring dd:5b:80:6a:c9:06, already seen it.
[D][BLEScan.cpp:97] handleGAPEvent(): Ignoring dd:5b:80:6a:c9:06, already seen it.
[D][BLEScan.cpp:97] handleGAPEvent(): Ignoring 2a:5a:db:2e:6f:f5, already seen it.
[D][BLEScan.cpp:97] handleGAPEvent(): Ignoring dd:5b:80:6a:c9:06, already seen it.
[D][BLEScan.cpp:97] handleGAPEvent(): Ignoring dd:5b:80:6a:c9:06, already seen it.
[D][BLEScan.cpp:97] handleGAPEvent(): Ignoring 2a:5a:db:2e:6f:f5, already seen it.
[D][BLEScan.cpp:97] handleGAPEvent(): Ignoring dd:5b:80:6a:c9:06, already seen it.
[D][BLEScan.cpp:97] handleGAPEvent(): Ignoring dd:5b:80:6a:c9:06, already seen it.
[D][BLEScan.cpp:97] handleGAPEvent(): Ignoring 2a:5a:db:2e:6f:f5, already seen it.
[W][BLEScan.cpp:69] handleGAPEvent(): ESP_GAP_SEARCH_INQ_CMPL_EVT
Devices found: 2
Scan done!
Forming a connection to [I][BLEDevice.cpp:614] addPeerDevice(): add conn_id: 0, GATT role: client
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:158] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:158] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
 - Connected to server
[I][BLEDevice.cpp:625] removePeerDevice(): remove: 0, GATT role client
[E][BLEUUID.cpp:116] BLEUUID(): ERROR: UUID value not 2, 4, 16 or 36 bytes
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown

chegewara, 2020-07-01:
2 problems:
1) `[I][BLEDevice.cpp:625] removePeerDevice(): remove: 0, GATT role client` <--- peer device is disconnected here
2) `[E][BLEUUID.cpp:116] BLEUUID(): ERROR: UUID value not 2, 4, 16 or 36 bytes`

baamiis, 2020-07-01:
Any idea as to why apart from the disconnection ??

chegewara, 2020-07-01:
No, no idea. There is not enough info.

baamiis, 2020-07-01:
I don’t understand What more info you need :)
You have the code and you have it’s output.

chegewara, 2020-07-01:
Its very simple, neither code nor logs explain disconnection reason. Or i am just dumb, that could explain.

h2zero, 2020-07-02:
> this is used from the idf examples the program hangs waiting for retrieve the service attributes indefinitly so I turned debug on and the output from debug is included

Are you using the esp32-arduino master branch? If so checkout v1.0.4 instead and try it. Master is using a bugged BLE controller.

baamiis, 2020-07-02:
You are not dump. The device is not disconnecting. And I just cannot see it disconnecting. Very strange :)

baamiis, 2020-07-02:
Thanks for your help h2zero. Much appreciated

baamiis, 2020-07-02:
just checked the version. I am using version v1.0.4. If any one has a solution to this please help thanks.

buxtronix, 2020-07-04:
You should check the return value of connect(), it might be false (ie failed to connect). And increase debugging level to Verbose.

closed 2020-09-16 by stale[bot]