espressif/arduino-esp32 #4128: arduino-esp32 as esp-idf component failes: error: 'DPORT_SPI_CLK_EN' undeclared

qt1, 2020-07-01:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do?

I am installing a fresh installation of arduino-esp32 

### Hardware:
Board:							N/A
Core Installation version:			?1.0.0? ?1.0.1-rc4? ?1.0.1? ?1.0.1-git? ?1.0.2? ?1.0.3?
IDE name:						IDF component
Flash Frequency:					40Mhz
PSRAM enabled:                                    no
Upload Speed:					115200
Computer OS:                                         Ubuntu?

### Description:

1. Installed and tested esp-idf stable then

2. Followed According to this guide:
https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md

When running 'make' there are a few warnings then some errors e.g.:
```
In file included from /home/baruch/esp2/esp-idf/components/soc/esp32/include/soc/dport_reg.h:20,
                 from /home/baruch/esp2/test/components/arduino/cores/esp32/esp32-hal-spi.c:28:
/home/baruch/esp2/test/components/arduino/cores/esp32/esp32-hal-spi.c: In function 'spiStartBus':
/home/baruch/esp2/test/components/arduino/cores/esp32/esp32-hal-spi.c:431:57: error: 'DPORT_SPI_CLK_EN' undeclared (first use in this function); did you mean 'DPORT_SPI3_CLK_EN'?
         DPORT_SET_PERI_REG_MASK(DPORT_PERIP_CLK_EN_REG, DPORT_SPI_CLK_EN);
```

It seems like `make `system mixes code from 
```
esp-idf/components
```
and 
```
components/arduino/cores
```

### Sketch:  
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
home/baruch/esp2/test/components/arduino/cores/esp32/libb64/cdecode.c: In function 'base64_decode_block_signed':
/home/baruch/esp2/test/components/arduino/cores/esp32/libb64/cdecode.c:42:23: warning: this statement may fall through [-Wimplicit-fallthrough=]
         *plainchar    = (fragment & 0x03f) << 2;
         ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~
/home/baruch/esp2/test/components/arduino/cores/esp32/libb64/cdecode.c:43:7: note: here
       case step_b:
       ^~~~
/home/baruch/esp2/test/components/arduino/cores/esp32/libb64/cdecode.c:53:23: warning: this statement may fall through [-Wimplicit-fallthrough=]
         *plainchar    = (fragment & 0x00f) << 4;
         ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~
/home/baruch/esp2/test/components/arduino/cores/esp32/libb64/cdecode.c:54:7: note: here
       case step_c:
       ^~~~
/home/baruch/esp2/test/components/arduino/cores/esp32/libb64/cdecode.c:64:23: warning: this statement may fall through [-Wimplicit-fallthrough=]
         *plainchar    = (fragment & 0x003) << 6;
[... 121 lines cut ...]
/home/baruch/esp2/esp-idf/components/soc/esp32/include/soc/dport_access.h:154:88: note: in definition of macro '_DPORT_WRITE_PERI_REG'
 #define _DPORT_WRITE_PERI_REG(addr, val) (*((volatile uint32_t *)(addr))) = (uint32_t)(val)
                                                                                        ^~~
/home/baruch/esp2/esp-idf/components/soc/esp32/include/soc/dport_access.h:183:46: note: in expansion of macro 'DPORT_WRITE_PERI_REG'
 #define DPORT_SET_PERI_REG_MASK(reg, mask)   DPORT_WRITE_PERI_REG((reg), (DPORT_READ_PERI_REG(reg)|(mask)))
                                              ^~~~~~~~~~~~~~~~~~~~
/home/baruch/esp2/test/components/arduino/cores/esp32/esp32-hal-spi.c:437:9: note: in expansion of macro 'DPORT_SET_PERI_REG_MASK'
         DPORT_SET_PERI_REG_MASK(DPORT_PERIP_CLK_EN_REG, DPORT_SPI_CLK_EN_1);
         ^~~~~~~~~~~~~~~~~~~~~~~
/home/baruch/esp2/test/components/arduino/cores/esp32/esp32-hal-spi.c:438:59: error: 'DPORT_SPI_RST_1' undeclared (first use in this function); did you mean 'DPORT_SPI2_RST'?
         DPORT_CLEAR_PERI_REG_MASK(DPORT_PERIP_RST_EN_REG, DPORT_SPI_RST_1);
                                                           ^~~~~~~~~~~~~~~
/home/baruch/esp2/esp-idf/components/soc/esp32/include/soc/dport_access.h:154:88: note: in definition of macro '_DPORT_WRITE_PERI_REG'
 #define _DPORT_WRITE_PERI_REG(addr, val) (*((volatile uint32_t *)(addr))) = (uint32_t)(val)
                                                                                        ^~~
/home/baruch/esp2/esp-idf/components/soc/esp32/include/soc/dport_access.h:180:46: note: in expansion of macro 'DPORT_WRITE_PERI_REG'
 #define DPORT_CLEAR_PERI_REG_MASK(reg, mask) DPORT_WRITE_PERI_REG((reg), (DPORT_READ_PERI_REG(reg)&(~(mask))))
                                              ^~~~~~~~~~~~~~~~~~~~
/home/baruch/esp2/test/components/arduino/cores/esp32/esp32-hal-spi.c:438:9: note: in expansion of macro 'DPORT_CLEAR_PERI_REG_MASK'
         DPORT_CLEAR_PERI_REG_MASK(DPORT_PERIP_RST_EN_REG, DPORT_SPI_RST_1);
         ^~~~~~~~~~~~~~~~~~~~~~~~~
make[1]: *** [/home/baruch/esp2/esp-idf/make/component_wrapper.mk:292: cores/esp32/esp32-hal-spi.o] Error 1
make: *** [/home/baruch/esp2/esp-idf/make/project.mk:590: component-arduino-build] Error 2

```

qt1, 2020-07-04:
Carefully matching the versions and reinstallation of esp-idf seems to resolve this.

closed 2020-07-04 by qt1

venkateshtata, 2020-11-17:
Hey @qt1 , got into the same issue. Can you please tell me a bit more insight into how you solved it ? would be a great help. Thank you

qt1, 2020-11-17:
Hello @venkateshtata , As far as I remember it was incompatible esp-idf and esp-arduino revisions.

You should go to the esp-arduino git tree and look for the recent esp-idf they used. Then you must make sure the esp-idf is the exact same revision.

Good luck :)

venkateshtata, 2020-11-17:
Thanks for the reply, I got it ! @qt1

qt1, 2020-11-20:
Here is a little bash script that display the revisions of arduino-esp32 and esp-idf, to make sure they are compatible.
This can probably be added to the environment setup script 

    #!/bin/bash
    git -C $IDF_PATH rev-parse --short HEAD
    awk  '/CONFIG_ARDUINO_IDF_COMMIT/{gsub("\"","");print $3}' < components/arduino/tools/sdk/esp32/include/config /sdkconfig.h

gb-123-git, 2021-09-21:
> Thanks for the reply, I got it to work ! @qt1

Could you tell me which versions are you matching ? I am getting the same error