espressif/arduino-esp32 #4130: Esp32 crashes during javascript interpreter init

curly-brace, 2020-07-02:
### Hardware:
Board:							lolin32
Core Installation version:			ARDUINO_ESP32_GIT_VER 0x46386288 :: ARDUINO_ESP32_RELEASE_1_0_4
IDE name:						Platform.io (platform version 1.12.4)
Flash Frequency:					80Mhz
PSRAM enabled:                                    no
Upload Speed:					115200
Computer OS:                                        Windows 10

### Description:
I don't have enough skills to debug this issue, but I'm almost certain that there is some sort of problem in esp32 memory management,

I'm implementing js scriptable device with gui. I've already tried using two js engines and both start to fail at some point. And this point depends on amount js functions, that I've added to them. Esp32 just aborts silently giving stack trace inside the guts of js engines. And at this time seems like js engines allocate lots of memory chunks for internal objects.

The heap is reported to be free a lot over 200kb and largest single alloc reports to be over 100kb.

I'm sure that compiled js code and objects and functions, added to engine, should fit fine into the heap.

I just don't get it.

The largest heap usage I've got without crashing the esp32 is around 234kb, which is minimum, but while my js app runs, it already reports 275kb of free heap, so it means that more memory was used during the compilation, but it was freed after it.

And if I add just a single more js function to the engine, everything crashes.

I've tried duktape and v7 engines. V7 worked much better somehow. But still both fail. I could try one more engine, but my codebase is already too large to just quickly change js engines.

curly-brace, 2020-07-02:
I believe this issue can be illustrated with this code in setup function
```
    int* p[7000];    
    for (int i=0; i<7000; ++i) {
        p[i] = (int*)malloc(10*sizeof(int));
        for (int j=0; j<10; ++j) {
            p[i][j] = 0;
        }
    }
```
It just allocates 70k, but is crashed in guru meditation. It is different error, in my case esp32 crashes silently, but I think it is related, because there is plenty of usable RAM in blank sketch, but it failes to allocate 70k.

Btw if allocated as single malloc, it works fine.

curly-brace, 2020-07-02:
I believe this code illustrates the issue, because it crashes the esp32:
```
setup() {
    byte* p[8000];    
    for (int i=0; i<8000; ++i) {
        p[i] = (byte*)malloc(10);
        for (int j=0; j<20; ++j) {
            p[i][j] = 0;
        }
    }
}
```
CORRUPT HEAP: multi_heap.c:432 detected at 0x3ffb844c
abort() was called at PC 0x4008abeb on core 1
Backtrace: 0x4008b580:0x3ffb1e80 0x4008b7ad:0x3ffb1ea0 0x4008abeb:0x3ffb1ec0 0x4008b1ce:0x3ffb1ee0 0x40084884:0x3ffb1f00 0x400848b9:0x3ffb1f20 0x40084d71:0x3ffb1f40 0x4000beaf:0x3ffb1f60 0x400d0f4e:0x3ffb1f80 0x400d2f7f:0x3ffb1fb0 0x40088235:0x3ffb1fd0

0x4008b580: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
0x4008b7ad: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
0x4008abeb: multi_heap_assert at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_platform.h:54
0x4008b1ce: multi_heap_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c:301
0x40084884: heap_caps_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:111
0x400848b9: heap_caps_malloc_default at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:140
0x40084d71: _malloc_r at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/syscalls.c:37
0x4000beaf: ?? ??:0
0x400d0f4e: _Z5setupv at D:\sources\PDA/src/main.cpp:228
0x400d2f7f: _Z8loopTaskPv at C:\users\waylander\.platformio\packages\framework-arduinoespressif32\cores\esp32/main.cpp:14
0x40088235: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

chegewara, 2020-07-02:
looper task stack, which is running setup and loop functions is only 8kB, thats why you get heap corruption:
`byte* p[8000]; `

curly-brace, 2020-07-02:
oh, my bad. there was also a typo. but this code also fails.
```
setup() {
    byte** p = (byte**)malloc(10000*sizeof(byte*));
    for (int i=0; i<10000; ++i) {
        p[i] = (byte*)malloc(10);
        for (int j=0; j<10; ++j) {
            p[i][j] = 0;
        }
    }
}
```

curly-brace, 2020-07-02:
And even better code:
```
setup() {
    byte** p = (byte**)malloc(9030*sizeof(byte*));
    for (int i=0; i<9030; ++i) {
        p[i] = (byte*)malloc(10);
        for (int j=0; j<10; ++j) {
            p[i][j] = 0;
        }
    }
}
```
with 9030 it works.
When it finishes, freeHeap is 73428 and max alloc is 73404. If I decrease the iterations, max alloc stays the same 73404. And if I run it with 9031, it crashes. So free heap goes below max alloc, while allocating another 10 bytes (+4 for arduino heap info).

chegewara, 2020-07-02:
Try this code first:
`Serial.println(10000*sizeof(byte*))`

curly-brace, 2020-07-02:
it outputs 40000. no surprises. you mean that there is memory needed for an array of pointers? but there is still plenty of heap free, but esp crashes.

chegewara, 2020-07-02:
There is 2 strange things:
1) why it crash when there is still plenty of heap
2) why it is taking 28 bytes to `malloc(10)` 
```
7321
12:57:42.838 ->     free 124624 allocated 187544 min_free 124604 largest_free_block 78576
12:57:42.838 -> 7322
12:57:43.004 ->     free 124596 allocated 187556 min_free 124576 largest_free_block 78576
12:57:43.004 -> 7323
12:57:43.169 ->     free 124568 allocated 187568 min_free 124548 largest_free_block 78576
12:57:43.169 -> 7324
12:57:43.302 ->     free 124540 allocated 187580 min_free 124520 largest_free_block 78576
12:57:43.302 -> 7325
12:57:43.467 ->     free 124512 allocated 187592 min_free 124492 largest_free_block 78576
12:57:43.467 -> 7326
12:57:43.600 ->     free 124484 allocated 187604 min_free 124464 largest_free_block 78576
```
```
p[i] = (byte*)malloc(10);
        
        for (int j=0; j<10; ++j) {
            p[i][j] = (byte)0;
        }
```

chegewara, 2020-07-02:
This code is causing crash:
```
for (int j=0; j<10; ++j) {
            p[i][j] = (byte)0;
        }
```
Without it run ok.

curly-brace, 2020-07-02:
yeah, i've also noticed that memset causes no errors. it is the actual write to the memory that causes it.

curly-brace, 2020-07-02:
here is similar issue, that was closed due to inactivity
https://github.com/espressif/arduino-esp32/issues/1163

chegewara, 2020-07-02:
And here is the answer:
```
13:27:50.555 -> 8 BIT
13:27:50.555 -> Heap summary for capabilities 0x00000004:
13:27:50.687 ->   Totals:
13:27:50.687 ->     free 251008 allocated 99664 min_free 245328 largest_free_block 113792
13:27:50.820 -> 8 BIT after 5000 mallocs
13:27:50.820 -> Heap summary for capabilities 0x00000004:
13:27:50.952 ->   Totals:
13:27:50.952 ->     free 111048 allocated 159680 min_free 111048 largest_free_block 78316
```

You have to check memory with this CAPS:
`heap_caps_print_heap_info(MALLOC_CAP_8BIT);`

chegewara, 2020-07-02:
I forgot about it, but Angus explained similar issue on forum.

curly-brace, 2020-07-02:
i've reading this now. and indeed xPortGetFreeHeapSize() reports that 24 bytes are available after my test code.
but what i don't understand, why works this:
```
    byte* p = (byte*)malloc(200000);
    for (int i=0; i<200000; ++i) {
        *p = 0;
    }
```
and why i can't use whole heap for my app?? i really need this memory for js to work. but i can't use without rewriting the source code of the engine?

chegewara, 2020-07-02:
Here you are allocating only 20000 bytes:
```
    byte* p = (byte*)malloc(200000);
    for (int i=0; i<200000; ++i) {
        *p = 0;
    }
```
The other code is allocating 40000 bytes and every iteration is allocating another bytes.
You can use this code to see how much bytes is allocated and where:
`heap_caps_print_heap_info(MALLOC_CAP_8BIT);`
> and why i can't use whole heap for my app?? 

You can use whole memory, but memory in esp32 has different capabilities. 
Maybe @projectgus can explain it better.

curly-brace, 2020-07-02:
i still don't understand. here is the code that allocates 10k of usable ram
```
    byte* p;
    for (int i=0; i<1000; ++i) {
        p = (byte*)malloc(10);
        for (int j=0; j<10; ++j) {
            p[j] = 0;
        }
    }
```
but according to this heap report function, it actually allocates almost 3x the amount.  - 27992
i understand that there is arduino heap info taking 4 bytes on each allocation, as it is described in forum post. but why 16 bytes are taken??? for what? 4 bytes of arduino heap info are already removed from this btw. if we have a lot of allocations, this build ups pretty fast.

no surprises that js engines run out of memory, while allocating and deallocating a lot of small and medium chunks.

and may be it is not the issue tho, because i've printed xPortGetFreeHeapSize and it showed 176792, before esp crashed. may be i'm looking in the wrong direction?

it just shows this:
abort() was called at PC 0x400e10af on core 1
0x4008bfc4:0x3ffb1b20 0x4008c1f5:0x3ffb1b40 0x400e10af:0x3ffb1b60 0x400e1197:0x3ffb1b90 0x400e121a:0x3ffb1bc0 0x400e12c5:0x3ffb1be0 0x400e131b:0x3ffb1c00 0x400e13c0:0x3ffb1c20 0x400e14ba:0x3ffb1c40 0x400e1767:0x3ffb1cb0 0x400e6445:0x3ffb1d10 0x400d5151:0x3ffb1d70 0x400d549f:0x3ffb1da0 0x400d6efc:0x3ffb1dc0 0x400d7b9e:0x3ffb1e00 0x400d8065:0x3ffb1e50 0x400eed76:0x3ffb1e80 0x400eedbb:0x3ffb1ec0 0x400ee167:0x3ffb1ee0 0x400eec4d:0x3ffb1f20 0x400f85f7:0x3ffb1f70 0x400f86c0:0x3ffb1f90 0x400dab6c:0x3ffb1fb0 0x400889cd:0x3ffb1fd0

0x4008bfc4: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
0x4008c1f5: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
0x400e10d7: gc_mark at D:\sources\PDA/src/v7.c:18225
0x400e11bf: gc_mark at D:\sources\PDA/src/v7.c:18225
0x400e1242: gc_mark_mbuf_pt at D:\sources\PDA/src/v7.c:18225
0x400e12ed: v7_gc at D:\sources\PDA/src/v7.c:18225
0x400e1343: maybe_gc at D:\sources\PDA/src/v7.c:18225
0x400e13e8: new_property at D:\sources\PDA/src/v7.c:18225
0x400e14e2: def_property_v at D:\sources\PDA/src/v7.c:18225
0x400e178f: def_property at D:\sources\PDA/src/v7.c:18225
0x400e646d: set_property at D:\sources\PDA/src/v7.c:18225
0x400d5145: _ZN5JSGUI10WidgetInitEP2v7 at D:\sources\PDA/src/js_gui.cpp:119
0x400d5483: _ZN5JSGUI4InitEP2v7P9_lv_obj_t at D:\sources\PDA/src/js_gui.cpp:17
0x400d6ee0: _ZN9JSWrapper7LoadAppE6StringS0_ at D:\sources\PDA/src/jswrapper.cpp:48
0x400d7bc6: _ZN8Launcher6RunAppE6String at c:\users\waylander\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\bits/vector.tcc:427
0x400d808d: _ZN8Launcher10AppClickedEP9_lv_obj_th at c:\users\waylander\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\bits/vector.tcc:427
0x400eed9e: lv_event_send_func at D:\sources\PDA/lib\lvgl\src\lv_core/lv_obj.c:2856
0x400eede3: lv_event_send at D:\sources\PDA/lib\lvgl\src\lv_core/lv_obj.c:2856
0x400ee18f: indev_proc_press at D:\sources\PDA/lib\lvgl\src\lv_core/lv_indev.c:355
0x400eec75: indev_button_proc at D:\sources\PDA/lib\lvgl\src\lv_core/lv_indev.c:355
0x400f861f: lv_task_exec at D:\sources\PDA/lib\lvgl\src\lv_misc/lv_task.c:331
0x400f8701: lv_task_handler at D:\sources\PDA/lib\lvgl\src\lv_misc/lv_task.c:331
0x400dab94: _ZN13TFTController6LVTaskEPv at D:\sources\PDA/src/tft_controller.cpp:112
0x400889cd: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

projectgus, 2020-07-02:
Hi @curly-brace @chegewara ,

Yes, as @chegewara says the ESP32 has different types of internal memory - IRAM and DRAM. IRAM (instruction RAM) is also added to the heap but it can only be accessed in aligned 32-bit accesses, so it's only useful for limited purposes. Calling the heap_caps APIs with `MALLOC_CAP_8BIT` as the capabilities argument will show the memory in DRAM (data RAM) which can be allocated via `malloc()`.

If you know that some memory will only be accessed as aligned 32-bit words, you can allocate it with `MALLOC_CAP_32BIT` (meaning it allocates from IRAM), instead of using up DRAM.

You can read more about the memory types and heap capabilities here:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/mem_alloc.html#memory-capabilities

> i still don't understand. here is the code that allocates 10k of usable ram
> i understand that there is arduino heap info taking 4 bytes on each allocation, as it is described in forum post.

ESP-IDF has an "heap poisoning" feature. Arduino is built with heap poisoning in the default "light" configuration. This allocates 4 "canary bytes" at the beginning and end of each allocation to help detect heap overflows:
https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/sdkconfig#L616

Full explanation of this feature:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/heap_debug.html#finding-heap-corruption

Also, all allocations are rounded up to the word size so allocating 10 bytes will effectively allocate 12.

So we have 12 (10 rounded up) + 4 (heap metadata) + 8 (canaries) = 24 bytes per allocation.

This only accounts for 24000/27992. Without seeing the full code (including all the debugging code used to measure the free RAM) it's hard to know what else might be allocating memory.

If you switch to building the firmware in ESP-IDF then you can change the poisoning mode to "disabled" and reclaim 8 bytes per allocation.

For a runtime engine like a JS engine that plans to do mostly very small short lived allocations of this kind, it may be better to move to a dedicated memory manager just for these small allocations rather than trying to use a general-purpose heap allocator like the one we have in ESP-IDF.

projectgus, 2020-07-02:
Regarding the crash, what does this line of source code say?

> 0x400e10d7: gc_mark at D:\sources\PDA/src/v7.c:18225

curly-brace, 2020-07-03:
wow! thanks for such definitive answer! now it is more clear to me concerning the memory.

at this line located this function:
```
void v7_interrupt(struct v7 *v7) {
  v7->interrupted = 1;
}
```
which I don't really know where is called, because it exists in engine source only as this function and its declaration. may be linenumbers aren't correct in backtrace?

curly-brace, 2020-07-03:
Here is the project. https://www.dropbox.com/s/ypuikozppctar1q/ESP32_V7_Test.zip?dl=0
It crashes a little bit in different way. In original case it crashed upon adding that Widget_prototype, but now it crashes upon executing. But in both cases the crash can be removed by commenting out one property insertion to js global object.

curly-brace, 2020-07-05:
sh*******t. i've run the test project on stm32 and it failed on the same principle. so it appears there is a bug in the js engine itself. and strangely enough, seems like previous duktape crash is unrelated to this problem.

curly-brace, 2020-07-05:
stupid me. there IS actually a limit for global properties. setting it higher solved the issue. thanks to st-link that i noticed this.

closed 2020-07-05 by curly-brace