espressif/arduino-esp32 #4132: Feature Request: Implementation from ESP32Mesh in Arduino ESP32

kugelkopf123, 2020-07-02:
### Hardware:
Board:							ESP DEV Module
Core Installation/update date:			latest Relase
IDE name:							PlatformIO
Flash Frequency:					40Mhz
PSRAM enabled:                                        no
Upload Speed:						921600
Computer OS:                                            OSX Mojave

### Description:
Its a Question/Feature Request.
I'm a bit of a Tasmota enthusiast and I recently heard that there's a[ MESH function,](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/mesh.html#tree-topology) developed by Espressif, for the ESP32. Apparently this option is already available in the latest IDF. 
Since a few months Tasmota is now also compatible with ESP32. Considering the range of Tasmota and the possibilities this software offers, it would be great if the mesh function would be implemented in arduino. 
This is the prerequisite for the development of this option in Tasmota.
So my question now is: Is it planned to implement this function in Arduino ESP32? If so, when?

Related feature request on [Tasmota Github](https://github.com/arendst/Tasmota/issues/8715). 

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

lbernstone, 2020-07-02:
For WiFi mesh, there is no real need for this in core.  It is available as a module at https://github.com/lbernstone/ESP_WiFiMesh.  Someone will probably make a library for BLE mesh once the core moves to IDF 4.x.

kugelkopf123, 2020-07-07:
> For WiFi mesh, there is no real need for this in core. It is available as a module at https://github.com/lbernstone/ESP_WiFiMesh. Someone will probably make a library for BLE mesh once the core moves to IDF 4.x.

Okay. Is there anything scheduled when the core rise to idf4?

closed 2020-07-07 by kugelkopf123

lbernstone, 2020-07-08:
beta is in the [esp32s2 branch](https://github.com/espressif/arduino-esp32/tree/esp32s2)