espressif/arduino-esp32 #4139: BLE provisioning turns off after failed WiFi authentication

maikeriva, 2020-07-06:
### Hardware
- Arduino IDE 1.8.12
- Fedora 31
- ESP32 WROOM32D

### Description:
Hello!

I am trying to provision my ESP32 device with the new unified provisioning API by using BLE and the dedicated Android application. To do so I am basing my code on the Arduino sketch WiFiProv.ino found in the arduino-esp32 repository.

After selecting the network on the Android application and setting the credentials, in the case the connection attempt fails (eg. when given credentials are wrong) the device cannot be detected anymore from the application home screen. It seems that after the failed attempt the device won't be reachable via BLE anymore, and I can't find a way through the given API to change this behaviour.

Digging a little bit under the hood, I found the function `wifi_prov_mgr_disable_auto_stop()` and I've built a custom 
`WiFiProvClass :: beginBLEProvision(wifi_prov_event_handler_t scheme_event_handler, wifi_prov_security_t security, const char * pop, const char *service_name, const char *service_key, uint8_t * uuid)` method which integrates it (and also which should allow to re-provision the device even when already provisioned, which is another thing I need). You can find the method here below:

```
WiFiProvClass :: beginBLEProvision(wifi_prov_event_handler_t scheme_event_handler, wifi_prov_security_t security, const char * pop, const char *service_name, const char *service_key, uint8_t * uuid) {
    prov_enable = true;
    wifi_prov_mgr_config_t config;
    config.scheme_event_handler = scheme_event_handler;
    config.app_event_handler = {
            .event_cb = prov_event_handler,
            .user_data = NULL
            };
    config.scheme = wifi_prov_scheme_ble;        
 
    if (wifi_prov_mgr_init(config) != ESP_OK) {
        log_e("Could not initialize provisioning manager");
    }

    WiFi.mode(WIFI_MODE_AP);
    service_key = NULL;
    if(uuid == NULL) {
        uuid=(uint8_t *)custom_service_uuid;
    }
    if (wifi_prov_scheme_ble_set_service_uuid(uuid) != ESP_OK) {
        log_e("Could not set provisioning service UUID");
    }

    if(service_name == NULL) {
        char service_name_temp[12];
        get_device_service_name(WIFI_PROV_SCHEME_BLE,service_name_temp,sizeof(service_name_temp));
        service_name = (const char *)service_name_temp;
    }

    log_i("Starting AP using BLE\n service_name : %s\n pop : %s",service_name,pop);
    
    if (wifi_prov_mgr_disable_auto_stop(1000) != ESP_OK) {
        log_e("Could not deactivate auto-stop on provisioning manager");
    }

    if (wifi_prov_mgr_start_provisioning(security,pop,service_name,service_key) != ESP_OK) {
        log_e("Could not start provisioning manager");
    }

    log_i("Provisioning manager started");
}
```
While with this it is possible to start the BLE provisioning operation even with WiFi credentials already present in NVS, it still doesn't solve the fact the device disappears from the Android application after a failed credentials attempt.

In a hacky attempt to solve the problem I event tried to deinit and reinit the WiFi provisioner after WIFI_PROV_CRED_FAIL events in `WiFiProv.cpp :: prov_event_handler()`, yet the firmware crashes and the backtrace fails to be decoded by EspExceptionDecoder.

Maybe the arduino-esp32 devs have an idea on how this could be accomplished? Is it an inherent limitation of the provisioning manager?

### Sketch:
WiFiProv.ino
```
#include "WiFi.h"
void SysProvEvent(system_event_t *sys_event,wifi_prov_event_t *prov_event)
{
[... 297 lines cut ...]
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 14 - AP_START
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START

Provisioning started
Give Credentials of your access point using " Android app "
[I][WiFiProv.cpp:212] beginBLEProvision(): Provisioning manager started
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 1 - SCAN_DONE

Received Wi-Fi credentials
	SSID : FREELANCERWIFI
	Password : fbbsjjkslofi
[I][WiFiProv.cpp:60] prov_event_handler(): WIFI_PROV_CRED_FAIL: Restarting provisioner
E (51814) wifi_prov_scheme_ble: bt_mem_release of classic BT failed 259
[I][WiFiProv.cpp:202] beginBLEProvision(): Starting AP using BLE
 service_name : BLE_XXX
 pop : abcd1234
aborCtO(R)R UwPaTs  HcEaAlPl:e d at head 40 8x3f5 onec. Expe

ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000

Backtrace: 0x4008ef74:0x3ffb2ef0 0x4008f1f1:0x3ffb2f10 0x40083155:0x3ffb2f30 0x40083279:0x3ffb2f60 0x400e060f:0x3ffb2f80 0x400dac11:0x3ffb3240 0x400d5048:0x3ffb3290 0x400951d9:0x3ffb32c0 0x400972c7:0x3ffb32f0 0x40096d73:0x3ffb3310 0x40083731:0x3ffb3330 0x4014466f:0x3ffb3350 0x401c4e21:0x3ffb3370 0x40144a53:0x3ffb33b0 0x401b7fdb:0x3ffb33d0 0x401b7bb5:0x3ffb3420 0x400ea35f:0x3ffb3470 0x400e80b1:0x3ffb3490 0x400e9370:0x3ffb34b0 0x400d2013:0x3ffb35e0 0x400d2092:0x3ffb3680 0x400e7a27:0x3ffb36c0 0x400e84b3:0x3ffb36e0 0x400d16db:0x3ffb3710 0x400d1a9b:0x3ffb3830 0x400941c1:0x3ffb3860

Rebooting...
```

sweetymhaiske, 2020-07-06:
This is currently a limitation of the underlying Wi-Fi provisioning component and the ESP IDF team is looking into how to handle it. Once the support is added there, it will be pulled into arduino

zenius06, 2020-07-15:
Hello,
First thank you so much to do this job, for me I have an issue to compile your example, the compiler don't found : wifi_prov_mgr_event_handler, wifi_prov_mgr_init', wifi_prov_mgr_is_provisioned, wifi_prov_scheme_ble_set_service_uuid
wifi_prov_mgr_start_provisioning, wifi_prov_mgr_deinit. If I look to the "esp-idf" I can found the source code into manager.c, but if I search for arduino I don't find the source code declared into manager.h, could you help me ? In advance thanks again for your work, best regards

maikeriva, 2020-09-15:
@zenius06 You are most likely not using the latest development version of arduino-esp32. Do not use the provided packages by espressif, rather simply use this repository's contents from its master branch.

closed 2020-11-29 by stale[bot]

smith-jordan-f, 2021-01-18:
It seems that the IDF may have some option for 'graceful handling' of bad credentials.  See below:

> Note
> 
> If the device fails to connect with the provided credentials, it won’t accept new credentials anymore, but the provisioning service will keep on running (only to convey failure to the client), until the device is restarted. Upon restart the provisioning state will turn out to be true this time (as credentials will be found in NVS), but device will again fail to connect with those same credentials (unless an AP with the matching credentials somehow does become available). This situation can be fixed by resetting the credentials in NVS or force starting the provisioning service. This has been explained above in Check Provisioning State.
> 

> If provisioning state needs to be reset, any of the following approaches may be taken :
> 
>         the associated part of NVS partition has to be erased manually
> 
>         main application must implement some logic to call esp_wifi_ APIs for erasing the credentials at runtime
> 
>         main application must implement some logic to force start the provisioning irrespective of the provisioning state
>