espressif/arduino-esp32 #4142: high light sleep power consumption

curly-brace, 2020-07-07:
### Hardware:
Board:							lolin32
Core Installation version: ARDUINO_ESP32_GIT_VER 0x46386288 :: ARDUINO_ESP32_RELEASE_1_0_4
IDE name: Platform.io (platform version 1.12.4)
Flash Frequency: 80Mhz
PSRAM enabled: no
Upload Speed: 115200
Computer OS: Windows 10

### Description:
After system start my esp32 consumes 78mA without any code running.

But when I enter light sleep with external interrupt wakeup, i'm getting 53 mA. Very little drop.

And when I change interrupt wakeup to timer wakeup it even decreases to 80 mA.

You may think that there is something in my circuit that drains power, but in flash mode it consumes only 23 mA and deep sleep consumes as little as 6 mA. So I assume that MCU doesn't go to sleep actually, but why?

I don't use WF or BT. Even tried disabling them manually. Tried disabling ADC too.

curly-brace, 2020-07-07:
It appears that it was TFT control pins RST, RS, CS that were draining the power.

But I don't understand why when setting no wakeup source, this is not happening? Esp32 automatically goes to deep sleep in this case??

lbernstone, 2020-07-07:
The system turns off unneeded power domains when it goes into sleep.
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/sleep_modes.html

curly-brace, 2020-07-07:
so it is a bug, because it turns power of periphery only if i don't set wakeup condition, it happens only with light sleep tho. deep sleep works fine. now i need to manually set outputs to inputs before sleeping in.

curly-brace, 2020-07-20:
no one is interested in a bug?

chegewara, 2020-07-20:
Even if it is a bug, then esp-idf is the place to report it.

closed 2020-10-03 by stale[bot]