espressif/arduino-esp32 #4145: Using PSRAM without ps_malloc in ESP2-WROVER-B

ccssw, 2020-07-08:
### Hardware:
Board:				ESP2-WROVER-B
Core Installation version:	??
IDE name:			Platform.io
Flash Frequency:		40Mhz
PSRAM enabled:                 yes
Upload Speed:			115200
Computer OS:                   Ubuntu

### Description:
How can I configure my ESP2-WROVER-B to allocate the object ```a``` and all its internal allocations in the PSRAM without using ```ps_malloc```? My code doesn't use the PSRAM.

I'm trying to use the configuration described [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/external-ram.html#integrate-ram-into-the-esp32-memory-map) and [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-spiram-use).

### Platformio.ini
```
[env:esp-wrover-kit]
board_build.partitions = /$PROJECT_DIR/custompart.csv
platform = espressif32
board = esp-wrover-kit
framework = arduino

upload_port = /dev/ttyUSB0
upload_speed = 921600

monitor_port = /dev/ttyUSB0
monitor_speed = 115200

build_flags =
;Debug level: 0 None, 1 Error, 2 Warn, 3 Info, 4 Debug, 5 Verbose
    -DCORE_DEBUG_LEVEL=3
    -DBOARD_HAS_PSRAM 
    -mfix-esp32-psram-cache-issue
        -DSPIRAM_CACHE_WORKAROUND=1
	-DSPIRAM_SUPPORT=1
	-DSPIRAM_BOOT_INIT=1
	-DSPIRAM_TYPE_AUTO=1
	-DSPIRAM_SIZE=-1
	-DSPIRAM_SPEED_40M=1
	-DSPIRAM_MEMTEST=1
	;-DSPIRAM_MALLOC_RESERVE_INTERNAL=32768
	;-DSPIRAM_MALLOC_ALWAYSINTERNAL=16384
	-DSPIRAM_USE_MEMMAP=1
    	; -DSPIRAM_USE_CAPS_ALLOC=1
    	; -DSPIRAM_USE_MALLOC=1
	-DSUPPORT_STATIC_ALLOCATION=1
	-DPICO_PSRAM_CS_IO=10
	-DSPIRAM_OCCUPY_VSPI_HOST
```

### main.cpp
```cpp
#include <Arduino.h>

unsigned int KB1   = 1000;
unsigned int KB10  = 10000;
unsigned int KB100 = 100000;
unsigned int MB1   = 1000000;

class A
{
public:	
	A()
	{
		this->array = new int8_t[KB10 * 5]; 
	};

	int8_t* myNewPointer(size_t r)
	{
		return new int8_t[r];
	};

	int8_t* myMalloc(size_t r)
	{
		return (int8_t*) malloc(r);
	};

	int8_t* array;
	
};

void printMemory()
{
	ESP_LOGI(TAG,"Total heap: %d", ESP.getHeapSize());
  	ESP_LOGI(TAG,"Free heap: %d", ESP.getFreeHeap());
	
  	ESP_LOGI(TAG,"Total PSRAM: %d", ESP.getPsramSize());
  	ESP_LOGI(TAG,"Free PSRAM: %d", ESP.getFreePsram());
}

A a;

void setup()
{			
	printMemory();	
}

void loop()
{	
	int8_t* b = a.myMalloc(KB1);
	if(b == nullptr) 
		ESP_LOGI(TAG,"b is null");
	
	int8_t* c = a.myNewPointer(KB1);
	if(c == nullptr) 
		ESP_LOGI(TAG,"c is null");

	vTaskDelay(pdMS_TO_TICKS(1000));
	printMemory();	
}
```

```
[I][main.cpp:33] printMemory(): ---
[I][main.cpp:34] printMemory(): Total heap: 380796
[I][main.cpp:35] printMemory(): Free heap: 305372
[I][main.cpp:37] printMemory(): Total PSRAM: 4194252
[I][main.cpp:38] printMemory(): Free PSRAM: 4194252
[I][main.cpp:33] printMemory(): ---
[I][main.cpp:34] printMemory(): Total heap: 380792
[I][main.cpp:35] printMemory(): Free heap: 303340
[I][main.cpp:37] printMemory(): Total PSRAM: 4194252
[I][main.cpp:38] printMemory(): Free PSRAM: 4194252
[I][main.cpp:33] printMemory(): ---
[I][main.cpp:34] printMemory(): Total heap: 380760
[I][main.cpp:35] printMemory(): Free heap: 301308
[I][main.cpp:37] printMemory(): Total PSRAM: 4194252
[I][main.cpp:38] printMemory(): Free PSRAM: 4194252
[I][main.cpp:33] printMemory(): ---
[I][main.cpp:34] printMemory(): Total heap: 380728
[I][main.cpp:35] printMemory(): Free heap: 299276
[I][main.cpp:37] printMemory(): Total PSRAM: 4194252
[I][main.cpp:38] printMemory(): Free PSRAM: 4194252
[I][main.cpp:33] printMemory(): ---
[I][main.cpp:34] printMemory(): Total heap: 380696
[I][main.cpp:35] printMemory(): Free heap: 297244
[I][main.cpp:37] printMemory(): Total PSRAM: 4194252
[I][main.cpp:38] printMemory(): Free PSRAM: 4194252
[I][main.cpp:33] printMemory(): ---
[I][main.cpp:34] printMemory(): Total heap: 380664
[I][main.cpp:35] printMemory(): Free heap: 295212
[I][main.cpp:37] printMemory(): Total PSRAM: 4194252
[I][main.cpp:38] printMemory(): Free PSRAM: 4194252
[I][main.cpp:33] printMemory(): ---
...
```

lbernstone, 2020-07-08:
You have to rebuild the compiled libraries in order to enable mmap for psram.  This can be done in platform.io with `framework = espidf, arduino` or much easier if you use the esp32s2 branch of this repo.

ccssw, 2020-07-08:
how can I use that specific branch in platform.io?

ccssw, 2020-07-08:
If I put ```framework = espidf, arduino``` in platformio.ini, when I build I get: 
```src/main.cpp:1:10: fatal error: Arduino.h: No such file or directory```

noukee, 2020-08-04:
Does anyone know how to use PSRAM with new (without using ps_malloc or malloc)?

closed 2020-10-17 by stale[bot]