espressif/arduino-esp32 #4147: OTA over serial: E (119053) esp_image: invalid segment length

Shuptuu, 2020-07-08:
Hi,
In a setup involving 2 ESP32 working in parallel, but only one with network access (wifi or ethernet), and with the ESPs communicating through Serial2, I created a way to upload a new firmware on the second ESP, using the first one as a bridge.
So for the user, it works like a web OTA, but the image is forwarded to the second ESP through the uart2.
Works really well, but... of course, there's a "but"...
Sometimes, I get an error and the OTA fails.
What I get on the monitoring tool:

> Try to start update (703520) 
> Written : 703520 successfully
> E (119053) esp_image: invalid segment length 0x31703f
> Error Occurred. Error #: 9 

Some parts of my code (I'm using Arduino IDE):
**UART setup on both devices**
```javascript
  Serial2.begin(256000, SERIAL_8N1, RX2, TX2);
  Serial2.setRxBufferSize(4096);
```
**On the first ESP (running a webserver)**
```javascript
  server.on("/updateB", HTTP_POST, []() {
    server.sendHeader("Connection", "close");
    server.send(200, "text/plain", "OK");
    delay(100);
    ESP.restart();
  }, []() {
    HTTPUpload& upload = server.upload();
    // Check if file seems valid .bin file
    if(!upload.filename.endsWith(F(".bin"))){
      server.send(500, F("text/plain"), F("ONLY BIN FILES ALLOWED\n"));
      return false;
    }
    int bufsize=0;
    if (upload.status == UPLOAD_FILE_START) {
      ota=1;
      Serial.setDebugOutput(true);
      UdpOsc.close();
      UdpPriv.close();
      udpstop();
      Serial.printf("Update B: %s\r\n", upload.filename.c_str());
      Serial2.write(FUPDATE);
      Serial2.write(filesize.b,4);          
    } else if (upload.status == UPLOAD_FILE_WRITE) {
      bufsize=upload.currentSize;
      Serial2.write(upload.buf, upload.currentSize);
    } else if (upload.status == UPLOAD_FILE_END) {
      //Serial2.write(FUPDATE-E);
      Serial.printf("Total sent = %u\r\n", upload.totalSize );
      Serial.setDebugOutput(false);
      udpconfig();
      delay(100);
      udpreceive();
      ota=0;
    } else {
      Serial.printf("Update Failed Unexpectedly (likely broken connection): status=%d\r\n", upload.status);
      Serial.setDebugOutput(false);
      udpconfig();
      delay(100);
      udpreceive();
      ota=0;
    }
 });
```
**On the second ESP**
```javascript
void mySerialAnswers (){
  cmd=Serial2.read();
  switch (cmd) {
    case FUPDATE:{
      long filesize=getlongfromEspA();
      if (filesize > 0) {
         Serial.printf("Try to start update (%u) \r\n", filesize );
         performUpdate(Serial2, filesize);
      }
    }break;
 }

void performUpdate(Stream &updateSource, size_t updateSize) {
   if (Update.begin(updateSize)) {
      while (!updateSource.available()){}      
      size_t written = Update.writeStream(updateSource);
      if (written == updateSize) {
         Serial.println("Written : " + String(written) + " successfully");
      }
      else {
         Serial.println("Written only : " + String(written) + "/" + String(updateSize) + ". Retry?");
      }
      if (Update.end()) {
         Serial.println("OTA done!");
         if (Update.isFinished()) {
            Serial.println("Update successfully completed. Rebooting.");
            ESP.restart();
         }
         else {
            Serial.println("Update not finished? Something went wrong!");
         }
      }
      else {
         Serial.println("Error Occurred. Error #: " + String(Update.getError()));
      }

   }
   else
   {
      Serial.println("Not enough space to begin OTA");
   }
}
```
So the question is why it works most of the time and fails just sometimes????
Thanks in advance for any help!

lbernstone, 2020-07-08:
Does it work more reliably at lower baud rates?  Occasional issues like this over the wire are often due to noise.  Look at the sheer number of upload issues we get due to bad cables.

Shuptuu, 2020-07-09:
Didn't test it at lower baud rates. Gonna do it. But I would be more interested by a higher baud rate! I need those two ESP to communicate as quick as possible. But I've not been able to  get a stable communication with more than 256000 b/s... And for the bad cable option: those 2 ESPs are soldered on a PCB. There's maybe 1cm between them. And wires on the pcb have been kept very short between them... Do you think by adding pullup resistors could make a difference?

lbernstone, 2020-07-09:
pullups are unlikely to help with transient noise, but who knows with EMI issues.  Better would be to ensure that there is distance b/w the lines on the board to provide adequate shielding from each other (and anything else they might be near).

Shuptuu, 2020-07-13:
Tests done:
-So, pullups don't change anything.
-And slowing done the baud rate doesn't change anything either. (test made with 115200)
Well I'm not sure the problem is linked to the wires. The strange thing is that Data seems to be correctly transfered and written on the second ESP (written bytes = file size). 
As I said, on my pcb, wires are really short, there's a good shielding, without any difference with wires for UART0. But through UART0 and a USB to uart interface , I'm able to upload a firmware up to 921600b/s... 
So what's the problem with OTA and/or other UARTs inside the ESP?     
Any Idea?

closed 2020-09-26 by stale[bot]

jaigarg2, 2023-03-14:
Can any one help on this I am facing the same issue