espressif/arduino-esp32 #4148: PC: 0x40170620: circular_queue ::push(unsigned int&&)

georgepopovici, 2020-07-08:
################
Backtrace:
.Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Core 1 register dump:
PC      : 0x40170620  PS      : 0x00060034  A0      : 0x8008135c  A1      : 0x3ffbe780  
A2      : 0x3ffc1288  A3      : 0x00000001  A4      : 0x0a417eb9  A5      : 0x3ffbe790  
A6      : 0x3ffbe7d8  A7      : 0x00000001  A8      : 0x8008125a  A9      : 0x00000000  
A10     : 0x3ffafdc8  A11     : 0x3ffbe78c  A12     : 0x8008ad87  A13     : 0x3ffbe760  
A14     : 0x00000008  A15     : 0x00000001  SAR     : 0x00000001  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
Core 1 was running in ISR context:
EPC1    : 0x40087435  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40170620

Backtrace: 0x40170620:0x3ffbe780 0x40081359:0x3ffbe7b0 0x40085185:0x3ffbe7d0 0x40087432:0x3ffba190 0x40083583:0x3ffba1b0 0x40089591:0x3ffba1d0

################
Decoded backtrace:

PC: 0x40170620: circular_queue ::push(unsigned int&&) at C:\Users\gpopovici\Documents\Arduino\libraries\EspSoftwareSerial\src\circular_queue/circular_queue.h line 269
EXCVADDR: 0x00000000

Decoding stack results
0x40170620: circular_queue ::push(unsigned int&&) at C:\Users\gpopovici\Documents\Arduino\libraries\EspSoftwareSerial\src\circular_queue/circular_queue.h line 269
0x40081359: __onPinInterrupt at C:\Users\gpopovici\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\esp32-hal-gpio.c line 232
0x40087432: spi_flash_op_block_func at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/cache_utils.c line 81
0x40083583: ipc_task at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/ipc.c line 62
0x40089591: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

#################
Serial output:

Going to sleep now
ets Jun  8 2016 00:22:57

rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
Boot number: 7
Wakeup caused by timer
Setup ESP32 to sleep for every 15 Seconds
Simple TinyGPS library v. 13
by Mikal Hart

Attempting to connect to SSID: priv1
.Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Core 1 register dump:
PC      : 0x40170620  PS      : 0x00060034  A0      : 0x8008135c  A1      : 0x3ffbe780  
A2      : 0x3ffc1288  A3      : 0x00000001  A4      : 0x0a4b349d  A5      : 0x00000000  
A6      : 0x3ffc1a50  A7      : 0xffffffff  A8      : 0x8008125a  A9      : 0x00000000  
A10     : 0x3ffafdc8  A11     : 0x3ffbe78c  A12     : 0x3ffba264  A13     : 0x0000abab  
A14     : 0x3ffc384c  A15     : 0x3ffba264  SAR     : 0x00000001  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
Core 1 was running in ISR context:
EPC1    : 0x40087440  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40170620

Backtrace: 0x40170620:0x3ffbe780 0x40081359:0x3ffbe7b0 0x40085185:0x3ffbe7d0 0x4008743d:0x3ffba190 0x40083583:0x3ffba1b0 0x40089591:0x3ffba1d0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
[... 277 lines cut ...]
  }
  
  Serial.println("Going to sleep now");
  delay(1000);
  Serial.flush(); 
  esp_deep_sleep_start();
  Serial.println("This will never be printed");
  //delay(20000); // Wait 20 seconds to update the channel again

}

void loop() {
  
}

void sendCallback(SendStatus msg) {
  // Print the current status
  Serial.println(msg.info());

  // Do something when complete
  if (msg.success()) {
    Serial.println("----------------");
  }
}
```

lbernstone, 2020-07-08:
You are trying to access something in an ISR that is not in IRAM.  This is often something trying to access flash.  You need to figure out what component is causing the issue.  Using SoftwareSerial is not recommended on esp32.

georgepopovici, 2020-07-09:
Thanks @lbernstone ! Would you be able to provide more details on how to figure out what component is causing the issue? Maybe some debug code that I could add or an approach I could follow to narrow this down?

Regards,
George

georgepopovici, 2020-07-09:
An update from my side:

1. Removed the software serial part and dependent code - eps32 board does not crash anymore
2. Added hardware serial replacement for the above:
#define RXD2 16
#define TXD2 17
 Serial2.begin(9600, SERIAL_8N1, RXD2, TXD2);
and used this serial for getting GPS data
This worked and there's been no crashes.

Thanks,
George

closed 2020-07-13 by georgepopovici