espressif/arduino-esp32 #4149: Support SPIFFS library for Arduino MKR Wifi 1010

anaouss, 2020-07-09:
Hello,

I'm working on web page project with Arduino MKR wifi 1010, I wrote my code HTML and I'm trying to control the pins of my Arduino with this page web, but the spiffs doesn't work on MKRwifi. (I got this error: Error compiling for board Arduino MKR WiFi 1010)

Could you help me ?

Regards,
Thanks in advance,
Anass

atanisoft, 2020-07-09:
This repository is not suitable for the Arduino MKR WiFi boards. You should use the boards setting as defined in the ide for the board and if you still have issues raise it in the Arduino community forum.

closed 2020-09-22 by stale[bot]

harsha7addanki, 2021-05-01:
stale yo rude

atanisoft, 2021-05-02:
@harsha7addanki As noted previously, this repository does not support the Arduino MKR WiFi boards which does not use an Espressif chip. It is based on the ARM Cortex-M0 processor with a u-blox NINA-W102. Nothing in this repository would work on that architecture.

stalebot did the correct thing here by closing an issue that was abandoned by the filer which should have closed it themselves since it was not filed to the correct team/company.