espressif/arduino-esp32 #4151: can't change wifi name and pass  on WiFi.softAP(ssid, password);

jimmisitepu88, 2020-07-10:
I Try active wifi access point on esp32 with dip switch, 
but odd, the wifi name and pass can't change as configure WiFi.softAP(ssid, password);
wifi name and pass still use program before

this my code

#include <WiFi.h>
#include <WiFiClient.h>
#include <WiFiAP.h>
WiFiServer server(80);

bool hold_active_wifi=false;

const byte inset1 = 34;
const byte inset2 = 35;

const char* ssid     = "kwh_jimmi";
const char* password = "123456";

#define ON 0
#define OFF 1

void setup() {
  // put your setup code here, to run once:
 Serial.begin(115200);
 pinMode(inset1, INPUT);
 pinMode(inset2, INPUT);

 if(digitalRead(inset1) == ON){
  if(hold_active_wifi == false){
    active_wifi();
    hold_active_wifi = true;
  }
  
  while(1){
    if(digitalRead(inset1) == OFF){
      ESP.restart();
    }
    WiFiClient client = server.available();
  }
 }
 
}

void loop() {
  // put your main code here, to run repeatedly:
 if(digitalRead(inset1) == ON){
    ESP.restart();
  }
  Serial.println("aplikasi running...");
  delay(1000);
}

void active_wifi(){
  Serial.println("active wifi...");
  WiFi.softAP(ssid, password);
  IPAddress IP = WiFi.softAPIP();
  Serial.print("AP IP address: ");
  Serial.println(IP);  
  server.begin();
}

lbernstone, 2020-07-10:
This site is for issues with the arduino-esp32 code, not your logic.  If you cut this to just the code in active_wifi(), does it start the AP with the correct credentials?

kjitender469, 2020-09-11:
@jimmisitepu88 make your password atleast 8 character long. It will work.

closed 2020-11-24 by stale[bot]