espressif/arduino-esp32 #4157: Using FreeRTOS xQueueCreateStatic:

motorek258, 2020-07-11:
### Hardware:
Board:							M5Stack Fire
Core Installation version:			1.0.4
IDE name:						Platform.io
PSRAM enabled:                                    Yes
Computer OS:                                        Mac OSX

### Description:
I'm trying to make use of a FreeRTOS static queues (xQueueCreateStatic) allocated in PSRAM.

I have:
`#define configSUPPORT_STATIC_ALLOCATION  1`

getting following error after compiling with with VSCode / PlatformIO:

.pio/build/m5stack-fire/src/main.cpp.o:(.literal._Z5setupv+0x50): undefined reference to `xQueueGenericCreateStatic'
.pio/build/m5stack-fire/src/main.cpp.o: In function `setup()':
/Users/Mikka/Documents/PlatformIO/Projects/PQA_Tester_Productive/src/main.cpp:87: undefined reference to `xQueueGenericCreateStatic'
collect2: error: ld returned 1 exit status
*** [.pio/build/m5stack-fire/firmware.elf] Error 1

- xQueueCreate (not static) works fine. The error comes up after compiling / while linking. Compiling works.

this is as the only reference to xQueueCreateStatic in my code:
`rx_queue_static  = xQueueCreateStatic(QUEUE_LENGTH, ITEM_SIZE, pointer_to_record, &xStaticQueue_1);`

I tried as well to import the newest queue.h from FreeRTOS.org, with no success...

lbernstone, 2020-07-11:
In order to change CONFIG_ settings, you must [recompile](https://github.com/espressif/arduino-esp32/blob/master/docs/lib_builder.md) the esp-idf libraries.  The esp32s2 branch has PSRAM memmapped into the default allocator, so this is probably an easier route ATM.  Note that you should typically not put something into psram that needs to have quick or ISR access, like queues.

chegewara, 2020-07-11:
I have app with 1500 elements queue created with xQueueCreate (not static) and it is automatically created in psram and works fine:

```
typedef struct{
    char topic[50];
    char data[250];
    size_t len;
}log_msg_t;
xQueueCreate(1500, sizeof(log_msg_t));
```

motorek258, 2020-07-12:
> In order to change CONFIG_ settings, you must [recompile](https://github.com/espressif/arduino-esp32/blob/master/docs/lib_builder.md) the esp-idf libraries. The esp32s2 branch has PSRAM memmapped [...]

Thanks for the explanation. I managed to recompile the esp-idf libraries using instructions from your link. Before building I changed the only thing in the sdkconfig:

`CONFIG_SUPPORT_STATIC_ALLOCATION=y`

Then I copied the created files from /out/tools/sdk to the platformio dir (/Users/mikka/.platformio/packages/framework-arduinoespressif32/tools) and replaced the original ones.

Unfortunately I am still getting the same error after compiling my project :(

Is there anything else I could try to get it running? I need a big buffer/queue to store up to 60s of CAN-Bus data, before I write them to the SD card or just delete them. The PSRAM seemed to be an ideal place for this buffer. Reduced speed (in comparison to an internal RAM) shouldn't be an issue in my case. Since I am using the ESP32 (without S2), the PSRAM can only be allocated manually...? 

@chegewara: Thank you for the code. Are you using the ESP32S2? I am running into "out of memory" error with your code in platformio/arduino - the queues created with xQueueCreate are always placed in the DRAM in my case (I'm already using some). BTW the PSRAM is enabled and I can successfully allocate the space for my big buffer with ps_malloc().

chegewara, 2020-07-12:
Its old esp32 with esp-idf v3.3

motorek258, 2020-07-13:
Ok, after few another tries I finally managed to run it.

To start from a scratch I reinstalled the platformio arduino-esp32 component (just deleted the dir /.platformio/packages/framework-arduinoespressif32), platformio downloads the whole framework again while building the ESP32 arduino project (this is not necessary, but I didnt know anymore which files I changed while trying to run it). 

Then I have build the v3.3 branch, following the instructions in the link from lbernstone with CONFIG_SUPPORT_STATIC_ALLOCATION=y. After that I took the libfreertos.a from /esp32-arduino-lib-builder/out/tools/sdk/lib (FreeRTOS queue functions) and replaced the old one in /.platformio/packages/framework-arduinoespressif32/tools/sdk/lib

Thank you!

closed 2020-07-13 by motorek258